lumapu/ahoy #843: keine Verbindung zum WR nach Upgrade von 0.5.17 auf 0.6.0

CBRocker, 2023-04-01:
## Hardware
  * [x] ESP8266

Modelname: ESP8266MOD 12

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [x] circuit board

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [x] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.6.0
Github Hash: 9c7d7a0

### Build & Flash Method:
* [ ] Arduino
* [x] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [x] Setup settings (use our templates ... to be added)

Hallo,
ich hatte bisher die Version 0.5.17 Ahoy DTU auf einem ESP8266, lief alles problemlos. Daten per MQTT an ioBroker übertragen.
Heute habe ich nun auf die Version 0.6.0 aktualisiert, hatte den ESP per USB-Kabel am PC. 
Installation war erfolgreich, ich konnte alle Einstellungen vornehmen.
Aber ich bekomme keine Daten vom Hoymiles HM-600.
Vor der Aktualisierung habe ich mir Screenshots von den wichtigsten Einstellungen im Setting von 0.5.17 gemacht und übernommen.

Wo ist der Fehler, daß ich noch keine Daten vom HM sehe bzw. der nicht verfügbar ist.
Hier ein paar Screenshots:

[image]

[image]

[image]

[image]

[image]

12:08:36 I: resetPayload: id: 0
12:08:36 I: (#0) enqueCommand: 0x01
12:08:36 I: (#0) enqueCommand: 0x0b
12:08:36 I: (#0) prepareDevInformCmd 0x01
12:08:36 I: TX 27B Ch75 | 15 81 02 13 03 82 66 73 72 80 01 00 64 28 02 a4 00 00 00 00 00 00 00 00 22 30 1a 
12:08:56 I: resetPayload: id: 0
12:08:56 I: (#0) prepareDevInformCmd 0x0b
12:08:56 I: TX 27B Ch3 | 15 81 02 13 03 82 66 73 72 80 0b 00 64 28 02 b8 00 00 00 00 00 00 00 00 e8 a2 54 

Mit 0.5.17 hatte ich keine Probleme, alles einzustellen. Hat auf Anhieb funktioniert.
Ich bitte um Hilfe.
Vielen Dank,

Grüße
Christoph

tastendruecker123, 2023-04-01:
Die Pins für LED0 und LED1 sollten auf "off" stehen. Die überschneiden sich bei dir mit IRQ und daher funktioniert der Empfang nicht. Ist ein Duplikat von issue #832.

CBRocker, 2023-04-01:
@tastendruecker123 danke, habe ich geändert. Aber keine Besserung

_Inverter #0: HM-600 (v0) is not yet available_

duracell, 2023-04-01:
Ähnliches Problem hier, nach einem Update auf 0.6.0 gibt's die Meldung:
> your NRF24 module can't be reached, check the wiring and pinout

sobald ich zurück auf v0.5.66 gehe klappt alles wieder.

[image]
[image]

CBRocker, 2023-04-01:
@duracell ich bin im Moment auch wieder auf meine ursprüngliche 0.5.17 zurückgegangen, dann war nach einem holprigen Start auch alles wieder da. Ich dachte, das Flashen auf 0.6.0 geht schnell in ner halben Stunde. Wollte dafür jetzt nicht den ganzen Samstag damit verbringen...
Sag mal, konntest du bei Power Limit Value was eintragen? Bei mir kommt da dann Ctrl Result: Error
[image]

Ollipop030, 2023-04-01:
@duracell Bei ESP32 musst du die Pins neu konfigururieren, siehe #812

duracell, 2023-04-01:
> @duracell Bei ESP32 musst du die Pins neu konfigururieren, siehe #812

Danke! Das war's. Sorry, dann passte das hier wohl nicht rein, weil es ein Problem beim ESP32 war. 

> @duracell ich bin im Moment auch wieder auf meine ursprüngliche 0.5.17 zurückgegangen, dann war nach einem holprigen Start auch alles wieder da. Ich dachte, das Flashen auf 0.6.0 geht schnell in ner [...]

Erstmal sorry für den Hijack des Tickets, dachte das Problem wäre das gleiche, aber ist wohl nicht der Fall.
Nachdem fix für mein Problem klappt das eintragen ohne Probleme. Ich gehe davon aus, dass er ohne eine Verbindung die Daten ja nicht senden kann. Also falls das jetzt mit der neuen Version der Fall war.

sweinreuter, 2023-04-02:
Guten Morgen. Hatte den gleichen Effekt nach Update von 0.5.66 auf 0.6.0 auf einem ESP8266. Nach dem Update habe ich die "disable night communication" Einstellung aktiviert, danach war bis auf WLAN die ganze Config weg. Alles neu eingetragen und jetzt genau das Problem: Es werden keine Daten empfangen. Die Verbindung zum RF-Modul ist aber wohl da. Die Konsolen-Ausgabe sah genau so aus wie bei CBRocker.

Der Versuch per OTA-Update auf 0.5.66 zurück zu gehen, hat jetzt aber alles gekillt, Ahoy verbindet sich nicht mehr mit dem WLAN und macht auch keinen AP auf. Da das Ding bei meinen Eltern steht, ist jetzt erst mal Feierabend. 

VG
Sascha

CBRocker, 2023-04-02:
@sweinreuter 
Hey, ich habe dann mit dem ESP Flashtool von Espressif die 0.5.17 komplett neu geflasht und alles wieder eingerichtet. Hatte zuvor Screenshots von den Einstellungen gemacht. 
Ich möchte, sobald ich Zeit habe, einen 2. ESP mit 0.6.0 nochmal probieren. Mein produktiv system rühre ich nicht mehr an, bis die 0.6.0 läuft. Habe noch genug ESPs rumliegen

sweinreuter, 2023-04-03:
So, nochmal kurzes Update. Die 0.5.66 läuft wieder, nachdem ich die mit esptool und "erase flash" drauf gepackt habe. Ohne das Erase war ein Downgrade nicht möglich. Sollte das OTA-Update evtl. ein Downgrade verhindern oder die Option für ein Erase anbieten?

Die 0.6.0 habe ich nicht zum Laufen bekommen - evtl. auch deshalb, weil ich ungefähr 10 mal das falsche WLAN-Passwort versucht habe ;-( - aber mal unabhängig davon hat sich damit das UI sehr komisch verhalten, da hat manchmal der Inhalt des Feldes "Device Name" gefehlt, dann war der Inhalt (bzw. Teile davon) der Inverter-Config mal weg und wie gesagt, beim ersten Mal war nach dem Aktivieren der "Disable Night Communication"-Einstellung erst mal alles weg. Da scheint ja auf jeden Fall was nicht ganz zu passen.

Den nächsten Versuch damit werde ich hier zu Hause machen, wenn mehr Zeit ist, dann kann ich evtl. auch genauere Informationen liefern.

ThiloGa, 2023-04-05:
> Die Pins für LED0 und LED1 sollten auf "off" stehen. Die überschneiden sich bei dir mit IRQ und daher funktioniert der Empfang nicht. Ist ein Duplikat von issue #832.

Für mich war das die Lösung....

CBRocker, 2023-04-07:
Hallo,
ich habe heute einen anderen ESP genommen und die Pin-Verkabelung neu gemacht, damit ich mit meinem produktiven ESP mit 0.5.17 nicht rumspielen muß. Ich war erst irritiert, weil bei den 2 fritzing-Schemata zum ESP8266 NodeMCU einmal 8 Pins und einmal nur 7 Pins eingezeichnet waren. Die Community bei Discord hat mir geholfen, außerdem auch gleich den Tip gegeben, die Dev. Version 0.6.4 zu installieren. Läuft bisher perfekt. Ich schließe den Issue als gelöst

closed 2023-04-15 by lumapu