lumapu/ahoy #844: Nokia 5110 Display Connection and Settings

jdrescher2006, 2023-04-01:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: D1 Mini NodeMcu ESP8266-12F
Retailer URL: https://www.azdelivery.de/products/d1-mini

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X] circuit board
* [ ] external antenna

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [X] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.6.0
Github Hash: 9c7d7a0

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [X] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

[image]

Hi, I am trying to get a Nokia 5110 Display to work. I connected as in the attached wiring.

In the table you can see the configuration.

| Display | D1 Mini | Ahoy | Ahoy Auswahlbox |
|---------|---------|------|-----------------|
|CE | D1 | CS | D1 (GPIO5, SCL) |
|DC | D2 | DC | D2 (GPIO4, SDA) |
|DIN | TX | DATA | TX (GPIO1) |
|CLK | RX | CLOCK | RX (GPIO3) |

The problem is there is nothing seen on the display, it just does not work.

[image]

knickohr, 2023-04-01:
Try using the D1 or D2 pin on D0. Don’t forget to Change the settings.

jdrescher2006, 2023-04-01:
Ok, so I connected CE with D0, and DC with either D1 or D2. Sometimes I have now a faint "AHOY" text on the display.

Has anyone connected this display to a D1 MINI and got it really working?

knickohr, 2023-04-01:
Ahh, you have a faint picture. Maybe you have a worse Display ?  Try pushing with a gentle force on the metallic Frame. A Lot auf cheap Displays have this issue.

bojs15, 2023-04-02:
my screen only lights up, are these two capacitors important??

jdrescher2006, 2023-04-02:
So I followed this guide:
[https://thesolaruniverse.wordpress.com/2021/01/15/nokia-5510-8448-lcd-and-the-esp8266-wemos-d1-mini-microcontroller-board/](url)
And voila: the display is working absolutely properly!

Reconnected it to ahoy: does not work. Sometimes a faint picture, thats all.

Notice that in the guide I linked, the reset pin is also connected to the D1 mini. Maybe this is needed?

bojs15, 2023-04-02:
[jdrescher2006](https://github.com/jdrescher2006)   dein link ohne funktion ??

jdrescher2006, 2023-04-03:
Sorry, habs korrigiert 👍

lumapu, 2023-04-05:
solved?

jdrescher2006, 2023-04-06:
> solved?

No.

I ordered an other display from china, just to make sure it's no hardware problem.

I made some tests and have the feeling that the problem may come from the used library u8g2.
Because with Adafruit_PCD8544 lib the display is working absolutely properly.

jdrescher2006, 2023-04-18:
Its working properly now. It was a hardware problem with my display. My connection is looking like:
Clock: D1
Data: D2
CS: RX
DC: TX

closed 2023-04-18 by jdrescher2006