lumapu/ahoy #845: Inverter An/Aus über MQTT nicht möglich

Telekatz, 2023-04-02:
### Platform

ESP32

### Model name

ESP32 DEVKITV1

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

bb5cb8b

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 114182903084
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 5
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 192.168.69.173
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Nach dem Update auf Version 0.6.0 funktioniert das Ein und Ausschalten des Inverters über den MQTT Topic "solar/ctrl/power/0" nicht mehr. Limitänderungen über "inverter/ctrl/limit/0" werden ausgeführt.

mager33, 2023-04-05:
Sicher, dass Du die richtigen Topics sendest? Entweder "solar/..." ODER "inverter/..." !!

Telekatz, 2023-04-05:
War schon der richtige Topic. Hab mich hier in der Beschreibung vertan.

SergejSuslik19877891, 2023-04-06:
Habe das gleiche Problem bei Version 6.0 als auch bei Version 6.3

beegee3, 2023-04-06:
es gibt kein Subscription mehr für "ctrl/power". Daher funktioniert das nicht mehr. Ist auch nicht mehr in der aktuellen Doku.
Ist wohl bei der Änderung von "ctrl/limit" weggefallen. @lumapu war das Absicht oder ein Versehen?

lumapu, 2023-04-06:
es ist ein Versehen, ich hatte da optimiert und war wohl zu eifrig 😀

commit f946b35d (2023-04-11): 0.6.5
* fix #845 MqTT subscription for `ctrl/power/[IV-ID]` was missing
closed 2023-04-19 by lumapu