lumapu/ahoy #847: [ESP8266] Display 0,96" SSD1306 geht nicht V0.6.0

PV-Sued-35b, 2023-04-02:
### Platform

ESP8266

### Model name

Wemos

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 10
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW

 #### Pinout Disp SSD1306
SCL > D1
SDA > D2
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
I: resetPayload: id: 0
I: (#0) enqueCommand: 0x0b
I: (#0) prepareDevInformCmd 0x0b
I: TX 27B Ch75 | 15 84 02 38 03 85 52 69 78 80 0b 00 00 00 00 00 00 00 01 80 00 00 00 00 c5 a1 00 
I: (#0) nothing received
I: resetPayload: id: 0
I: (#0) enqueCommand: 0x0b
I: (#0) prepareDevInformCmd 0x0b
I: TX 27B Ch3 | 15 84 02 38 03 85 52 69 78 80 0b 00 00 00 00 00 00 00 01 80 00 00 00 00 c5 a1 00 
W: (#0) Frame 2 missing: Request Retransmit
I: TX 11B Ch23 | 15 84 02 38 03 85 52 69 78 82 ec 
W: (#0) Frame 2 missing: Request Retransmit
I: TX 11B Ch40 | 15 84 02 38 03 85 52 69 78 82 ec 
I: procPyld: cmd:  0x0b
I: procPyld: txid: 0x95
I: alarm ID incremented to 386
I: (#0) enqueCommand: 0x11
I: resetPayload: id: 0
I: (#0) prepareDevInformCmd 0x11
I: TX 27B Ch61 | 15 84 02 38 03 85 52 69 78 80 11 00 00 00 00 00 00 00 01 82 00 00 00 00 df c3 60 
I: procPyld: cmd:  0x11
I: procPyld: txid: 0x95
I: Alarm #2 'DTU command failed' start: 23:59:50, end: 11:59:50
```

### Error description

nach update  von 0566 auf 060 steht im Display nur 
AHOY! 
ahoydtu.de
0.6.0

Setting auf 1106 brachte keine Änderung
nur Display 180 drehen und zurück funktioniert

PV-Sued-35b, 2023-04-03:
nachdem der Inverter spannungsfrei(Nacht) war, kommt die Anzeige nun :-)

lumapu, 2023-04-04:
also gelöst?

schulle61, 2023-04-09:
Gibt es dazu schon eine Lösung? Weil seit dem Update geht mein Display auch nicht mehr

matrix1000, 2023-04-11:
Hi, 

ja, bei mir war die Config für das Display weg. Einfach wieder einstellen.  [altes Bild](https://user-images.githubusercontent.com/99898355/224476694-b886772c-0b9c-423d-9c3c-04b8bd7238ee.png)
Und beachten: Turn off while inverters are offline [x] [->] dann geht das Display auch aus, wenn der Inverter Offline ist.

Dann geht es auch in v0.6.0 auf dem esp8266.

schulle61, 2023-04-11:
Danke Matrix 1000.
Die Einstellungen habe ich gestern auch gefunden.  Einfach umgestellt und schon ging es wieder...

closed 2023-04-19 by lumapu