lumapu/ahoy #848: Verhindern der YieldDay Rücksetzung bis Tagesende

knickohr, 2023-04-02:
Seit ein paar Tagen fällt mir auf das der Yield Day hin und wieder Abends zurück gesetzt wird wenn sich ein Inverter ausschaltet und dann doch nochmal aufwacht.

Das Ganze ist wetterbedingt. Hin und wieder kommt es Abends vor das die Sonne durch Wolken abgeschattet wird, der Inverter geht offline. Kommt die Sonne dann noch nochmal durch die Wolken, dann startet der Inverter neu und resettet dabei sein YieldDay.

Das Rücksetzen des YieldDay der einzelnen Inverter sollte verhindert werden bis zum Tagesende, meinetwegen Mitternacht.

Ich könnte mir auch vorstellen das es Morgens auftritt, da fällt es halt nur weniger auf.

beegee3, 2023-04-03:
der YieldDay Reset kommt vom Inverter selbst, wenn er sich ausschaltet und dann doch nochmal aufwacht.
Das wurde in #773 schon mal angesprochen. Eine Korrektur seitens Ahoy ist wahrscheinlich sehr aufwändig (wegen der variablen "Reset" Möglichkeiten).

knickohr, 2023-04-03:
Verdammt, ich habe mir fast schon sowas gedacht 😢 ich muß das mal beobachten ob der Inverter dann nur wirklich bei 0 anfängt oder ob er doch noch zum alten Wert dazu addiert, falls er wieder richtig aufwacht am selben Tag.

Letzteres wäre ideal, dann könnte ich einfach per Max. Wert abfragen und mir es zu Mitternacht selbst resetten.

sumerland, 2023-06-27:
Ich habe gestern das gleiche Problem beobachtet, als bei schlechtem Wetter der Himmel zu lange zu dunkel war.

Könnte man nicht in einem regelmäßigem Intervall eine Kopie des vorigen YieldDay Werts mit dem aktuellen vergleichen, und im Fall des Resets eine Kopie des alten Werts in einer Variablen (beispielsweise YieldDayMax, die zu Beginn des Tages zu Null gesetzt wird) aufaddieren? Über MQTT könnte man dann eine alternative YieldDay Variante anbieten, die die Summe aus aktuellem YieldDay (vom Inverter) und dem Wert von YieldDayMax ist. Um Mitternacht wird YieldDayMax dann wieder genullt.

knickohr, 2023-06-27:
Im Node-Red mache ich das schon. Einfach nur den Max-Wert halten. Mir geht es weniger darum das da nachdem ein Inverter aussteigt und wieder neu anfängt, sondern eher darum das eben das Total weg ist. So habe ich zumindest das aufsummierte über den Tag. Die paar Wattstunden die dann am Abend fehlen machen das Kraut nicht fett. Und wenn es frühmorgens passiert ist es erst gar nicht schlimm.

Schön wäre halt, wenn es aus der DTU halt korrekt raus kommt. Dann wäre auch der Wiedereinstieg mit Hilfe einer Variable leicht machbar, denn die DTU weiß ja ganz genau welche Inverter einen kurzen Blackout hatten. Aber das muß halt jemand programmieren der etwas von diesem C, C++ und weißgottwas versteht.

knickohr, 2023-08-18:
Ich rolle das Thema nochmal auf und stresse es etwas 😅

Kann man die YieldDay Rücksetzung der einzelnen Inverter nicht unterbinden bis die DTU am Tagesende in den Status offline geht ?  So würde selbst bei einem Inverterausfall, -Reboot, oder Gewitter das YieldDay des Inverters bei Neustart dort weitergezählt werden wo es aufhörte.

lumapu, 2023-09-16:
besteht das Problem immer noch? Oder hat die neue Statemachine jetzt hier für Ruhe gesorgt oder gar die Heuristik der Kanalvorhersage?

knickohr, 2023-09-17:
Das Problem wird immer bestehen solange die Inverter unter Tags offline gehen, bzw. sogar die komplette DTU (alle Inverter). Ich denke das Problem kann man so nicht lösen, außer man bindet die Rücksetzung an den Night Behavour oder Mitternacht. Wie man das aber am sinnvollsten macht und wie es dann allen Recht ist, steht auf einem anderen Blatt.

sumerland, 2023-09-18:
Ich bin auf 0.7.36 unterwegs und heute Abend wurde YieldDay wieder wetterbedingt vorzeitig resettet.

lumapu, 2023-09-18:
wie viele Wh wurden danach noch geloggt? Evtl. macht es Sinn ab diesem reset einfach nicht mehr zu loggen (kein MqTT mehr)

sumerland, 2023-09-18:
Leider gar keine mehr. Es war eh schon kurz vor Tagesende. Ich finde knickohrs Vorschlag eigentlich gut - Rücksetzen bis Sonnenuntergang oder Mitternacht (ggf. als Option in der GUI anwählbar).

knickohr, 2023-09-18:
Das Problem ist das halt der Inverter zurück setzt und wieder bei Null anfängt. Somit müßte man sich in der DTU den alten Wert speichern und was dann noch kommt aufaddieren.

lumapu, 2023-09-18:
stimmt, die DTU lebt ja noch, daher weiß die ja Bescheid. Das klingt machbar, damit wären wir für die nächste Sonnenfinsternis gerüstet 😊

sumerland, 2023-09-19:
Keine Scherze bitte :-) Ein dunkler Gewitterhimmel reicht schon. Oder gestern der Regen am späten Nachmittag und das kurze Aufklaren danach.

beegee3, 2023-09-19:
@lumapu: folgende Idee, um das in den Griff zu bekommen:
was man braucht, sind für jeden channel von jedem Inverter für YieldDay ein Offset und den letzten Messwert (kann man vielleicht als zusätzliche Felder in `deviceFieldAssignment` einbauen). Nenne sie hier mal offYD und lastYD.
Wenn ich das richtig sehe, werden neue Werte in `addValue` bzw. `setValue` (bei MI Invertern) verarbeitet.
Im Fall FLD_YD müsste man nun den aktuellen Wert (hier actYD) mit lastYD vergleichen. Ist lastYD > actYD, so wird offYD um lastYD erhöht (so werden auch mehrere Inverter Neustarts am Tag berücksichtigt). Als YieldDay wird offYD + actYD eingetragen, hier mal für `addValue`:
```
...
} else if (FLD_YD == rec->assign[pos].fieldId) {
    float actYD = (REC_TYP)(val) / (REC_TYP)(div) * generalConfig->yieldEffiency;
    if (lastYD > actYD)
        offYD += lastYD;
    lastYD = actYD;
    rec->record[pos] = offYD + actYD;
} else {
...
```
Bei einem gewünschten YieldDay Reset müssen offYD und lastYD natürlich auch auf 0 gesetzt werden.

Damit sollten die einzelnen channel der Inverter immer den korrekten YieldDay Wert haben und alle darauf basierenden Berechnungen ohne weitere Anpassungen richtig sein.

knickohr, 2023-09-19:
Ja, so habe ich das auch in Node-Red realisiert 😉

beegee3, 2023-09-19:
@knickohr: hast du wohl schon mal beschrieben und ich hab' die Idee da geklaut 😆

lumapu, 2023-09-19:
braucht man wirklich zwei Variablen - bin hier ein Sparfuchs?
Wohl nur bei mehreren Neustarts an Tag oder, sonst könnte man einfach pauschal `lastYd` + den aktuellen Messwert rechnen, unter der Bedingung, das `lastYd` größer ist.

Werde es wohl wie beschrieben umsetzen, finde ich gut 😀

beegee3, 2023-09-19:
> Wohl nur bei mehreren Neustarts

ja, genau! ... und ich wette, irgendjemanden passiert genau das 😁

LeSpocky, 2023-10-08:
Wir hatten das letztens auch schon mal im Discord diskutiert.  Ich find es nur nicht wieder.  Letztlich war mein Vorschlag aber ähnlich.  Man merkt sich einen Offset und den letzten Wert vom Inverter.  Der Offset wird um 0:00 Uhr auf 0 gesetzt.  Kommt ein neuer Wert vom Inverter, vergleicht man ihn mit dem letzten.  Ist der neue Wert kleiner als der letzte, dann addiert man den letzten Wert auf den Offset.  Der aktuelle Wert ist dann immer Offset plus Wert vom Inverter.  So müsste es passen, oder? 🤔

(Und ja, ich fände das Feature ganz nett, weil dieses "Inverter weg" bei dunklen Wolken auch tagsüber auftreten kann und nicht erst kurz vor Sonnenuntergang.  Hatte ich jetzt dieses Jahr aber auch nur eine Hand voll Male.)

stefan123t, 2023-10-08:
> Eine Korrektur seitens Ahoy ist wahrscheinlich sehr aufwändig (wegen der variablen "Reset" Möglichkeiten).

Ich denke das sollte doch relativ einfach möglich sein: man muß sich nur um 0:00 Uhr den YieldTotal des Inverters merken und kann dann den ganzen Tag über die Differenz vom `aktuellen_YieldTotal - _YieldTotal_0:00_Uhr` als YieldDay ausgeben.
Dumm natürlich nur wenn der ESP mal für kleine Entwickler neu booten muß :wink:

LeSpocky, 2023-10-09:
> Ich denke das sollte doch relativ einfach möglich sein: man muß sich nur um 0:00 Uhr den YieldTotal des Inverters merken und kann dann den ganzen Tag über die Differenz vom `aktuellen_YieldTotal - [...]

Oder wegen Watchdog neu startet. Kam bei mir häufiger vor bevor es "Reboot @ midnight" gab.

stefan123t, 2023-12-21:
Siehe https://github.com/tbnobody/OpenDTU/issues/1397 für das Feature / Problem in OpenDTU.

knickohr, 2024-06-02:
Zwar nicht resolutes, aber anscheinen programmtechnisch nicht so einfach umsetzbar. Werde trotzdem schließen da ich das Problem mit Node-Red „umschifft“ habe.

closed 2024-06-02 by knickohr