lumapu/ahoy #849: HM-1500 keine volle Panelleistung bei 600w

Suedfinne, 2023-04-03:
Hallo zusammen,
ist es normal das wenn man den HM-1500 auf 600w gedrosselt hat die einzelnen Panels nur noch ca. 155w bringen?
Ich habe 2 Panels Richtung Süd/Ost und 2 Richtung West damit ich über den ganzen Tag die volle Ausnutzung habe. Wenn der Wechselrichter mit voller Leistung läuft haben die Panels die Richtung Süd/Ost liegen morgens ca.300 Watt jeder, läuft der Wechselrichter auf 600w haben die Panels nur noch ca. 155w.
Das finde ich blöd weil könnte morgens meine 600w erreichen geht aber nicht weil er die Pandels drosselt, oder mache ich da was verkehrt?
Gruß Gunnar

## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

rejoe2, 2023-04-03:
Es ist "völlig normal", dass die Inverter die Limitierung intern aufteilen. 
Versuche mal unzustöpseln, also je ein Panel Süd/Ost und West je Inverterseite.

Suedfinne, 2023-04-03:
ok ich hatte gedacht das man gerade eine Himmelsrichtung auf ch1/2 und die andere ch 3/4 stöpselt. 
werde es versuchen, danke

gimehrzdbo, 2023-04-03:
Ich halte dies für einen Fehler! 
Der inverter scheint via ahoy nur relativ limitiert werden zu können. 
Limit 600 Watt sollen das bewirken was da steht.
Also normale Produktion, nur nicht über 600 Watt.  
Aus den 600 Watt werden allerdings 42 % gemacht.

rejoe2, 2023-04-03:
Was genau meinst du mit "dies"?

Woran auch Ahoy nichts ändern kann ist der Fakt, dass (bei den 4-Kanaligen) der Inverter intern den Zielwert aufteilt und jeder der beiden Teil-Inverter eben die Hälfte machen soll. 
Kommt von den Panels auf einer Seite mehr an als die Hälfte und auf der anderen weniger, ist das halt so. Darauf kann man nur extern reagieren (also über eine Automatisierungslösung, die ggf. dann den (zugelassenen) Gesamtwert erhöht usw.).

Falls ein Fehler bei der Umrechnung passieren sollte, wäre das ein anderes Thema.

GallusMax, 2023-04-03:
> Es ist "völlig normal", dass die Inverter die Limitierung intern aufteilen. Versuche mal unzustöpseln, also je ein Panel Süd/Ost und West je Inverterseite.

Wenn das "normal" ist, war es dennoch nicht zu erwarten. Ich bin gerade etws ratlos; Die Limitierung erteile ich _dem gesamten Inverter_ wieviele Module da nun produzieren, sollte erstmal egal sein.

Wenn eine Limitierung auf 600W (was 42% entspricht) dazu führt, daß einzeln aktive 300W-Module nicht mehr 300W bringen. läuft da imho was falsch.

rejoe2, 2023-04-03:
> Wenn eine Limitierung auf 600W (was 42% entspricht) dazu führt, daß einzeln aktive 300W-Module nicht mehr 300W bringen. läuft da imho was falsch.

Du kannst dich gerne beim Hersteller beschweren. Falls du eine firmware bekommst, die das anders macht, lass es uns wissen...

Ahoy kann jedenfalls mit einem einzelnen Befehl nicht eine Beschränkung umgehen, die auf dem Inverter selbst besteht. Ist halt so, gefällt mir auch nicht unbedingt - daher steht das "völlig normal" ja auch von vornherein in Anführungszeichen.

Ollipop030, 2023-04-03:
> Wenn eine Limitierung auf 600W (was 42% entspricht) dazu führt, daß einzeln aktive 300W-Module nicht mehr 300W bringen. läuft da imho was falsch.

Naja, der HM1500 (genau wie HM1200 HM1000 und HM800...) hat zwei unabhängige Tracker, auf denen die Leistung aufgeteilt wird. Bei 600W Limit bekommt jeder Tracker 300 Watt maximale Leistung. 200W + 400W oder 100W + 500W rechnet er nicht um, bei 300W ist Schluss, auch wenn der andere Tracker nur 50W produzieren sollte. So arbeiten die Wechselrichter nunmal.

GallusMax, 2023-04-04:
> Naja, der HM1500 (genau wie HM1200 HM1000 und HM800...) hat zwei unabhängige Tracker, auf denen die Leistung aufgeteilt wird. Bei 600W Limit bekommt jeder Tracker 300 Watt maximale Leistung.

Nur zum Verständnis: wie sind die Tracker auf die Ausgänge verteilt?
ch1+ch2 "links" auf einen und ch3 + ch4 "rechts" auf den anderen, oder doch anders?

rejoe2, 2023-04-04:
Davon war ich bisher ausgegangen, (meine eigene Verkabelung wegen ungleichmäßiger "Ausleuchtung" der Panels ist so organisiert), und auch die Antwort zur Verkabelungsfrage von Suedfinne impliziert das...

Ollipop030, 2023-04-04:
> Nur zum Verständnis: wie sind die Tracker auf die Ausgänge verteilt?

Die Bezeichnung der Eingänge hab ich grad nicht im Kopf, aber wenn du den WR vor dir liegen hast ist ein Tracker links, der andere rechts.

Suedfinne, 2023-04-04:
Ich kann es leider auf die schnelle nicht ausprobieren, da zwei Kabel dann zu kurz wären. Aber ich habe mir ein Skript geschrieben wo ich nur die P_AC auf 600 Watt einstelle. Funktioniert recht gut, müsste nur abwarten wenn Wolken vor die Sonne ziehen um zu sehen wie schnell er reagiert. Er regelt immer um 3% das Limit. Also P_AC > 610w dann um 3% das Limit runter und wenn P_AC < 570w dann 3% das Limit rauf. Nur wenn eine Wolke vor die Sonne zieht geht er ja sehr hoch mit dem Limit und wenn die Wolke weg ist habe ich viel zu viel P_AC. Aber da finde ich noch was. Nur ich verstehe nicht das die bei Hoymiles es nicht auch nach dem was hinten raus kommt regeln.
Gruß Gunnar

gimehrzdbo, 2023-04-07:
> Ahoy kann jedenfalls mit einem einzelnen Befehl nicht eine Beschränkung umgehen, die auf dem Inverter selbst besteht.

Trotzdem gibt es 2 Möglichkeiten einzustellen? Relativ und absolut. Im Ergebnis gibt es dann keinen Unterschied, richtig? 

Ich hatte den 1500 gewählt, um optional zusätzlich  zu meinen nach Süd ausgerichteten 2 Paneelen (ohne jegliche Beschattung) noch ein Paneel nach Osten zu ergänzen. 
Und für dieses Szenario auf Output 600 Watt gerüstet zu sein. Da lag ich wohl falsch. 
Wäre eine Einstellung für mich dann bei 1200 Watt (mit nur 2 Paneel) richtig? Sodass ich im Ergebnis 600 Watt aus einem "Kanal" erhalte?

rejoe2, 2023-04-07:
_Glaube_ auch, dass es im Ergebnis keinen Unterschied macht, auf dem Inverter scheint das immer in einen Absolutwert umgerechnet zu werden.

Bei einer Einstellung 1200 Watt bekommst du vermutlich mehr wie die 600W, weil erfahrungsgemäß auch "abgeschattete" Panels irgendwas liefern. Vermutlich liegst du mit irgendwas um die 800W "im Mittel" (halbwegs) richtig, wenn du auf der einen Seite eines der Süd-Panels mit dem Ost anschließt. 
Faktisch läßt sich das wohl nur über eine dynamische Anpassung anhand der konkret gelieferten Werte lösen...

gimehrzdbo, 2023-04-14:
Nun, jetzt also nochmal. 
Die zwei Module die ich angeschlossen habe sind auf einer Seite des wr eingesteckt, also nur in einem mppt. Diese hängen unmittelbar nebeneinander im gleichen Winkel und unbeschattet. 
Das Limit wurde absolut eingegeben wird aber wohl immer relativ (!) verrechnet. (siehe Screenshot, hier nach update auf die neueste freigegebene Version). 
Und was mich nun wundert, ist, dass die beiden Module, in Summe,  über 300 Watt dc sind. Das widerspricht den bisherigen geäusserten Theorien.

[image]
[image]

Ollipop030, 2023-04-14:
Aber du regelst doch die AC Seite. In deinem Screenshot 302,2 Watt, also alles richtig.

rejoe2, 2023-04-14:
> Das Limit wurde absolut eingegeben wird aber wohl immer relativ (!) verrechnet. (siehe Screenshot, hier nach update auf die neueste freigegebene Version).

Diese Vermutung scheint sich hartnäckig zu halten, nach Aussage von @baloo auf discord ist das aber ein längst behobener bug (oder eine Fehlinterpretation): 
https://discord.com/channels/984173303147155506/1029761098381017098/1096195904589009106

Bitte daher nicht einfach weiterverbreiten.

Suedfinne, 2023-04-14:
Antwort von Hoymiles auf mein Problem:

Sehr geehrter Herr ....,

ich danke Ihnen für Ihre Email. 

Der Wechselrichter begrenzt seine Ausgangsleitung auf 600W (AC-seitig). Wenn der WR auf 600W gedrosselt ist, dann muss die summe aller PV-Eingänge (PV1+PV2+PV3+PV4)) dicht an 600W sein. D.h, wenn die Summe von PV1+PV2= bei ca. 600W liegt, dann muss PV3 + PV4= ca. 0W, was hier nicht der Fall ist (alle 4 Module sind aktiv). 

Wenn der WR nicht gedrosselt ist, kann er folgendes erreichen:  PV1+PV2+PV3+PV4=300x4=1200W oder sogar 375X4=1500W (mit entsprechendes Modul).

Aus technischer Sicht ist das eine normale Leistungsregelung des WRs im gedrosselten Betrieb. Wenn Sie anstatt 4 nur 2 Module mit dem WR verbinden, dann wird er versuchen, die 600W (AC-seitig) aus den 2 Modulen zu erreichen. 

Ich hoffe Ihnen damit weitergeholfen zu haben.

 

Mit freundlichen Grüßen-Best Regards-Atentamente
Jose Sayago
Dr.-Ing. (TU)

Micro& Hybrid Inverter Technology
Technical Support Engineer
Hoymiles Power Electronics Inc.

sri-elgora, 2023-05-03:
Wozu überhaupt die Limitierung? Da der Wechselrichter bauartbedingt mehr als 600 Watt erreicht wird der Betrieb nicht legal, indem Du ihn softwareseitig limitierst.

GHolli, 2023-07-13:
> Der Wechselrichter begrenzt seine Ausgangsleitung auf 600W (AC-seitig). Wenn der WR auf 600W gedrosselt ist, dann muss die summe aller PV-Eingänge (PV1+PV2+PV3+PV4)) dicht an 600W sein.

Ich finde diese Erklärung verwirrend. Weil offenbar wird das Limit auf die MPPT aufgeteilt und nicht die maximale Ausgangsleistung begrenzt:
https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-1292102738

GHolli, 2023-07-13:
> 

> 

> Wozu überhaupt die Limitierung? Da der Wechselrichter bauartbedingt mehr als 600 Watt erreicht wird der Betrieb nicht legal, indem Du ihn softwareseitig limitierst.

> Wozu überhaupt die Limitierung? Da der Wechselrichter bauartbedingt mehr als 600 Watt erreicht wird der Betrieb nicht legal, indem Du ihn softwareseitig limitierst.

So pauschal würde ich das nicht formulieren. In Österreich ist zum Beispiel für eine derartige Kleinsterzeugungsanlage eine Engpassleistung von weniger als 800 W vorgeschrieben.

GHolli, 2023-07-13:
> Wozu überhaupt die Limitierung? Da der Wechselrichter bauartbedingt mehr als 600 Watt erreicht wird der Betrieb nicht legal, indem Du ihn softwareseitig limitierst.

So pauschal würde ich das nicht formulieren. In Österreich ist zum Beispiel für eine derartige Kleinsterzeugungsanlage eine Engpassleistung von weniger als 800 W vorgeschrieben.

DanielR92, 2023-08-25:
Anscheinend sind alle Fragen vom Tisch. Ich close diesen Beitrag vorzeitig. Handelt es sich um ein Versehen bitte einfach neu öffnen. 
Danke

closed 2023-08-25 by DanielR92