lumapu/ahoy #850: Max Module Power erscheint nicht in Einstellungen

MM7201, 2023-04-03:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.6.0
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ x] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Nach Eingabe der Wechselrichterseriennummer sollten sich laut einiger Videos im Netz eine Eingabemöglichkeit für die Modulleistung ergeben. Dies erscheint nicht. Stattdessen die Meldung " A max module power value of 0 disables the channel in live view".
Eine Verbindung zum Wechselrichter wird laut Log-Daten nicht aufgebaut. Er antwortet nicht. Was habe ich falsch gemacht?

sweinreuter, 2023-04-03:
Klingt wie https://github.com/lumapu/ahoy/issues/843 - Werte fehlen teilweise und sporadisch bei der 0.6.0 und die WR-Verbindung funktioniert nicht (was wohl durch Änderung der PIN-Belegungen zu beheben ist, habs aber noch nicht versucht).

sweinreuter, 2023-04-03:
Das hier ist wohl der ursprüngliche Report: https://github.com/lumapu/ahoy/issues/828 - ist also bekannt.

stefan123t, 2023-12-21:
@MM7201 wir sind bei aktuelleren Versionen 0.7.6x bzw. 0.8.2x, ist das Problem immer noch vorhanden ?

Ist das Problem auch auf dem ESP32 reproduzierbar oder nur auf dem ESP8266 der etwas wenig Speicher hat ?
Wieviele Wechselrichter und welche Modelle konfigurierst Du, die 4-chan Modelle brauchen mehr Speicherplatz im Heap / RAM.

closed 2024-01-18 by lumapu