lumapu/ahoy #851: [ESP8266] No reconnect after WLAN down (only after upgrade to 0.6.0)

elisenjens, 2023-04-03:
### Platform

ESP8266

### Model name

Don't know how to find this

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

Don't know...

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

Inverter 0  Communication Enable  Serial Number (12 digits)* 116183069356 Name* Hoymiles Max Module Power (Wp) 420 420 420 420 Module Name ObenLinks UntenLinks ObenRechts UntenRechts Yield Total Correction [kWh] 0 0 0 0 Delete Note  A 'max module power' value of '0' disables the channel in 'live' view  General  Interval [s] 60 Max retries per Payload 5 Reset values and YieldDay at midnight

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

#Device Name
AHOY-DTU
Dark Mode

System Config
Pinout

CS

D8 (GPIO15)
CE

D4 (GPIO2)
IRQ

D3 (GPIO0)
LED0

off / default
LED1

off / default
Radio (NRF24L01+)

Power Level

MIN
Serial Console

print inverter data

Serial Debug

Interval [s]
5
=============================
Connected to Fritzbox 6490

### Debug Serial Log output

```shell
na
```

### Error description

1) Go to http://ahoy-dtu/setup
2) configure above settings: standard
3) Reboot
4) Switch off WLAN while Ahoy is online and switch on WLAN again
5) In 0.5. after WLAN is coming up in the morning the Ahoy connected automatically; after upgrade to 0.6. Ahoy stays disconnected in the morning until it's powered down and up again.

elisenjens, 2023-04-03:
just tested 0.6.1: issue still existing

elisenjens, 2023-04-03:
Retested with a fresh and clean installation 0.6 from scratch (no upgrade) but still the no reconnect after WLAN comes back again

TobiScience, 2023-04-04:
I noticed the same problem, instead of reconnecting into wifi it stays in ap mode.

- reboot after changing settings -> stuck in ap mode
- had the local live view open for ~30minutes -> maybe crashed or something -> stuck in ap mode

never had these issue with the old version

elisenjens, 2023-04-04:
Today tested with 0.6.2: issue still exists

lumapu, 2023-04-04:
kann es hier nachvollziehen, das liegt an der letzen Umstellung der loop aus der Wifi-Klasse von PR #652.

<details>
<summary>Log (click to expand)</summary>

```
connect to network 'xxxxxx' ...
I: [WiFi] Connection Lost
scanning APs with SSID xxxxxx
reconnect in 3 seconds
reconnect in 2 seconds
reconnect in 1 seconds
scanning APs with SSID xxxxxx
reconnect in -1 seconds
reconnect in -2 seconds
reconnect in -3 seconds
scanning APs with SSID xxxxxx
reconnect in 0 seconds
reconnect in 9 seconds
scanning APs with SSID xxxxxx
reconnect in 0 seconds
reconnect in 9 seconds
scanning APs with SSID xxxxxx
reconnect in 0 seconds
reconnect in 9 seconds
scanning APs with SSID xxxxxx
reconnect in 0 seconds
reconnect in 9 seconds
scanning APs with SSID xxxxxx
reconnect in 0 seconds
reconnect in 9 seconds
scanning APs with SSID xxxxxx
reconnect in 0 seconds
reconnect in 9 seconds
scanning APs with SSID xxxxxx
reconnect in 0 seconds
reconnect in 9 seconds
```
</details>

elisenjens, 2023-04-05:
0,6,3 still reconnection issue

lumapu, 2023-04-05:
I tested several times with success @TobiScience can you test too?

elisenjens, 2023-04-05:
Retest is working fine now; please close ticket

elisenjens, 2023-04-05:
Hallo Lukas,

 

das ist ja Magie: jetzt funzt der Reconnect nachdem mein WLAN aus/an gegangen ist. Hab das Ticket geschlossen; danke!

 

VG, Jens

closed 2023-04-05 by elisenjens