lumapu/ahoy #853: [ESP8266] Prometheus Endpoint override

moritz31, 2023-04-04:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### Inverter
#### Inverter 0
    - Address: XXXXXX
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 0,0
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

I have setup ahoydtu with default values and using prometheus. The bug is introduced with the new default to only show the aggregation and not the default panel names. In the metric endpoints i have now 3 metrics

ahoy_solar_YieldTotal_kilowattHours{inverter="HM-600",channel=""} 0.000
ahoy_solar_YieldTotal_kilowattHours{inverter="HM-600",channel=""} 24.527
ahoy_solar_YieldTotal_kilowattHours{inverter="HM-600"} 24.527

Which is equivalent to channel 1, channel 2 and complete. Without the channel set, prometheus pics only the first metric and ignores the other 2 metrics

fsck-block, 2023-04-05:
Wenn die channel-id fehlt dann ist das  (für prometheus user) ein problem.
Die Frage ist, warum fehlt plötzlich die channel id bzw. wer putzt sie aus den internen Datenstrukturen raus.

@lumapu ich schau' mal rein

fsck-block, 2023-04-05:
Ich kann das Problem zwar nachstellen, aber nur dann wenn bei der Inverterkonfiguration keine Modulname eingetragen ist.
Wenn ich den Inverter mit Modul Namen eintrage 

[image]

erhalte ich auch die Metriken mit passenden Labels.
```
ahoy_solar_YieldTotal_kilowattHours{inverter="TestInverter",channel="Modul1"} 0.000
ahoy_solar_YieldTotal_kilowattHours{inverter="TestInverter",channel="Modul2"} 0.000
ahoy_solar_YieldTotal_kilowattHours{inverter="TestInverter"} 0.000
```

@moritz31 Kannst du bitte deine Ahoy Konfig prüfen oder exportieren und das json anhängen.

moritz31, 2023-04-06:
@fsck-block genau, wenn ich ein Modulname eintrage funktioniert es auch! Meine Annahme oder Erwartung wäre jedoch wenn ich nur den Inverter hinzufüge, dass ich dann auch eine Prometheus Metric bekomme. Also entweder wie in der UI nur das YieldGesamt anzeigen wenn label channel = "" metric gar nicht exposen oder den channeln default values geben.

fsck-block, 2023-04-06:
> Meine Annahme oder Erwartung wäre jedoch wenn ich nur den Inverter hinzufüge, dass ich dann auch eine Prometheus Metric bekomme. 

OK, jetzt verstehe ich. Bin gar nicht auf die Idee gekommen daß man nicht wissen will was die Panels liefern. :wink:

> Also entweder wie in der UI nur das YieldGesamt anzeigen wenn label channel = "" metric gar nicht exposen oder den channeln default values geben.

Ich würde mich an die Note im UI halten `A 'max module power' value of '0' disables the channel in 'live' view` und dann keine Metrik liefern wenn max module power nicht gesetzt ist. Wenn max module power gesetzt ist dann sollte aber auch ein Modulname gesetzt sein.

@lumapu : Ich würde die Änderung auf Basis `development03` machen und dir einen PR schicken. OK?

lumapu, 2023-04-06:
sehr gerne, danke für deine Mühen schon mal

closed 2023-07-23 by lumapu

badsmoke, 2023-09-06:
hey,

Fehler besteht in der version: "0.7.36" immer noch/ schon wieder, kann bitte mal jemand drüber gucken

fsck-block, 2023-09-07:
Hallo @badsmoke 

Leider kann ich deinen Hinweis (mit v.0.7.46) nicht nachvollziehen. Wenn ich bei mir die "Max Module Power" auf Null setze werden für dieses Modul keine Metriken generiert.

Könntest du mal deine Settings und eine Metricsdatei anhängen?

badsmoke, 2023-09-08:
Hey  @fsck-block 

ich bezieh mich auf den fehler das keine channel default werte generiert werden, wenn es keinen "Module Namen" gibt.
in der ausgabe ist das nicht Prometheus konform, es wird nur der erste wert genommen, es darf nicht zwei metricken mit den identischen namen geben.

außerdem ist mir noch aufgefallen das ich diese Konstellation auch noch nie gesehen habe.

 ````
 ahoy_solar_YieldTotal_kilowattHours{inverter="Hm-600",channel="Ulf"} 3.820
ahoy_solar_YieldTotal_kilowattHours{inverter="Hm-600",channel="Peter"} 3.737
ahoy_solar_YieldTotal_kilowattHours{inverter="Hm-600"} 7.557
 ````

wenn man im prometheus/grafana das hier eingeben würde

`ahoy_solar_YieldTotal_kilowattHours{inverter="Hm-600"}`

sollten zwei graphen im selben dashboard rauskommen einmal mit channel=ulf und einmal mit channel=peter.
 
 in diesem fall kommen aber 3 raus, "ohne channel parameter", channel=ulf und channel=peter
 
 
 der richtige weg währe entweder das zusammenrechner der beiden channel mit hilfe von 
 sum()
 `sum(ahoy_solar_YieldTotal_kilowattHours{inverter="Hm-600"})`
 
oder einen metric entpunkt zu erstellen der 

z.b. `ahoy_solar_YieldTotal_kilowattHours_total{inverter="Hm-600"}`

heißt

MfG

badsmoke

fsck-block, 2023-09-08:
Hallo @badsmoke 

Vorbemerkung: 
Der Prometheus-EP ist eher gewachsen als entwickelt. Ursprünglicher Gedanke war wohl dass alle Werte die es im System gibt, als Metrik bereitgestellt wurden. Ungeachtet irgendwelcher Spezifikationen. 

> ich bezieh mich auf den Fehler das keine channel default werte generiert werden, wenn es keinen "Module Namen" gibt.
Das Kennzeichen für ungenutzte Channels ist das Attribut "Max Module Power". Wenn man das Modul unterdrücken will, dann muss man "Max Module Power" auf 0 setzten. Ansonsten einen Namen vergeben. 
Aber vllt könnte man dann auch einen Default-Namen einsetzen wie zb. ch0, ch1, ...

> in der Ausgabe ist das nicht Prometheus konform, es wird nur der erste wert genommen, es darf aber nicht zwei Metricken mit 
> den gleichen Namen geben.
Korrekt.

> in diesem fall kommen aber 3 raus, "ohne channel parameter", channel=ulf und channel=peter
Korrekt.

> der richtige weg währe entweder das zusammenrechner der beiden channel mit hilfe von sum ...
Könntest du mir einen Link auf eine Spezifikation für "der richtige weg" geben. 
Ich kenne leider nur https://prometheus.io/docs/instrumenting/exposition_formats/ und dort habe ich dies so nicht rausgelesen.

> oder einen metric entpunkt zu erstellen der
> z.b. ahoy_solar_YieldTotal_kilowattHours_total{inverter="Hm-600"} heißt
Wie ist denn der Metric-Type "Summary" zu verstehen? Wäre das nicht genau hier anwendbar als `ahoy_solar_YieldTotal_kilowattHours_sum{inverter="Hm-600"}`?

badsmoke, 2023-09-08:
ich finde das super das ahoy metriken ausgibt, ich wünsche mir das für jedes Programm ;-)

ein default model name wie ch0,ch1 würde das problem schon lösen.

ich weiß auch nicht wo man die richtige spezifikation abrufen kann, der [node-exporter](https://gist.github.com/ipl31/888bf56ae33097fa178b0b643c2862be) ist aber ein sehr bekanntes beispiel mit sehr vielen werten, das ist die quasi Referenz 

die [summary metric](https://prometheus.io/docs/concepts/metric_types/) scheint spezieller zu sein, es wird in verbundung mit count und {quantile="<φ>"} verwendet

`_total` wäre schon ok

fsck-block, 2023-09-11:
@badsmoke 
Ich hab's mal codiert. Wenn es in den development branch übernommen wurde kannst du es ja mal ausprobieren.

lumapu, 2023-09-12:
wurde gestern übernommen 😉

fsck-block, 2023-09-15:
@badsmoke 
Ist im development branch drin. Wäre nett wenn du es auch bei dir testen könntest und Rückmeldung gibts.