lumapu/ahoy #857: Zero export solution implementation

dexterbot80, 2023-04-05:
Hello,
First of all, I want to congratulate you for your initiative - such a project is really necessary and must be perfected in the smallest details.
To get an idea of what I want to propose next, please watch this [VIDEO](https://drive.google.com/file/d/1Ys1RuMraD-KG359aM8GFS5ILdYvn3meH/view?usp=share_link).

The main idea would be to make this custom DTU to be able to read the data from a smart meter via modbus RTU and thus be able to control the output power of the microinverters according to the consumption of the house.
For example: If my house now consumes 500W, the DTU sends the command to the inverter to produce 500W.
If the consumption of the house is reduced to 200W, the DTU sends the command to the inverters to produce only 200W.
It would also be very useful if this DTU could communicate with single-phase and three-phase smart meters from several manufacturers.
I am attaching the modbus communication protocol for some common models.
[CARLO GAVAZZI](https://drive.google.com/drive/folders/1z8KK8Rwt7lmXtIR-w4jIyge5CnuyXWqh?usp=share_link)
[CHINT CHINA](https://drive.google.com/drive/folders/1Qv2kpiK0jWcGmBprH1Lkg91glC6n1Q34?usp=share_link)
[EASTRON CHINA](https://drive.google.com/drive/folders/1gfOEX4l05sCM7DDmP_L8i83U8oh0P6ro?usp=share_link)

rmayergfx, 2023-04-05:
First of all, this is not an issue, this is a wish from you.
AhoyDTU should be stay ligthweigth and working stable. 
You can do this already by using a 3d party software like HomeAssistant, ioBroker e.g. there are other solutions also on the market. An ESP8266 will never can handle this. 
Also this is not relevant to all users in all countries. If you connect to the grid, you need to meet the regulations from your country and there is no place for any wishes.

Do you actually know when an inverter is most effective? It is not a good  idea to have 4 panels with 1600 Wp with full sun power and reglement the inverter to 200W, this will cook your inverter. If you want a stable inverter with long service life put your not needed Energy into Battery or downgrade your system.

dexterbot80, 2023-04-05:
> First of all, this is not an issue, this is a wish from you. AhoyDTU should be stay ligthweigth and working stable. You can do this already by using a 3d party software like HomeAssistant, ioBroker [...]

Guys, please forgive me if I dared to ask more.
I was asking if it is possible because HOYMILES DTU PRO can do this. And more than that, any inverter built by a respectable brand can do this.
I posted a video created by HOYMILES that refers exactly to this bold request of mine.
I say let's be more optimistic and not limit ourselves only to the esp8266 and what this microcontroller can do.
There is also a much more powerful esp32 that could cope with these requirements.

[https://drive.google.com/file/d/1Ys1RuMraD-KG359aM8GFS5ILdYvn3meH/view?usp=share_link](https://drive.google.com/file/d/1Ys1RuMraD-KG359aM8GFS5ILdYvn3meH/view?usp=share_link)

rmayergfx, 2023-04-05:
What is the project goal? A cheap and stable solution without buying the DTU from the manufacturer.  a DTU-Pro from HM is about 250 bucks, and with a few effort from your side you can implement all this and much more in already existing open source home automation software.

Of course, suggestions and proposals are welcome, but these should then not be set as an issue and, moreover, consider beforehand what the goal of a project is.

Look at the size and power consumption of the DTU-Pro and compare that with this project. 

Why should the developer do that? There are already enough open construction sites with the already supported devices and if in addition still further devices from the home automation come in addition, how should all then function. Keep in mind that this project is managed by one person in his spare time and that for free.

Therefore, I find such requests a bit impertinent. 

Translated with www.DeepL.com/Translator (free version)

dexterbot80, 2023-04-05:
I posted here because I couldn't find another topic with proposals, It is an opensource project created by a quality man in his precious free time.
Just as he started this project and for the same reasons, he may still find time to develop more.
As for impertinence and other human "qualities", I say you should find a specialized forum to discuss it there.
Let's not forget that this project was developed precisely to replace DTU.
I believe that the developer of this project has the best answer to my request.

Regards

mager33, 2023-04-05:
Guys, stay calm! This was a normal feature request. The feature has been requested before. It is a logical wish. I do not understand why this would be impertinet.
Stay calm!
I suggest to close this issue.

reserve85, 2023-04-06:
@dexterbot80 : if your Smart Meter is able to publish the current power via webapi/JSON, then have a look at my repository: https://github.com/reserve85/HoymilesZeroExport

rmayergfx, 2023-04-06:
Attention you will need a NRF24L01+ otherwise you can´t get connected to your inverter!
The case for the omega rail mounted inside an electrical cabinet can lead in different issues. If the cabinet is made of steel, you need to place the NRF24L01+ Antenna outside the cabinet to get a stable wireless, amplifier will not help, as it does not amplify reception. 
also you need take care in cooling the system to prevent overheating. closed cases are not always a good idea.

dexterbot80, 2023-04-06:
> Attention you will need a NRF24L01+ otherwise you can´t get connected to your inverter! The case for the omega rail mounted inside an electrical cabinet can lead in different issues. If the cabinet [...]

Thank you for the recommendations,
I specified that I will use an ESP32 with thernet precisely to be able to mount the device in a steel cabinet.
I didn't specify that I'm going to mount the NRF24 antenna on the outside of the electrical panel because you kind of scolded me at the beginning and I think that if we've come this far, each of us knows what a Faraday cage is.
Regarding the overheating of the devices in the housing on the omega rail - I think that natural ventilation and a well-ventilated housing of 4 modules is sufficient. 

I'm glad though that now you don't see my idea as really bad anymore and we can have a civilized discussion here.
Ideally, the developer of this project would activate the discussions here, and then my proposal would not have reached the issues section.
[Google say that is verry easy ](https://www.google.com/search?q=how+to+activate+discussion+on+github&rlz=1C1KNTJ_enRO1005RO1005&sxsrf=APwXEde3JP3QFMbAczbG0ktgaAJ6TAPQWg%3A1680815860758&ei=9DYvZOH3LeaQ7_UP17yWuAc&oq=how+to+activate+discu+on+github&gs_lcp=Cgxnd3Mtd2l6LXNlcnAQARgBMgoIIRCgARDDBBAKMgoIIRCgARDDBBAKOgoIABBHENYEELADOggIIRCgARDDBEoECEEYAFDwGVi2J2CUSmgCcAF4AIABuAGIAbMGkgEDMC41mAEAoAEByAEIwAEB&sclient=gws-wiz-serp)
This would probably be useful too - I believe that the human species has evolved by sharing ideas:)

rmayergfx, 2023-04-08:
Please do not misunderstand. I still don't like the idea. There are several reasons for that. If the manufacturer sees that too many adaptations replicate the main functions of his paid DTU, the time will come that he will seal off his devices and it may then no longer be possible to work with this project. A firmware change can have fatal consequences for the community around the DTU projects. After all, the investors want to see money and if the in-house DTUs are no longer sold because there are cheap or free solutions, it will sooner or later come to that the devices are sealed off.
An ESP32 cannot provide these services at all without massive restrictions. It consumes depending upon application up to 170 mAh, a DTU PRO 2,5W to 5W, there one sees already that here a completely different more efficient hardware is installed, so that the whole can take place also in real time.

We don't even need to start with an ESP8366. But that is one of the advantages of this project, that these are also supported, otherwise you could have installed the sister project, which only supports the ESP32 variants.

If you want to have such a function, then buy the DTU from the manufacturer and all your wishes are fulfilled. 

Alternatively you work yourself into the programming and develop such an alternative then you will find out how time-consuming such a development can be. The costs for the original DTU are then a bargain.

mager33, 2023-04-08:
I totally disagree: good, feature-rich open source software increases the value of Hoymiles inverters. They will sell more inverters. There is no connection between the capabilities and the power consumption. Zero export is easy to implement: subscribe to the energy meter's MQTT for current power, adjust power production on each update. If the maintainer does not like it, a fork would help...

stefan123t, 2023-12-21:
See #781 and #782 for individual Feature Requests to implement this for multiple inverters and using a variety of Smart Meters as source of the current consumption of your metered installation / house.

lumapu, 2024-03-02:
We started development of a zero-export version of Ahoy, check the branch `zero-export` which is available now.