lumapu/ahoy #858: [Bug] Inverter PowerLimit not published to homeassistant

tammof, 2023-04-06:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT Msoquitto installed and working
MQTT generally working
homeassistant working

When using the "Discovery Config (homeassistant)" function, the mqtt topic "inverter/Invertername/ch0/active_PowerLimit" is not integrated into homeassistant. The corresponding topic "homeassistant/sensor/Invertername" is missing. Any workarounds?

### Error description

When using the "Discovery Config (homeassistant)" function, the mqtt topic "inverter/Invertername/ch0/active_PowerLimit" is not integrated into homeassistant. The corresponding topic "homeassistant/sensor/Invertername" is missing. Any workarounds?

tammof, 2023-04-06:
I Published this config to the corresponding topic but HA still shows "not Available".

homeassistant/sensor/Invertername/ch0_active_PowerLimit/config

{
  "name": "Invertername active_PowerLimit",
  "stat_t": "inverter/Invertername/ch0/active_powerlimit",
  "unit_of_meas": "var",
  "uniq_id": "112183823262_ch0_active_PowerLimit",
  "dev": {
    "name": "Invertername",
    "ids": "112183823262",
    "mdl": "Invertername",
    "cu": "http://192.168.178.110",
    "mf": "Hoymiles"
  },
  "exp_aft": 95
}

tammof, 2023-04-06:
I think it is a Bug in the 0.6.0 autodiscover feature for HA.

lumapu, 2023-04-06:
you can be right, I will have a look into that. I think the topic `inverter/Invertername/ch0/active_PowerLimit` is not included in autodiscover for now. If it's missing I will add it.

tammof, 2023-04-07:
> you can be right, I will have a look into that. I think the topic `inverter/Invertername/ch0/active_PowerLimit` is not included in autodiscover for now. If it's missing I will add it.

while you are at it, please include total values cumulated over all inverters in /api/record/live or in seperate endpoint.
Thanks again

mr-p666, 2023-04-25:
Maybe I'm wrong, but subscribing on:
inverter/Invertername/ch0/#
doesn't show PowerLimit nor active_PowerLimit with v0.67.
If it's already fixed with v0.69 sorry for bothering you. But I can't test it because of the webif issue.
Thanks a lot for your help.

mr-p666, 2023-05-27:
Hej @lumapu,
r there any news on this topic? Without this value it's very difficult to handle the power limit function from the Hoymiles WR.
Would be very happy about a little update!
Thx a lot! :relaxed:

FusselTV, 2023-06-21:
@lumapu 
Same problem for me I am using a zeroexport script and want to know how much energy is overproduced but the value is not autodiscovered in homeassistant
[image]
[image]