lumapu/ahoy #860: FW Update: Vorgehensweise für DAUs ?

flummy1978, 2023-04-06:
Holla,

Ich habe den Rest nicht ausgefüllt, weil es ja für alle Versionen Configs etc gleich ist: 

Kann mir jemand sagen, wie man am einfachsten OTA Updaten kann, nachdem die Software einmal läuft? 

Klar, jetzt werden einige sicherlich sagen: "Einfach über die Update Funktion" -.- 
Aber genau hier ist mein Problem: 
Ich habe zwei identische Setups - Beide auf einem Wemos D1 Mini. Auf beiden war 0.5.66 drauf. Nun wollte ich eben auf 0.6.0 updaten und habe somit die **_230327_ahoy_0.6.0_9c7d7a0_esp8266.bin_** ausgewählt - die dann wohl falsch war. Danach ging gar nichts mehr und ich musste manuell Firmware neu drauf machen (Was in der Tat sehr besch.... ist, wenn die Anschlüsse bereits fest verlötet sind :( ) 
Nun würde ich das gern auf meinem Live-System vermeiden wollen. Welche Datei sollte ich dafür auswählen? Oder gibt es einen Trick einen Vorabcheck zu machen, bevor man updatet etc? 

Vielen Dank im Voraus
VG
Andreas

mroenne2022, 2023-04-06:
Hallo @flummy1978 
Eigentlich hast du es genau richtig gemacht, wenn es wirklich ein Wemos D1 Mini esp8266 ist, dann hast du die richtige Datei ausgewählt.
Ich hab 3 Stck davon und hab alle 3 kürzlich erfolgreich und ohne Probleme auf dem Wege upgedatet.
Es muss offensichtlich irgendein anderes Problem vorliegen, oder du bist evtl. doch in der Zeile verrutscht auf eine andere bin !? 
[image]

jimknopf63, 2023-04-07:
Was heisst den ging gar nichts mehr? Erkläre das mal bitte genauer...

flummy1978, 2023-04-07:
@jimknopf63 Naja nach der Installation klappte kein Neustart, kein Zugriff per Wlan, kein Accesspoint nichts mehr.... Das Einzige was funktioniert hat, war eine Neuinstallation von Vers 0.6.0 

@mroenne2022  Jetzt isses für mich umso verwirrender, weil ich genau diese Datei auch genommen habe. Ich hab extra drauf geachtet die korrekte Datei auszuwählen, weil mir sowas schon mal passiert ist (seitdem auch nie wieder ;) ) 

Vielen Dank vorab für Eure Hilfe schon jetzt :)

CBRocker, 2023-04-07:
Hey,
ich habe bisher auch schon ein paar Mal Ahoy installiert. Aber ich habe es bisher immer per USB-Kabel gemacht.
Welche Version hattest du vor dem Update auf dem ESP? Hast du den ESP vor dem Update mal kurz stromlos gemacht, kann ja nicht schaden.
Das Update wurde erfolgreich installiert?

Ich habe im Moment noch als Produktivsystem 0.5.17 laufen, habe heute aber einen neuen ESP zusammengesteckt und gleich die Development Version 0.6.4 installiert.

rmayergfx, 2023-04-08:
Das Update per OTA funktioniert nur 100% gut wenn das WLAN hervorragende Verbindung hat, das kann ich hier mehrfach nachstellen, irgendwas in der Update Routine passt noch nicht so ganz. Zeigt z.B. das WLAN Symbol in der Gui nur 1 oder 2 Balken an, so kann ich per OTA kein Update einspielen, es dauert ewig und nichts passiert da anscheinend die *.bin nicht sauber übertragen wird. Das ist unabhängig ob ein ESP8266 oder ESP32 im Einsatz ist, der ESP32 ist beim Update nur etwas schneller. Alle anderen Änderungen in der GUI werden auch bei schwachem WLAN übertragen und gesetzt, speichern, reboot kein Problem.

flummy1978, 2023-04-08:
@CBRocker  Ich habe es OTA gemacht, wie ich sagte. Dort dann 0.5.66 drauf gehabt und versucht auf 0.6.0 upzudaten. 
Über USB ? - habe ich tatsächlich bis dato noch nie probiert. 

@rmayergfx In diesem Fall kann ich das zu nahezu 100 % ausschließen. Das entsprechende Teil war über den Access Point erreichbar (also nicht mit dem einem Wlan verbunden) der PC mit dem ich an dem Wemos gearbeitet habe, war ca 1 M Luftlinie entfernt. D.h. einen besseren Empfang könnte ich wahrscheinlich niemals haben.

stundenblume, 2023-04-09:
Moin zusammen,
ich habe das gleiche Verhalten.
Von 0.5.66 auf 6.0 mit `230327_ahoy_0.6.0_9c7d7a0_esp8266.bin`.
OS: Linux
Firefox hat nachdem auf `update` geglickt wurde gesagt dass kein Verbindungsaufbau möglich ist. Neu laden der Seite funktionierte und zeigte dass kein Update stattgefunden hat.
Bei Chromium habe ich die Meldung `success` erhalten. Seit dem erreiche ich den ESP nicht mehr.
ESP hängt 10cm neben einem Fritzbox Powerline AP.
Grüße,
Sebastian

mroenne2022, 2023-04-09:
@stundenblume : Wie ich oben schon schrieb , bei mir hab ich das Verhalten echt nicht gehabt. Hab 3 Stck. von 0.5.66 auf 0.6.0 über das Menü im Ahoy aktualisiert, alles lief einwandfrei. Alles über einen Laptop / Win11.  Ich hab dazu leider keine Idee.
VG Michael

momoluna, 2023-04-09:
Ich habe einen esp32 und würde auch gerne ein update machen,
was ist denn der unterschied
esp32_prometheus
esp32s3
esp32
?
Viele Grüße

stundenblume, 2023-04-10:
Es war jetzt kein Problem, der ESP ist nur gesteckt und ich glücklicherweise vor Ort (Installation der Schwiegereltern).

Evtl. hat der ESP nach dem Update aber auch eine neue IP bekommen und ich habe das nicht gemerkt. Zumindest nach dem erfolgreichen Update hatte er eine neue IP.

Ich werde das auch nochmal an einem Testsystem Zuhause probieren.

mroenne2022, 2023-04-10:
@stundenblume : Kleiner Tip - Im Router einstellen das immer die gleiche IP vergeben wird, dann gibt's solche Probleme oder Irrtümer nicht mehr, das klappt wunderbar 👍

Sinned42, 2023-04-11:
Ich habe den Verdacht, dass z. B. Pins nicht richtig übernommen werden und es dann beim Start zur exception kommt. Einfach mal den esp von allem abziehen (bis auf Strom natürlich ;) ), und schauen, ob dieser dann wieder erreichbar ist. Dann ggf. neu konfigurieren auf das eigene setup und wieder zusammen stecken.

stundenblume, 2023-04-11:
Ich bin mir recht sicher dass auch getestet zu gaben. Es kann aber, wie gesagt, auch daran gelegen haben dass ich die falsche IP angenommen habe.

tavoc, 2023-04-15:
Bei mir hatte beide esp8266 von der 0.5.66 auf die 0.60 eine andere IP bekommen. 

Auch per mqtt heißen sie nun anders. Hatte auch dir normale bin genommen. 
Am besten im Router nachschauen

mroenne2022, 2023-04-15:
Wie gesagt, im Router kann man trotz DHCP für jeden Client festlegen das er IMMER die gleiche IP bekommt.  Das hab ich für meine DTUn so gemacht, noch nie haben sie nach einem Update  eine andere IP bekommen.  Ich empfehle es also.

DonTermi, 2023-04-17:
Ich hatte auch nach dem Update das selbe Problem. Der ESP8266 kam nicht wieder. Bin extra 2 mal zum Gehege raus, Strom ab, Strom dran ... Aber kam nicht wieder. Daher habe ich das Konstrukt aus dem Dosenverteiler genommen und wollte es drin nochmal probieren. Als ich den an den Strom steckte funktionierte es auf einmal. Anscheinend war bei mir Lösung den ESP einfach mal etwas länger vom Strom zu nehmen. Jetzt läuft alles.

flummy1978, 2023-04-17:
Danke für die rege Teilnahme an meiner Fragestellung.... Scheinbar hat mein ESP einfach nur kurz pech gehabt beim denken 🤣
@tavoc Bei allem Respekt, aber wer mit so etwas hantiert, wie dem hier, sollte ganz sicher nicht an einer DHCP Adresse scheitern. Das würde ich schon voraus setzen, dass man nach sowas schaut.....

Ich werde heute oder morgen mein aktiv Gerät von 0.566 auf 0.6 ziehen, mal sehen was da passiert 😉

VG
Andreas

tavoc, 2023-04-17:
@flummy1978 
Ein IP Wechsel ist kein Problem, es kam nur unerwartet. 

Das andere ist der Wechsel des Mqtt Gerätenames zu dem devicename-wert

Aktuell heisst es so. AHOY_DTU_C5596B

Das sind alles keine Probleme, ich hatte es nur im changelog über sehen und bei allen anderen updates war es nicht so.

EXECdesign, 2023-04-23:
> Moin zusammen, ich habe das gleiche Verhalten. Von 0.5.66 auf 6.0 mit `230327_ahoy_0.6.0_9c7d7a0_esp8266.bin`. OS: Linux Firefox hat nachdem auf `update` geglickt wurde gesagt dass kein [...]

^^Exakt so war es bei mir auch: Von 0.5.66 auf 6.0, neu laden der Seite funktionierte und zeigte dass kein Update stattgefunden hat. Habe es dann einfach nicht nochmal versucht - und nach den Berichten des instabilen WLANs der 0.6.9 bin ich auf diese auch nicht wirklich scharf...

Was ich halt immer noch nicht verstehe -im Sinne der Benutzerfreundlichkeit- ist, warum es so schwer ist, im UI einen Link direkt zu den Updates zu platzieren. Es gibt einen Update-Reminder samt Changelog für Produktivversion und Download-Link samt Changelog für die Development Firmware. Warum dann keinen Direktlink zur Downloadliste der Produktivversion?

> Support this project:
>     [Changelog](https://github.com/lumapu/ahoy/blob/main/src/CHANGES.md)
>     [Download](https://nightly.link/lumapu/ahoy/workflows/compile_development/development03/ahoydtu_dev.zip) & Test development firmware, [Changelog](https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md)

Allein deshalb hatte ich schon auf diesen Thread geklickt "FW Update: Vorgehensweise für DAUs?"
Bin zwar nicht ganz DAU, möchte mich aber dennoch nicht immer zur korrekten FW-Datei durchklicken müssen (unnötig).

Wichtigster bisheriger DAU-Hinweis hier: "Diesem Netzwerkgerät immer die gleiche IPv4-Adresse zuweisen." bei FRITZ!BOX & besser. Mache ich bei solchen Geräten standardmäßig - hilft.

lumapu, 2023-04-25:
> Das einzige was ich mir wünschen würde, wäre das eine Prüfung stattfindet, damit nicht jemand versehentlich eine ESP32 auf einen ESP8266 einspielen kann oder umgekehrt, d.h. das Update quitiert mit einer Fehlermeldung "Falsche Version ausgewählt"

Das könnte man tatsächlich Recht einfach einbauen 😊