lumapu/ahoy #862: Activate /json with config_override.h

tammof, 2023-04-07:
Hy, 
the manual states that there is an option to enable JSON Live data 
http://192.168.178.x/json 

by activating it in the config_override.h.

`<html>
<body>

/json | gets live-data in JSON format | json output from the livedata | no - enable via config_override.h
-- | -- | -- | --
/metrics | gets live-data for prometheus | prometheus metrics from the livedata | no - enable via config_override.h

</body>
</html>`

Any hint how to do this? The examples in /ahoydtu/src/config are not mentioning this.
Is the data from the "total" section included? Or is this replaced by the rest api in general?

Cheers!

fsck-block, 2023-04-07:
The /json endpoint was removed because this functionality is covered by the REST api with much more details available.

It seems that the documentation is missing that fact.

tammof, 2023-04-07:
Is the data from the "total" section included? I couldnt find it.

fsck-block, 2023-04-07:
Have a look at `/api/record/live`. For my configuration i will find
[image]
Is it that what you are looking for?

tammof, 2023-04-07:
Is this from the Total section of the live page? I have 6 inverters configured and would Like to request the Total values over all inverters.

fsck-block, 2023-04-07:
> Is this from the Total section of the live page? 

I only have one inverter configured, but as far as i can see the REST api provides data for each inverter independently.
The old /json endpoint also reports only inverter by inverter without summarizing the inverter totals.

tammof, 2023-04-07:
Thanks again. I'll figure something out.

lumapu, 2023-04-13:
can you do the sum on the client side? Ahoy does the calculation with Javascript on the client side by looping over all inverters

closed 2023-07-23 by lumapu