lumapu/ahoy #863: Ahoy DTU seems to crash (reboot) if i connect to webif from mobile device

x23piracy, 2023-04-07:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

### Power Stabilization:
* [X]  220uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [X] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hi,

i just build my own Ahoy DTU and everything went smoothly, i build it before i own my own Hoymiles inverter :P
While playing around with the webif i could see that i can bring it easily to a reboot with just connecting to its webif with a mobile browser (iphone with safari or firefox).

When doing this on firefox on my computer this does not happen.
There is no inverter configured currently.

I've read about better stability with esp32 but i would expect that ahoy with wemos d1 mini should at least alive a night without doing nothing :) any idea? 

Best Regards X23

tastendruecker123, 2023-04-07:
This should be fixed in the newer dev versions starting with 0.6.3.

x23piracy, 2023-04-07:
Ok but what is it? I have not switched over to the dev version which i will do.

x23piracy, 2023-04-08:
0.6.4 seems to fix it: https://github.com/lumapu/ahoy/issues/828

closed 2023-04-09 by lumapu