lumapu/ahoy #864: 0.6.0: /api/live doesn't contain inverter data

christophhhh, 2023-04-07:
## Hardware
  * [ ] ESP8266
  * [ X] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

### Power Stabilization:
* [ X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.0
Github Hash: 9c7d7a0

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools
* [X ] Precompiled/OTA

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hi all!
I'm running Ahoy DTU 0.6.0 and the data reception works fine. I would like to use [solaranzeige](http://www.solaranzeige.de) to display the inverter production over time.

That project relies on the REST Api of Ahoy DTU (api/live). Unfortunately there seems to be a breaking change between 0.5.66 and 0.6.0. 

As api/live doesn't contain any live data from the inverter, I wonder if that is intended or a bug or my installation is broken.

my /api/live looks like this (0.6.0):
`{"generic":{"wifi_rssi":-53,"ts_uptime":9163,"menu_prot":false,"menu_mask":61,"menu_protEn":false,"esp_type":"ESP32"},"refresh":10,"ch0_fld_units":["V","A","W","Hz","","°C","kWh","Wh","W","%","var"],"ch0_fld_names":["U_AC","I_AC","P_AC","F_AC","PF_AC","Temp","YieldTotal","YieldDay","P_DC","Efficiency","Q_AC"],"fld_units":["V","A","W","Wh","kWh","%"],"fld_names":["U_DC","I_DC","P_DC","YieldDay","YieldTotal","Irradiation"],"iv":[true,false,false,false,false,false,false,false,false,false]}`

tammof, 2023-04-07:
seems to have moved to http://192.168.178.110/api/record/live
but only each inverter on its own, no cumulated values available.

beegee3, 2023-04-08:
there was a change due to problems with JSON
now you will find the values in /api/inverter/id/0
change the id number as needed: 0 = first inverter, 1 = second inverter, ...

closed 2023-04-19 by lumapu