lumapu/ahoy #865: Switch to change information on display

x23piracy, 2023-04-07:
Hi,

is i possible to use a switch to change informations on the display on demand maybe in combination of some customization via webif?
A better readable font would also be nice if possible.

Best Regards X23

DB1BMN, 2023-04-29:
+1

Hi,

I would appreciate to add some buttons / switches to navigate through several display pages (eg. one seperate page per Inverter or panel etc. - printing some diagrams would be *WOW!*).

To not wase any "precious" ports I would suggest using the ADC 'A0' port.
By connecting a binary stepped voltage divier (eg. 1k-2k-4k-8k...) you could <s>easily</s> distinguish which button is pressed and even if more buttons at the same time are pressed...

Of course this has to be debounced properly by a window comparator and state machine...

I've prepared a PCB having connected two pushbuttons on 'A0'. See https://github.com/lumapu/ahoy/issues/911

So I could easily test some firmware having implemented this.

stefan123t, 2024-10-27:
Noch ein Feature Request für einen Button: #865 #751 und #541.