lumapu/ahoy #866: [Bug]ESP8266 0.6.4 Exception beim Limit setzen

sude22, 2023-04-08:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.4

### Github Hash

d114fd67

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

4xHM-1500

### Debug Serial Log output

```shell
0x4025718b in ieee80211_setup_ratetable at ??:?
0x40257120 in ieee80211_setup_ratetable at ??:?
0x402570c7 in ieee80211_phy_init at ??:?
0x4025621c in cnx_update_bss_more at ??:?
0x40256245 in cnx_update_bss_more at ??:?
0x40253ca8 in scan_parse_beacon at ??:?
0x40100590 in interrupt_handler at ??:?
0x401002cc in ets_post at ??:?
0x40250005 in ieee80211_parse_beacon at ??:?
0x401002cc in ets_post at ??:?
0x402549a3 in sta_input at ??:?
0x40254412 in sta_input at ??:?
0x40267edb in pp_tx_idle_timeout at ??:?
0x4026779f in ppPeocessRxPktHdr at ??:?
0x4023234a in loop_task(ETSEventTag*) at core_esp8266_main.cpp:?
0x401056ed in call_user_start_local at ??:?
0x401056f3 in call_user_start_local at ??:?
0x4010000d in call_user_start at ??:?
0x401000ab in app_entry_redefinable at ??:?
0x4025c6fc in cont_ret at cont.S.o:?
0x4025c6ad in cont_continue at cont.S.o:?
0x401002cc in ets_post at ??:?
0x40105e85 in ets_timer_disarm at ??:?
0x401002cc in ets_post at ??:?
0x40102354 in pp_post at ??:?
0x401055eb in lmacRxDone at ??:?
0x40102edf in rcReachRetryLimit at ??:?
0x401030bc in rcReachRetryLimit at ??:?
0x401002cc in ets_post at ??:?
0x40103576 in wDev_ProcessFiq at ??:?
0x401002cc in ets_post at ??:?
0x401044e3 in lmacProcessTXStartData at ??:?
0x401044e0 in lmacProcessTXStartData at ??:?
0x40103447 in wDev_ProcessFiq at ??:?
0x40208000 in u8g2_UpdateRefHeight at ??:?
0x40208000 in u8g2_UpdateRefHeight at ??:?
0x40103298 in wDev_ProcessFiq at ??:?
0x401002cc in ets_post at ??:?
0x4023c0db in _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c:519
0x40105e85 in ets_timer_disarm at ??:?
0x4023b99f in quorem at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c:74 (discriminator 1)
0x4023b971 in quorem at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c:63
0x401002cc in ets_post at ??:?
0x40102354 in pp_post at ??:?
0x401055eb in lmacRxDone at ??:?
0x40102edf in rcReachRetryLimit at ??:?
0x401030bc in rcReachRetryLimit at ??:?
0x4023dc66 in __d2b at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c:779
0x4023d42d in _Balloc at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c:128
0x4023dc66 in __d2b at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c:779
0x4023c759 in _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c:854
0x401043f1 in lmacIsIdle at ??:?
0x4023c759 in _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c:854
0x401002cc in ets_post at ??:?
0x401002cc in ets_post at ??:?
0x40102354 in pp_post at ??:?
0x401055eb in lmacRxDone at ??:?
0x40238acd in __cvt at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_float.c:102
0x401030bc in rcReachRetryLimit at ??:?
0x40103576 in wDev_ProcessFiq at ??:?
0x4023de78 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:179
0x40103298 in wDev_ProcessFiq at ??:?
0x4022a3f9 in MqttClient::_sendPacket() at ??:?
0x4022b0fe in MqttClient::_checkOutbox() at ??:?
0x4022a4ac in MqttClient::publish(char const*, unsigned char, bool, unsigned char const*, unsigned int) at ??:?     
0x4020c0bf in PubMqtt<HmSystem<(unsigned char)10, Inverter<float> > >::publish(char const*, char const*, bool, bool) at ??:?
0x4022a4ac in MqttClient::publish(char const*, unsigned char, bool, unsigned char const*, unsigned int) at ??:?     
0x40239e55 in snprintf at /workdir/repo/newlib/newlib/libc/stdio/snprintf.c:87 (discriminator 4)
0x4022a4f6 in MqttClient::publish(char const*, unsigned char, bool, unsigned char const*, unsigned int) at ??:?     
0x4022af92 in MqttClient::_checkIncoming() at ??:?
0x4022a7bb in MqttClient::_checkPing() at ??:?
0x4022b224 in MqttClient::loop() at ??:?
0x40232466 in yield at ??:?
0x4020c0c2 in PubMqtt<HmSystem<(unsigned char)10, Inverter<float> > >::publish(char const*, char const*, bool, bool) at ??:?
0x4020fdf7 in PubMqtt<HmSystem<(unsigned char)10, Inverter<float> > >::sendData(Inverter<float>*, unsigned char) at 
??:?
0x40232466 in yield at ??:?
0x40211b9d in PubMqtt<HmSystem<(unsigned char)10, Inverter<float> > >::sendIvData() at ??:?
0x40211d56 in PubMqtt<HmSystem<(unsigned char)10, Inverter<float> > >::tickerSecond() at ??:?
0x40232466 in yield at ??:?
0x4020b7d3 in ah::Scheduler::loop() at ??:?
0x40220000 in ahoywifi::getAvailNetworks(ArduinoJson::V6211PB::JsonObject) at ??:?
0x40232466 in yield at ??:?
0x4021d063 in app::loopStandard() at ??:?
0x40208974 in std::_Function_handler<void (), std::_Bind<void (PubSerial<HmSystem<(unsigned char)10, Inverter<float> > >::*(PubSerial<HmSystem<(unsigned char)10, Inverter<float> > >*))()> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?

last failed alloc call: 4022A4F6(48)
```

### Error description

Die DevVersion 0.6.4 [ESP8266] läuft sehr stabil.

Das setzen der Limits per MQTT führt aber leider zu dem gezeigten Log und einem Reboot.
Vereinzelt klappt es auch und der ESP läuft normal weiter.

Sollte ich für das Setup mit vier HM-1500 lieber auf einen ESP32 wechseln?

DukeWolle, 2023-04-08:
@sude22 bei 4x würde ich definitiv auf einen esp32 wechseln.

sude22, 2023-04-08:
Ich wollte halt erstmal, bevor ich alles wieder umbaue, fragen ob es bei anderen (evtl. auch mit ESP32) ebenso auftritt.
Ich kann den CallStack nicht so richtig deuten, aber scheinbar passiert ja ein "failed alloc call: 4022A4F6(48)" beim 
`0x4022a4f6 in MqttClient::publish(char const*, unsigned char, bool, unsigned char const*, unsigned int) at ??:?`
ein paar Zeilen vorher kommt auch schon ein publish...
Also entweder kein Speicher verfügbar oder .... ?

sude22, 2023-04-08:
Habe jetzt auf ESP32 umgebaut. 
Keine Fehler mehr beim Setzen der Limits, egal wie schnell nacheinander.
Auch deutlich stabileres WebUI.
Danke @DukeWolle  für den Hinweis.
Ich werde die Konfiguration jetzt so belassen und der Nulleinspeisung steht nichts mehr im Weg :)

closed 2023-04-19 by lumapu