lumapu/ahoy #867: nur Daten vom Wechselrichter, wenn PC-Verbindung "WebInstaller" geöffnet ist

allesclip, 2023-04-08:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

{"wifi":{"ssid":"","pwd":"","dev":"AHOY-DTU","adm":"","prot_mask":61,"dark":false,"ip":"","mask":"","dns1":"","dns2":"","gtwy":""},"nrf":{"intvl":10,"maxRetry":5,"cs":15,"ce":2,"irq":0,"pwr":3},"ntp":{"addr":"pool.ntp.org","port":123},"sun":{"lat":0,"lon":0,"dis":false,"offs":0},"serial":{"intvl":5,"show":false,"debug":false},"mqtt":{"broker":"192.168.1.108","port":1883,"user":"XXXX","pwd":"","topic":"inverter","intvl":0},"led":{"0":255,"1":255},"plugin":{"disp":{"type":3,"pwrSafe":true,"logo":true,"pxShift":true,"rot180":true,"wake":0,"sleep":0,"contrast":60,"pin0":4,"pin1":5}},"inst":{"en":false,"rstMidNight":false,"rstNotAvail":false,"rstComStop":false,"iv":[{"en":true,"name":"HM-600","sn":1897309046XXXX,"yield":[0,0,0,0],"pwr":[410,410,0,0],"chName":["rechts","links","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]}]}}

### Error description

Hallo,

ich habe das Problem, dass ich keine Daten vom Wechselrichter empfange. Das komische ist, dass wenn ich dem Wemos D1 per WebInstaller verbinde, sehr wohl Daten vom Wechselrichter erhalte. Trenne ich die Verbindung, zwischen PC und WebInstaller und dem Wemos, dann empfange ich auch keine Daten mehr vom Wechselrichter.

Hat jemand hierzu eine Erklärung? Ich habe schon diverse Versionen getestet. Alle haben das gleiche Problem.

VG

tastendruecker123, 2023-04-08:
Klingt eher nach einer Hardware-Sache als nach einem Ahoy Bug. Was passiert wenn man den ESP mit einer Powerbank mit Strom versorgt?

yogi2563, 2023-04-08:
In deinem Setup sind nach dem ersten Inverter noch viele weitere Inverter ohne Daten.
Bei mir sieht der letzte Eintrag nach dem ersten Inverter so aus:
... "chName":["GH vorne","GH hinten","",""]}]}}

In deinem Setup so:
"chName":["rechts","links","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]},{"en":false,"name":"","sn":0,"yield":[0,0,0,0],"pwr":[0,0,0,0],"chName":["","","",""]}]}}

Vielleicht werden durch die vielen, nicht vorhandenen Inverter die Daten überschrieben? Keine Ahnung ob das zu dem Fehler führt.

allesclip, 2023-04-08:
Hallo, 
ich habe nun sowohl den Wemos direkt an einem Netzteil angeschlossen, als auch die Konfig bereinigt. Das Phänomen ist aber weiterhin vorhanden. Die Verbindung zum Wechselrichter funktioniert nur, wenn der Wemos über WebInstall (COM-Port ausgewählt) verbunden ist. 

VG

tastendruecker123, 2023-04-08:
Wie verhält er sich an einer Powerbank?

allesclip, 2023-04-08:
Powerbank habe ich keine da, aber sollte sich das anders als direkt am Netzteil verhalten?

tastendruecker123, 2023-04-08:
Netzteile sind ungeerdet aber kapazitiv an das Stromnetz gekoppelt, Powerbanks haben keine Verbindung zum Netz. Es kann Unterschiede in Bezug auf Funkstörungen geben. PC USB Ports sind geerdet. Wenn er an einer Powerbank nicht funktioniert, würde das eher auf einen Hardware-Defekt hindeuten.

lumapu, 2023-04-08:
wir hatten schon mal so einen ähnlichen Issue, muss den mal rauskramen - wenn ich ihn wieder finde ...

tastendruecker123, 2023-04-08:
> wir hatten schon mal so einen ähnlichen Issue, muss den mal rauskramen - wenn ich ihn wieder finde ...

An[ den hier](https://discord.com/channels/984173303147155506/992030956313657384/1013785193737367673) aus dem Discord kann ich mich erinnern.

DerHerrW, 2023-04-12:
Klingt, als ob eine Kommunikationsleitung nur dann toggeln kann, wenn der PC angeschlossen ist. Wechsle mal die Verbindungen für CS, CE und IRQ. Bei meinem Wemos D1 mini Pro musste ich IRQ von D3 verlagern (habe D1 genommen). Natürlich nicht vergessen, unter den Settings den geänderten Pin einzustellen.

closed 2023-07-23 by lumapu