lumapu/ahoy #868: WLAN nach Unterbrechung bis zum Neustart weg

z7ths, 2023-04-08:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- 1 Hoymiles Inverter set
- WLAN with IP and Subnet set (no DNS or Gateway)
- MQTT Broker set
- NTP set
- LAT and LONG set

### Error description

Ich verwende ein TP-LINK WLAN AP der nun mal nicht mehr als eine Woche stabil läuft. 
Daher habe ich ein täglichen Neustart vom WLAN AP konfiguriert. 

Leider ist ab dem Zeitpunk der Ahoy DTU weg bzw. nicht mehr erreichbar. 
Die Erwartung ist dass der WLAN AP mehrere Minuten braucht und der Ahoy DTU daher wieder ein Fallback auf seinen eigenen AP Mode macht? Allerdings erscheint nicht als eigener AP soweit ich gesehen habe.
Erst nachdem die Stromversorgung entfernt und wieder verbunden wird, verbindet sich der Ahoy DTU wieder mit dem WLAN.

Könnte die WLAN Verbindung nach einer Trennung (i.e. AP Neustart) in automatisch wieder aufgebaut werden? 
Der AP Braucht ca. 2 bis 3 Minuten bis er wieder erreichbar ist und es sollte daher mehrere Verbindungsversuche geben.
Ggf. eine Einstellung z.B. Wenn 30 Minuten lang der WLAN AP nicht erreichbar ist dann fallback auf eigenen AP?

z7ths, 2023-04-08:
[image]

z7ths, 2023-04-08:
Habe jetzt mit Discord Hilfe die GIT SHA: d114fd6 :: 0.6.4 installiert und werde es mal damit versuchen.

rmayergfx, 2023-04-08:
Tipp:
Im Router für AhoyDTU den Haken setzen damit immer die gleiche IP Adresse genutzt wird. TP-Link nach Möglichkeit so konfigurieren, das er nur im 2,4 Ghz Bereich arbeitet.

lumapu, 2023-04-08:
die aktuelle dev `0.6.4`, wie es @z7ths geschrieben hat löst das Problem nach meiner Ansicht.

lumapu, 2023-04-08:
duplicate #851

z7ths, 2023-04-09:
Vielen Dank! Mit 0.6.4 funktioniert es perfekt!

closed 2023-04-09 by z7ths