lumapu/ahoy #869: Kein zugriff auf Ahoy DTU seit Update auf 0.5.66

schulle61, 2023-04-08:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo, 
Ich hatte gestern Abend auf mein HM600 das Update 0.6.0 gemacht, wobei sich dann die Ahoy DTU nicht mit dem WR verbunden hatte. Das Problem an was es liegt habe ich hier schon gelesen. 
Dann habe ich die Version 0.5.66  draufgespielt und die Ahoy DTU Webseite hat wieder alles Live angezeigt und so. Nur bei der DTU selbst hat sich das Display nicht mehr aktualisiert. Daraufhin hatte ich die DTU kurz vom Netz getrennt und wieder angeschlossen. Und seitdem geht nix mehr. Komme auch nicht mehr auf die Seite des DTU. Der WR funktioniert einwandfrei.
Ist das Gerät jetzt defekt oder an was könnte es liegen?

yogi2563, 2023-04-08:
Hast Du daran gedacht? Bei mir war das beim ESP32 auch so. Die Kabel für den RFI müssen geändert werden.

Changelog v0.6.0

Warning: The default pinout for ESP32 has changed. The SPI Pins MOSI, MISO and SCLK are now configurable. If you do an update from an older version make sure that these pins do not have 0 as their value.
This definition is our default for ESP32: MISO = 19, MOSI = 23 and SCLK = 18

rmayergfx, 2023-04-08:
Nicht die Kabel müssen geändert werden sondern die Settings in der System Config, siehe: https://github.com/lumapu/ahoy/issues/812

lumapu, 2023-04-08:
ich hoffe das Problem ist die Pin-Config, wenn nicht @schulle61 wieder öffnen.

closed 2023-04-08 by lumapu

schulle61, 2023-04-09:
Den Zugriff auf der Ahoi DTU Webseite bekomme ich nach erneuten aufspielen der Aktuellen FW. Und die PIN Belegung ist auch geändert.  Aber mein Display zeigt inmernoch nix an. Und wenn ich eine ältere FW von hier drauf spiele, dann habe ich wieder kein Zugriff auf der Ahoy DTU.  Und muss wieder von der Webseite aus die aktuelle Version draufspielen das ich wieder Zugriff habe