lumapu/ahoy #870: Fehlerhaftes aufritzend in der Doku

knickohr, 2023-04-08:
Das Fritzing für den NodeMCU ist fehlerhaft, es fehlt eine Verbindung und ist falsch verdrahtet. Bitte entfernen oder austauschen.

des Weiteren bitte auch nur noch Verdrahtungen mit Elko publizieren.

Außerdem wäre es sinnvoll auch Fritzings und Schaltpläne mit Display anzubieten.

[image]
[image]
![8824AE3A-C21D-4012-BC78-833A9B32DBD7](https://user-
[image]
[image]
images.githubusercontent.com/63048210/230733136-6e703fee-b2ce-4675-b767-6d175acca963.jpeg)
[image]

Fritzings mit ohne Display kann ich nachliefern wenn gewünscht.

mager33, 2023-04-08:
Nett. Kann man irgendwas machen, dass man bei den ESP32 auch die Pinbeschriftung lesen kann?

Joh49, 2023-04-09:
die gewünschte Schaltung in Word oder eine eigene Datei kopieren dann kann man zoomen

rmayergfx, 2023-04-09:
Ähm wo genau war da jetzt der Fehler im Fritzing?
Meiner Meinung nach sind da zu viele Elkos in der Schaltung. Statt hier 3x 100 uF zu verbauen genägt hier ein etwas größerer, da es sich um eine parallel Schaltung der Kondensatoren handelt. 5V und GND, somit ergibt sich Cges=C1+C2+C3

mager33, 2023-04-09:
Kein Fehler, nur Pin-Beschriftung des ESP32 eben kaum lesbar. Sehe keinen Sinn in mehr als 100 uF

knickohr, 2023-04-09:
Na, wenn das kein Fehler ist, bzw. Sogar 2 Fehler. Es fehlt ein Anschluß am NRF-Modul und die Verbindung zum D2 ist falsch, zumindest paßt sie nicht zum Schaltplan.

[image]

Und bitte auch mal genau schauen wo „meine“ Kondensatoren angeschlossen sind. Über den 3. am Display läßt sich noch streiten, aber die andern beiden sind NICHT parallel geschaltet !

rmayergfx, 2023-04-09:
Genau lesen was ich geschrieben habe bzgl. der Kondensatoren!  
Wo hast du diese Grafik gefunden, bitte mal die Quelle angeben!

knickohr, 2023-04-09:
In der offiziellen Doku hier auf Git :

https://github.com/lumapu/ahoy/blob/main/Getting_Started.md#symbolic-view-1

rmayergfx, 2023-04-09:
Ah, du meinst Pin8(IRQ) am NRF24L01+

Loetnase, 2023-04-09:
Vielleicht auch die Farben in Schematic und Symbolic View für die Verbindungen gleich halten.
Z.B. gelbe Verbindung usw. die passen nicht zusammen.
[image]

knickohr, 2023-06-11:
Das olle fehlerhafte Fritzing ist ja immer noch drin 😲

closed 2024-02-17 by lumapu