lumapu/ahoy #871: Aktuelle Leistung in Watt per API auslesen

SteveBaeumer, 2023-04-08:
Hallo,

da ich neu hier bin Hofe ich die richtige Seite erwischt zu haben für meine Frage.

ich würde gern die aktuelle Leistung AC die der WR erzeugt aus der DTU auslesen um sie in meinem Smart home zu visualisieren.
Über REST API sehe ich unter http://192.168.178.38/api/record/live die Daten die ich benötige. Aktuell: "fld":"P_AC","unit":"W","val":"31.90"
Wie kann ich genau diese Daten auslesen.
Über get-Input kann ich die Adresse eintragen, aber welcher Befehl liefert mir die in diesem Beispiel angezeigten 31.9 Watt?
Habe leider nicht wirklich viel Erfahrung in dem ganzen Thema und hoffe daher auf Hilfe.
Hoffe ich konnte mich einigermaßen verständlich ausdrücken.

lieben Dank,
Steve

## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

mager33, 2023-04-08:
Das hängt ganz von Deinem Smarthome-System ab, da bist Du hier falsch. Bei HomeAssistant geht es über MQTT inkl. Autodiscov ery. MQTT und REST sollte überall gehen.

SteveBaeumer, 2023-04-08:
Danke,
genau da habe ich ja mein Problem. 
Wie lautet denn die entsprechende Abfrage über REST API.
Gibt es nicht einen Befehl der mit genau die Daten aus diesem Feld ausgibt? "fld":"P_AC","unit":"W","val":"31.90"

Also sowas wie {"fld":"P_AC","unit":"W","val":"31.90"} ? Kenne mich halt mit diesen JSON Befehlen nicht aus.

dknubben, 2023-04-08:
Also ich gebe auch per REST die aktuellen Daten an einen 2. ESP32 mit LCD-Display weiter und maches das über ein Skript (auf dem MQTT-"Server"), das mir die Werte filtert. Die API kann das nicht direkt leisten. DIe URL lautet aber (seit 0.5.90 meine ich) http://[IP]/api/inverter/id/0 (erster Inverter). Die Ausgabe kann dann in ein Array konvertiert und nach dem Schlüssel/Array [ch][0] gefiltert werden. Das sind die Daten, die Du willst.

SteveBaeumer, 2023-04-08:
Vielen lieben Dank.Mein Problem ist dann wohl der Array nach dem gefiltert wird. Sagt mir Laien nämlich einfach gar nix. 😉--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

closed 2024-01-02 by lumapu

KSNWH, 2024-01-10:
Hallo,
ich hänge mich mal hier rein:
Bisher habe ich keine Erfahrung mit der Ahoi DTU, werde aber demnächst eine für mein BKW einsetzen,
um die Werte in eine bestehende DB und Grafana-Umgebung einzulesen.
Dies möchte ich schon vorbereiten, bevor das BKW läuft.
Meine FRAGE:
a) gibt es ein curl-Kommando, mit dem ich alle relevanten Daten (in Summe) auslesen kann,
z,B. [diese Liste?](https://github.com/lumapu/ahoy/blob/main/User_Manual.md#topicinverter_name_from_setupch0)

b) kann mir jemand das Kommando UND das (ASCII-) Egebnis zur Verfügung stellen.
    PS ich bin Purist und will daraus die relevanten Parameter (per String-Verarbeitung) extrahieren und direkt in die DB schreiben. (one über MQTT, Broker,... zu gehen)

Herzlichen Dank im Vorraus!
Klaus

mager33, 2024-01-10:
Moin, der Link war der Richtige... weiter unten ist die REST API erklärt.

KSNWH, 2024-01-12:
Danke für die Info.
Dann wäre ich dankbar, wenn mir jemand das Ergebnis dieses Kommandos senden könnte:
http://<Ahoi-IP>/api/inverter/id/0
Klaus

mager33, 2024-01-13:
{"id":0,"enabled":true,"name":"HM300","serial":"GEHEIM","version":"10014","power_limit_read":0,"power_limit_ack":false,"ts_last_success":1705163547,"generation":0,"status":1,"alarm_cnt":0,"ch":[[0,0,0,0,0,23.5,30.639,29583,1.3,0,0,0],[26.4,0.05,1.3,29583,30.639,0.433,1.3]],"ch_name":["AC","Battery"],"ch_max_pwr":[null,300]}