lumapu/ahoy #872: How to set limit in HM 1200

aenemy1, 2023-04-09:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: HM 1200
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 230327_ahoy_0.6.0_9c7d7a0_esp8266.bin
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [x ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

How can I set the limit in HM 1200 inverter ? I'm receiving error message "Deserialization failed"

aenemy1, 2023-04-09:
[https://ibb.co/m8gxs1Y](url)

lumapu, 2023-04-13:
laut der URL im Screenshot bist du mit dem AP verbunden. Ich würde vorschlagen Ahoy zuerst mit deinem lokalen WiFi Netzwerk zu verbinden und dann das ganze nochmal zu probieren. Es könnte auch klappen, wenn du Ahoy direkt mit der AP-IP `http://192.168.4.1` verbindest.

skywalkerxxl, 2023-07-18:
I am having the same problem here.

The DTU is in my local Wi-Fi network and  I am connecting to it via  http://192.168.178.52/

I also have tested ahoy version 0.6.9 and 0.7.11.

Does anyone have an idea how to solve that?

lumapu, 2023-07-22:
I tested all versions since `0.6.9` connected to a HM1200 Inverter using the limit function l. I use the Webinterface without any error.
Can you check the heap fragmentation in /system before you try to set a limit. After an OTA update the ESP often has a high heap frag. Then you need to do another reset to get it down below 10%

skywalkerxxl, 2023-07-23:
Thanks for your email.I solved the problem by flashing 0.6.9 again, just adding the inverter and directly changing limits. Afterwards I changed Wi-Fi and the other settings.-- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

closed 2023-08-05 by lumapu