lumapu/ahoy #873: Anfänger hat ein paar Fragen.

giGa977, 2023-04-09:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: Hoymiles HM-600
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X ] external antenna (SMA)

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.0
Github Hash: bb5cb8b 
### Build & Flash Method:
* [X ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo zusammen, 

ich bin ganz neu auf diesen gebiet und habe seit 2 Tagen die Mini PV anlage am laufen und seit heute die Ahoy DTU.
und habe ein paar evtl. auch blöde fragen an euch

Frage 1: werden alle daten gelöscht wenn die Ahoy-DTU vom strom genommen werden. 

Frage 2: was genau wird dann resetet bei diesen punkten im Setup.

Reset values and YieldDay at midnight  -> wenn ich das nicht auswähle zählt er immer weiter denn tages wert ?

Reset values when inverter polling pauses at sunset -> das sagt mir gar nichts
Reset values when inverter status is 'not available'  ->  das sagt mir gar nichts

ich hoffe das ihr für mich licht ins dunkel bringt. 

mit besten grüßen und frohe Ostern Dennis.

QuickStopGroceries, 2023-04-10:
Zu deiner 1. Frage:

Der Gesamtertrag (Yield Total) wird im Wechselrichter gespeichert und bleibt immer erhalten, auch wenn die DTU vom Strom getrennt wird. Ebenso die Gesamterträge der einzelnen MPPT-Eingänge. Das kann auch von AHOY nicht manipuliert werden - stell es dir vor wie den Kilometerstand beim Auto. 

Die Tageserträge setzt der WR automatisch auf 0 zurück, wenn er morgens aufwacht, also genügend DC Power von den Panels geliefert wird.

Die Reset-Funktionen sind nach meinem Verständnis eher für Anwender gedacht, welche die DTU per MQTT regelmäßig abfragen. Hier kann es wünschenswert sein, den Tagesertrag zurückzusetzen, sobald die Sonne untergeht (at sunset) oder der Inverter abschaltet (not available) oder sobald der neue Tag beginnt (at midnight), damit Statistiken/Auswertungen nict verfälscht werden.

Ich nutze die Reset-Funktionen nicht, ggf. kann das jemand anderes präziser erklären.

giGa977, 2023-04-10:
Vielen Dank dir erstmal. Dann hab ich die grund dinge ja schon mal richtig verstanden. ebenflas lese ich alles über mqtt aus und lasse es in eine influxdb schreiben um es dann über grafana auzuwerten jedoch. das klappt soweit ganz gut bis auf das der YieldTotal wert direkt in kW/h und nicht wie YieldDay in W/h ausgegeben wird. 
Dadurch habe ich eine hürde zu überwinden beim errechnen des ersparnisses mit der Formel /1000 * Price pro kW/h bekomme ich halt keinen wert angezeigt. 
Wenn ich das Ganze mit dem wert aus YieldDay nehme der in W/h ausgegeben wird kann ich mir nur den Täglichen wert anzeigen lassen der dann auch passt. Hat da Jemand tipp für mich ?

beegee3, 2023-04-11:
wieso /1000 ? Üblicherweise gibt dir der Energieversorger doch den Preis in Cent pro kW/h an.
Also einfach YieldTotal * Preis rechnen. Evtl. /100 um das Ergebnis in Euro zu erhalten.

Mart-Muc, 2023-04-12:
Nicht vergessen, die Yield-Daten sind auf der DC-Seite!
Mann sollte sie also mit 95% Effizienz multiplizieren um den AC-Ertrag zu bestimmen und daraus die Ersparnis ausrechnen zu können.
Somit müssen für beide AC-Yields eh neue Sensoren erschaffen werden, dann fällt  der Wh/kWh Unterschied automatisch raus.

giGa977, 2023-04-12:
@ beegee3 - die Frage ist berechtigt die Vorlage ist aus dem Netz (grafana Board) , und da ist es so vorgegeben denn der wert Yieldday ist in der influxdb so eingetragen 4000 ( wird als w/h ) eingetragen und dagegen YieldTotal wird 4.000 als kW/h eingetragen und mathematisch gesehen in grafana geht das rechnen nicht mit einen Punkt im zahlen wert irgendwie nicht plausibel erst wenn ich die Anzeige auf einen Dezimal wert von 15 nehme bekomme was angezeigt was aber nicht passt wenn ich manuel gegenrechne mit der ausgelesenen kW/h wert. 

@ Mart-Muc das ist schon sinnig, aber es soll ja auch nicht auf den cent genau sein. Ist für eine grobe richtung gedacht.

Ich bin ja auch noch nicht sehr gut im umgang mit grafana und der DTU geschichte. Hätte ja sein können das sich jemand damit auskennt und mir sagt wie ich denn ausgelesenen wert veränder geschrieben bekomme 

Hier mal die Daten aus der influxdb

> select * from yieldday order by time desc Limit 2
name: yieldday
time                value
----                -----
1681323489022331567 1463    < entspricht 1,465 kh/h
1681323474027075900 1463
> select * from yieldtotal order by time desc Limit 2
name: yieldtotal
time                value
----                -----
1681323489023722691 11.528   < entspricht 11,528 kw/h
1681323474028667189 11.528

wenn ich die 11.528 umgeschrieben bekomme auf 11528  dies dann durch 1000 teile und mit dem preis der kw/h strom multipliziere habe ich eine grobe einschätzung des ersparnisses 

ich hoffe das es jetzt etwas plausibler ist. 

grüße Dennis

beegee3, 2023-04-12:
Der Punkt bei 11.528 entspricht dem Komma in der deutschen Schreibweise. Ist halt die internationale (englische) Schreibweise. Damit kann man rechnen.
Ich nutze influxdb nicht, denke aber es lässt sich mit einer etwas anderen Abfrage lösen.
Angenommen, dein kW/h Preis ist 45 Cent, dann versuch doch mal

`select max(yieldtotal*45) from yieldtotal`

bzw. in Euro (achte auf die Schreibweise, wieder mit Punkt):

`select max(yieldtotal*0.45) from yieldtotal`

und noch mit @Mart-Muc's Effizienz Faktor:

`select max(yieldtotal*0.95*0.45) from yieldtotal`

Evtl. lassen sich die Faktoren auch vor max(...) schreiben (geht nicht bei allen DBs; bei influxdb ???):

`select 0.95*0.45*max(yieldtotal) from yieldtotal`

giGa977, 2023-04-14:
Hallo zusammen,

ich habe es so versucht das ergebnis was mir angezeigt wird ist 0.00 €

> `select max(yieldtotal*45) from yieldtotal`
> 
> `select max(yieldtotal*0.45) from yieldtotal`
> 
hätte ja klappen können. oder aber ich bin nicht schlau genug um dies richtig zu verarbeiten. 

grüße

closed 2023-07-23 by lumapu