lumapu/ahoy #875: Wechsel AhoyDTU 0.6.0 zu ältere Version?

schulle61, 2023-04-10:
## Hardware
  * [ ] ESP8266
  * [ x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo,

Wie kann ich von der Aktuellen Ahoy DTU  zu einer älteren wechseln? Wenn ich es einfach über Software aktualisieren machen, dann schreibt er zwar erfolgreich,  aber danach kann ich nicht mehr auf die Ahoy DTU zugreifen.
Seit ich auf die Version 0.5.66 und 0.6.0 aktualisiert habe, geht bei der gebauten DTU das Display nicht mehr. Es zeigt garnix mehr an

Wie kann ich auf OpenDTU wechseln?

tastendruecker123, 2023-04-10:
Zurück geht's am besten mit dem [Flash Tool](https://www.espressif.com/en/support/download/other-tools) von Espressif mit folgenden Einstellungen:

[image]

Erst Erase, dann Start, Wichtig ist, dass man das Häkchen setzt vor der zu flashenden Datei.

schulle61, 2023-04-10:
So habe es so auf ner älteren Version geflasht. Ist auch alles grün und finish. Aber trotzdem finde ich die AhoyDTU im Netzwerk nicht. Mache ich was falsch?

Das Display hatte vor den neuen Updates funktionert. Die Pinbelegung die man bei der neuen Version ändern muss, die sind so eingestellt wie die sein sollen
[image]
[image]
[image]
[image]

/

closed 2023-04-10 by schulle61

schulle61, 2023-04-10:
.

tastendruecker123, 2023-04-10:
Vorher Erase gemacht? Macht er einen Access Point auf?