lumapu/ahoy #877: Inverter Connection only with serial Debug[Bug]

LukasFritz, 2023-04-10:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus, square dot

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Tested Inverter Intervall 30 seconds and 10 seconds

Verschiedene Intervalle getestet

### Error description

Inverter Vebrindung wird nur aufgebaut nur wenn die serielle USB Debug Konsole geöffnet ist.

tastendruecker123, 2023-04-10:
Bitte mal testen mit Stromversorgung über Powerbank (nicht Netzteil). Ist das Web Interface erreichbar wenn die Konsole nicht verbunden ist?

LukasFritz, 2023-04-10:
Diverse Spannungsversorgungen (Auch Powerbank) getestet, immer selbes Fehlerbild.
Webinterface usw. erreichbar. Auch im Seriel out im Web-Interface gehen die Telegramme raus.
Sobald man die USB Debug Konsole schließt (Web-Debug von ahoydtu.de als auch via Putty) verliert der ESP die Verbindung zum Inverter.

By the Way, es handelt sich um einen WEMOS D1 Pro, einziger unterschied ist dabei jedoch die WLAN Antenne sowie mehr Flash Speicher. Sehe ich also nicht als Problem an.

LukasFritz, 2023-04-10:
Sind beide auf Off
IRQ ist verbunden

closed 2023-04-10 by LukasFritz

LukasFritz, 2023-04-10:
D3 also wie im Göttingen Started beschrieben.

tastendruecker123, 2023-04-10:
Ich hab so ein bisschen Pin D8 in Verdacht, weil der auch mit dem Reset Schaltkreis des USB-Serial Adapters verbunden ist. Der Pro nutzt etwas andere Widerstandswerte und es ist ein anderer USB Chip verbaut.

Wenn du es mit dem Pro noch mal versuchen solltest, würde ich dir empfehlen für CS nicht den Pin D8 sondern D1 zu benutzen. Wenn es dann immer noch nicht funktioniert, würde ich CE mit D2 verbinden statt mit D4. Und für IRQ könnte man D0 statt D3 benutzen.

D8 wäre aber der erste Kanditat, den ich austauschen würde. Ob die Konsole auf ist oder nicht, bekommt eigentlich nur der USB-Serial Chip mit, nicht der ESP, und D8 ist (soweit ich sehe) der einzige Pin, der sowohl mit dem USB-Serial Chip als auch mit dem NRF verbunden ist.