lumapu/ahoy #879: 0.6.5 - LED active high/low

knickohr, 2023-04-12:
NOTE: incompatible change: renamed led_high_active to act_high, maybe setting must be changed after update

Funktioniert leider immer noch nicht.

Manuelles editieren der JSON mit anschließendem Import funktioniert jetzt leider auch nicht mehr 😢

Bei der Gelegenheit auch gleich mal verhindern das man den Import-Button auch drücken kann wenn keine Datei ausgewählt wurde. Latsche da mit meinen Wurstfingern immer auf den Button drauf. Oder den Button nicht so nahe am Auswählen-Button platzieren 😉

knickohr, 2023-04-13:
Danke für die Mühe 😉 Brachte aber fast gar nichts 😲
Zwar kann man jetzt wieder den Trick über das Editieren der JSON machen, aber in der GUI speichert er das LED active high immer noch nicht ab. Steht jedesmal nach dem Reboot wieder low drin.

Bezüglich des Import Buttons hat sich bei mir rein gar nichts verändert 🤔

[image]

tastendruecker123, 2023-04-13:
Ich denke, dass ich das Problem gefunden habe. Ich habe einen PR erstellt, der dieses und andere Problem beim Speichern behebt.

knickohr, 2023-04-14:
Yeahhh ! 👍

closed 2023-04-14 by knickohr