lumapu/ahoy #880: 0.6 - no module and inverter data under http://.../api/live anymore

edmunch, 2023-04-12:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

{"wifi":{"ssid":"O2","pwd":"","dev":"AHOY-DTU","adm":"","prot_mask":61,"dark":false,"ip":"","mask":"","dns1":"","dns2":"","gtwy":""},"nrf":{"intvl":30,"maxRetry":5,"cs":15,"ce":2,"irq":0,"sclk":0,"mosi":0,"miso":0,"pwr":2},"ntp":{"addr":"pool.ntp.org","port":123},"sun":{"lat":5,"lon":1,"dis":false,"offs":300},"serial":{"intvl":5,"show":false,"debug":false},"mqtt":{"broker":"","port":1883,"user":"","pwd":"","topic":"inverter","intvl":0},"led":{"0":255,"1":255},"plugin":{"disp":{"type":0,"pwrSafe":false,"pxShift":false,"rotation":0,"contrast":60,"data":255,"clock":255,"cs":255,"reset":255,"busy":255,"dc":255}},"inst":{"en":false,"rstMidNight":false,"rstNotAvail":false,"rstComStop":false,"iv":[{"en":true,"name":"HM-600","sn":18xxxxx,"yield":[0,0,0,0],"pwr":[380,380,0,0],"chName":["M1","M2","",""]}]}}

updated via web interface from 0.5.66 to 0.6
using the json api to connect to openHAB

### Error description

Hi, thank you for your work.

did you changed the behaviour of the JSON frontend?

no module and inverter data under http://.../api/live anymore after upgrade to 0.6

i just get:

{"generic":{"wifi_rssi":-77,"ts_uptime":2402,"menu_prot":false,"menu_mask":61,"menu_protEn":false,"esp_type":"ESP8266"},"refresh":30,"ch0_fld_units":["V","A","W","Hz","","°C","kWh","Wh","W","%","var"],"ch0_fld_names":["U_AC","I_AC","P_AC","F_AC","PF_AC","Temp","YieldTotal","YieldDay","P_DC","Efficiency","Q_AC"],"fld_units":["V","A","W","Wh","kWh","%"],"fld_names":["U_DC","I_DC","P_DC","YieldDay","YieldTotal","Irradiation"],"iv":[true,false,false,false,false,false,false,false,false,false]}

**before (with 0.5.66) there have been the inverter and module data
so that i could fetch the inverter AC voltage  for example with JSONPATH:$.inverter[0].ch[0].[0]**

now i see under http://..../api/record/live some values but without information of the inverter or module:

{"inverter":[[{"fld":"U_DC","unit":"V","val":"32.50"},{"fld":"I_DC","unit":"A","val":"0.52"},{"fld":"P_DC","unit":"W","val":"16.90"},{"fld":"YieldDay","unit":"Wh","val":"815.00"},{"fld":"YieldTotal","unit":"kWh","val":"28.66"},{"fld":"Irradiation","unit":"%","val":"4.45"},{"fld":"U_DC","unit":"V","val":"33.00"},{"fld":"I_DC","unit":"A","val":"0.61"},{"fld":"P_DC","unit":"W","val":"20.10"},{"fld":"YieldDay","unit":"Wh","val":"940.00"},{"fld":"YieldTotal","unit":"kWh","val":"28.48"},{"fld":"Irradiation","unit":"%","val":"5.29"},{"fld":"U_AC","unit":"V","val":"229.30"},{"fld":"I_AC","unit":"A","val":"0.15"},{"fld":"P_AC","unit":"W","val":"35.30"},{"fld":"Q_AC","unit":"var","val":"0.00"},{"fld":"F_AC","unit":"Hz","val":"49.99"},{"fld":"PF_AC","unit":"","val":"1.00"},{"fld":"Temp","unit":"°C","val":"11.90"},{"fld":"ALARM_MES_ID","unit":"","val":"3.00"},{"fld":"YieldDay","unit":"Wh","val":"1755.00"},{"fld":"YieldTotal","unit":"kWh","val":"57.14"},{"fld":"P_DC","unit":"W","val":"37.00"},{"fld":"Efficiency","unit":"%","val":"95.41"}]]}

thank you for your help

Maik

dknubben, 2023-04-12:
The REST-URL is now  http://[IP]/api/inverter/id/0 (the last number is for your inverter(s)).
This is not well documented for now.

lumapu, 2023-04-12:
duplication #864

closed 2023-07-23 by lumapu