lumapu/ahoy #882: Feature Request: Reboot when WiFi is lost

prnzngr, 2023-04-13:
I have the Problem, that my ESP8266 is loosing its WiFi everyday and after a Power Cycle its working again.
Is it possible to implement an automatic reboot if WiFi is down for more than ~ 3 minutes?

knickohr, 2023-04-13:
Definitely not !

There must be another solution as rebooting the DTU in 3 minute intervals.

lumapu, 2023-04-13:
@prnzngr welche Version wird eingesetzt? Hier wurde seit der Release was geändert, daher empfehle ich die aktuelle dev-Version auf jeden Fall zu testen.
Geplante Reboots von Ahoy sind ein no-go.

prnzngr, 2023-04-15:
> Definitely not !
> 
> There must be another solution as rebooting the DTU in 3 minute intervals.

Es war in keinem Wort die Rede von eine drei Minuten Intervall. Sollte das Wlan länger als drei Minuten weg sein dann reboot. Kann auch 5 Minunten sein. War nur ein Vorschlag. 

> @prnzngr welche Version wird eingesetzt? Hier wurde seit der Release was geändert, daher empfehle ich die aktuelle dev-Version auf jeden Fall zu testen. Geplante Reboots von Ahoy sind ein no-go.

Version 0.6.0
Und auch von geplanten Reboots war nie die Rede. Ich verliere EINMAL am Tag die Wlan Verbindung und ein reboot würde helfen. Nicht mehr und nicht weniger

knickohr, 2023-04-15:
Wenn das WLAN weg ist bootet Dir die DTU alle 3 oder 5 Minuten im  Kreise rum bis das WLAN wieder da ist. Ich denke das ist nicht Sinn der Sache 😲

Allenfalls könnte man sich überlegen ob man einen Watchdog in Ahoy einbauen kann der meinetwegen nach 10 Minuten versucht die WLAN neu zu initialisieren. Das muß Dir aber ein Entwickler sagen ob und wie und vielleicht.

onemorename, 2023-04-15:
WLAN fliegt bei mir ebenso nach einigen Stunden immer weg, Version 0.6.7. Grad wieder nur Konsole möglich

I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect
I: MQTT disconnected, reason: TCP disconnect

lumapu, 2023-04-15:
bevor wir hier weiterdiskutieren: wurde die letzte dev-version getestet? 
https://nightly.link/lumapu/ahoy/workflows/compile_development/development03/ahoydtu_dev.zip

onemorename, 2023-04-16:
Ja, tritt genau mit der letzten Dev Version auf, hatte deshalb auch dazugeschrieben 0.6.7. Wlan läuft mit den letzten Fixes stabiler aber bricht trotzdem weg. DTU steht einen Meter vom WLAN AP weg, der via Kabel versorgt wird. An der Stabilität vom Wlan selbst hängts also nicht. Kommunikation mit den Invertern läuft weiter einwandfrei, die DTU zeigt mir am Display korrekte Werte. Aber ist eben via Wlan nicht mehr erreichbar bis zum nächsten Reset

lumapu, 2023-04-17:
oh, die Versionsangabe habe ich übersehen. Verstehe nicht was genau passiert. Kannst du per PC bitte mitloggen, war in Fehlerfall ausgegeben wird?

onemorename, 2023-04-17:
Hatte den Tag über das Logging laufen, als ich heute Nachmittag vom REchner ging war das Webinterface noch erreichbar. Anbei das Log mit dem aktuellen Stand
[ahoylog.txt](https://github.com/lumapu/ahoy/files/11254243/ahoylog.txt)

onemorename, 2023-04-17:
Danke für die schnelle Antwort! Ich hab das Logging nach wie vor laufen und interessanterweise ist er netzwerkseitig wieder zurück gekommen. Wenn ich in das Protokoll schau aber nach wie vor sehr instabil. Der Name von meinem Wifi ist 'Home'. 50% der Pings laufen aktuell auf Timeout. Da gerade keine Sonne scheint sollte auch der MQTT eigentlich ja nichts tun, das entsprechende Setting ist gesetzt dass der nur was durch schickt falls es Änderungen gibt

[ahoylog2.txt](https://github.com/lumapu/ahoy/files/11254812/ahoylog2.txt)

onemorename, 2023-04-17:
Hab am Ende vom letzten Log jetzt nochmal angestückelt was aktuell passiert - immer wieder connect / disconnect. Pings laufen aktuell stabil - letzte Meldung ist auch MQTT connected und aktuell keine weiteren Events. 
Glaub nicht dass es von den Invertern kommt da das sonst ja aktuell (Nachts) keine Auswirkungen mehr haben sollte. MQTT tut ja auch nichts, ist also nur 'user' vom Interface das offline geht. Hoff meine Interpretation macht Sinn soweit. Kann man da irgendwie noch bessere Logs ziehen? Das was da kommt scheint mir bisschen spärlich

onemorename, 2023-04-17:
Ist schon wieder tot. Erstmal würde das nur ein kompletter Neustart wieder lösen.

Pinging 192.168.100.85 with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.100.140: Destination host unreachable.
Reply from 192.168.100.85: bytes=32 time=145ms TTL=255
Reply from 192.168.100.85: bytes=32 time=3657ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.100.85: bytes=32 time=3558ms TTL=255
Reply from 192.168.100.85: bytes=32 time=78ms TTL=255
Reply from 192.168.100.85: bytes=32 time=265ms TTL=255
Reply from 192.168.100.85: bytes=32 time=182ms TTL=255

Beim letzten erfolgreichen Ping jetzt auch wieder ein MQTT connect in der Serial Console

onemorename, 2023-04-17:
Da ich gerade aufs Interface komme, hier noch ein paar technische Daten falls das was hilft:

mac
7C:87:CE:2F:6A:C8
sdk
v4.4.2
cpu_freq
240
heap_free
182416
sketch_used
1074
wifi_rssi
-63
ts_uptime
31595
esp_type
ESP32
heap_total
303132
chip_revision
3
chip_model
ESP32-D0WDQ5
chip_cores
2
Radio
nrf24l01+
is connected
Datarate
250 kbps
Power Level
MAX
TX count
5240
RX success
4613
RX fail
4
RX no answer
583
RX fragments
19770
TX retransmits
6858
Sun
Sunrise
18.4.2023, 06:17:44
Sunset
18.4.2023, 20:06:22
Communication start
18.4.2023, 06:17:44
Communication stop
18.4.2023, 20:06:22
Night Communication
disabled

lumapu, 2023-04-17:
@beegee3 weißt du, ob es normal ist wenn man mehrere APs hat, dass so viele Verbindungsversuche gemacht werden? Ich dachte er sucht nach den stärksten und verbindet sich mit diesem. Ich beziehe mich auf Log2.txt

lumapu, 2023-04-17:
@onemorename könntest du testweise MqTT ausschalten?

beegee3, 2023-04-18:
Oje, das läuft ja gar nicht rund.
Den Logs nach scheint mir die MQTT Verbindung instabil zu sein. Es wird permanent 'MQTT disconnected, reason: TCP disconnect' gemeldet, ohne dass dazwischen 'Wifi event: 5' kommt (5 = SYSTEM_EVENT_STA_DISCONNECTED). Das lässt darauf schließen, dass zunächst nur die MQTT Verbindung abreißt.
Der normale Ablauf bei einem (echten) Wifi Verbindungsverlust wäre, dass Ahoy die Standardschleife (`loopStandard`)  beendet und nur die Wifischleife (`loopWifi`) durchläuft, bis die Wifi Verbindung wiederhergestellt ist. D.h. ohne Wifi Verbindung läuft keine Radio Kommunikation und kein MQTT Ticker sondern nur der Wifi Verbindungsversuch.
(nebenbei bemerkt: daher irritiert mich auch das feature request. Ahoy baut, wenn möglich, die Wifi Verbindung wieder auf!)
Auch sind mir in den Logs die Wifi Fehlermeldungen "Software caused connection abort" aufgefallen, die bei einer 'normalen' Wifi Unterbrechung nicht auftreten (z.B. Hardwareausfall, Router Reset usw.) . Hab' danach gegoogelt. Die Meldungen weisen darauf hin, dass irreguläre (verstümmelte) Wifi Kommunikation den ESP zum Wifi Verbindungsabbruch genötigt hat. Dies wird insbesondere im Zusammenhang mit MQTT genannt, wenn MQTT Server und ESP in verschiedenen Subnetzen sind. Auch die in den Logs gesehene Fehlermeldung "Connection reset by peer" weist auf eine abgerissene MQTT Verbindung hin. In den gegoogelten Hinweisen wird auch beschrieben, dass die MQTT Verbindung eine zeitlang funktioniert, bevor sie abbricht. Dann aber wird das ESP Wifi so 'durcheinander gebracht', dass nur noch der Reboot hilft.
Im Prinzip beschreibt das genau das hier beobachtete Problem bis hin zu dem, dass letztlich die eigentliche Wifi Verbindung abgebrochen wird und sich der anschließende neue Verbindungsaufbau sehr problematisch gestaltet (eigentlich sollte die Verbindung zum stärksten AP nach ein paar Sekunden stehen und keine permanenten 'Wifi event: 5' kommen). Daher vermute ich die Ursache in der MQTT Verbindung. Ist es die oben genannte Subnetz Problematik?
Andererseits ist das Wifi RSSI mit -63 nicht gerade das beste. Bei 4 APs würde ich erwarten, dass zumindest einer ein besseres Signal liefert. Wie sieht RSSI nach einem Reboot aus? Ich weiß, andere haben bei der Signalstärke nicht diese Probleme, doch prinzipiell bedeuten schwache Signale, dass die Wifi Kommunikation intern häufig Pakete mehrfach senden muss, bevor sie fehlerfrei empfangen werden. Und das verlangsamt die Kommunikation. Bei 5 Invertern könnte ich mir vorstellen, dass tatsächlich in MQTT ein Datenstau entsteht und der ESP überfordert wird (evtl. durch ein 'run out of memory').
Auch haben diverse AP's Probleme mit IP4 Clients, wenn sie in ein IP6 Netzwerk eingebunden sind, was dann ebenfalls zu den genannten Fehlermeldungen "Software caused connection abort" führt. Wenn möglich, würde ich versuchen, eine direkte Router Verbindung herzustellen, um das auszuschließen.
Hinzu kommt, dass die Inverter Kommunikation auch nicht die beste ist. Zumindest wenn es HM's und keine 2. Gen. MI's sind, ist TX retransmits viel zu hoch. Es sollte nicht mehr als 1/3 des TX count Werts sein. Hier würde ich mal den NRF Power Level runterschrauben auf MIN oder LOW. NRF funkt im selben Frequenzband wie Wifi und Bluetooth und kann durch deren Signale gestört werden. Bei der NRF Reichweite von bis zu 200m ist der Power Level MAX in der Regel zu hoch und sorgt für eine größere Störanfälligkeit. Während der TX retransmits bleibt dem ESP kaum Zeit die asynchrone Wifi Kommunikation zu bedienen, so dass ein Datenstau nicht wirklich abgearbeitet werden kann (falls vorhanden).
Alles in allem ist es möglich, dass eine Reihe von kleinen Problemen sich zu dem hier beobachteten Phänomen aufsummieren.
Wie @lumapu schon schreibt, erscheint es sinnvoll, zunächst eine "Minimalkonfiguration" mit nur einem Inverter und ohne MQTT aufzubauen. Dazu eine direkte Router Verbindung (RSSI ansehen!) und den Power Level anpassen (Radio Statistik ansehen!) Wenn es dann läuft, nacheinander die anderen Komponenten wieder hinzufügen. Aber immer nur eine und dann wieder beobachten. Viel Erfolg!

onemorename, 2023-04-18:
Danke für die ausführliche Antwort! 
Ich habe gestern während der Probleme parallel einen Ping laufen lassen, ab zwei abreißenden Pings hintereinander war dann auch immer der MQTT in der Console disconnected. Sobald der Ping wieder durch ging war auch MQTT jeweils immer zurück auf connected. Bin mir daher unsicher ob MQTT hier wirklich die Ursache ist oder einfach die Verbindung weg war - hab es aber jetzt abgeschaltet und lass den Ping mal weiterlaufen. 
Was meines Erachtens auch gegen MQTT und interferenzen mit der Inverter Kommumikation spricht ist, dass das Verhalten auch (insbesondere) Abends auftritt wo egtl. nicht gepollt wird. Sprich keine Kommunikation mit den Invertern, auch MQTT tut nichts aber die Verbindung ist immer mal wieder weg und kommt zurück. Mal schneller mal ist sie länger weg.
MQTT und ESP sind im selben Netz, handelt sich um ein Mesh Netzwerk auf Basis von AVM Fritz, mit einer 7590AX im Repeater Modus 2m daneben. Hab aktuell einen RSSI von -66 (bei stabiler Verbindung). In der Fritz Übersicht sehe ich, dass der Ahoy auch mit diesem Repeater verbunden ist. Werd mal beobachten ob das anders ist wenn es zu den Abbrüchen kommt. Hab das mal nachgelesen, zwischen -70 und -80db redet man von einer sehr guten stabilen Verbindung daher sollte -66 doch exzellent sein oder?
Grade wo ich schreibe ist Ping wieder tot, mit inaktivem MQTT - das scheidet also wie vermutet als Ursache aus. Kommunikation mit den Invertern läuft weiter und die Daten am Display stimmen, es bleibt also spannend :-)
Würd als nächstes mal die Sendeleistung so weit wie möglich zurück nehmen mit nur einem Inverter
Im übrigen sehe ich das auch so, dass ein täglicher Reboot schwachsinn ist. Wenn es hier etwas gibt das schief läuft sollte es an der Ursache gefixed werden. Nur der Thread hier schien mir related zu sein deshalb hab ichs hier geteilt, lasst mich gern wissen wenn das besser in nem separaten Issue # aufgehoben ist

onemorename, 2023-04-18:
Ne, stehen am selben Tisch an unterschiedlichen Enden :-) 
Bin durch diesen Artikel zu dem Schluss gekommen dass die -63 bzw. -66 ja ok sein dürften - lieg ich da falsch?
https://www.metageek.com/training/resources/understanding-rssi/

knickohr, 2023-04-18:
Ansich ist der Wert OK, aber nicht in 2m entfernten Repeater mit Sichtverbindung. Da müßten es mindesten -30 oder besser sein.

ich schätze mal deine DTU connected nicht auf diesen Repeater sondern auf einen andern.

onemorename, 2023-04-18:
Ne, das hab ich gecheckt, der connected auf die FritzBox 7590AX die hier am anderen Ende vom Tisch als Repeater via Ethernet mit der Fritzbox im EG upgelinked ist, fehlende Verbindungsqualität scheints daher nicht zu sein. Neben der Fritzbox ist ne Ziegelwand, vielleicht reflektiert da was blöd

onemorename, 2023-04-18:
Gibt ein (Labor) update für die FritzBox: 

- **Verbesserung** Verhalten bei Anmeldungen von WLAN-Geräten auf 2.4 GHz verbessert 

Ich fang damit mal an, und hangel mich dann vor

gungam, 2023-04-18:
Ich bin nicht sicher ob es das selbe Problem ist, aber ich habe auch ein Problem mit regelmäßig aufgegangenen ESP8266 die sich durch Powercycle wiederbeleben lassen. 
In seltenen Fällen gerappelt sich das System auch durch warten wieder, dann aber in der Regel nur sehr kurz.
Interssant ist, dass ich die AhoyDTU ebenfalls mit eine Fritzbox (allerdings das ältere 7490 System) und MQTT nutze. 
Aktuell installiert ist die 0.6.0. 
gibt es neben der Seriellen Schnittstelle noch andere Loggingfunktionen?

Erdbeerfranz, 2023-04-19:
Hallo, hab mich extra wegen dem Problem hier angemeldet. Habe ebenfalls den Softwarestand 0.6.0 auf einem ESP8266 bei mir hängt sich das ganze mehrmals am Tag auf, sodass auch nur ein Powercycle Abhilfe schafft, dass sich das ganze "neu verbindet". Sobald sich das ganze aufgehangen hat, steht nur noch "offline" auf dem Display oder aber der Startbindschirm mit Ahoy DTU wird am Display abgebildet. Habe seit kurzem das MQTT aktiviert um die Daten auf IOBroker zu senden und meiner Meinung nach tritt erst seit dem aktivieren des MQTT´s der Fehler auf. Das ganze ist bei mir über eine Vodafone Box mittels Repeater verbunden, weiß nicht ob das hilfreich ist, wobei ja im Raum stand,  dass es an der Fritzbox liegen könnte... denke das ist nicht der Fall, da ich mit der Vodafone Box das Gleiche Problem habe. 
Zum Abstellen des Fehlers habe ich den ESP8266 erneut geflashed, was aber keine Besserung gebracht hat. 
Einen Hardwaredefekt schließe ich aus, da ich noch eine weiter DTU zusammengebaut habe und da hatte ich das Gleiche Problem. Wenn´s hilft könnte ich auch beide DTU´s parallel laufen lassen (ka ob das geht dass beide DTU´s mit dem Wechselrichter kommunizieren) und anschließend berichten.

lumapu, 2023-04-19:
@gungam solange Wifi geht, kann man die Meldungen auch über das Webinterface unter `/serial` abfragen.

@Erdbeerfranz: das haben schon mehrere beobachtet und ich bin auch schon dran dies zu lösen. Das Problem lässt sich wie folgt zusammenfassen: Es wird bei Empfang von neuen Wechselrichterdaten das versenden der MQTT Nachrichten angestoßen. Da dies je nach Wechselrichter (oder auch mehrere) sehr viele Daten sind kommt der ESP kurz ins schwitzen und veruscht alles zu puffern. Das gelingt nicht immer was mit einem Absturz quitiert wird. Abhilfe wird eine asychnrone Abarbeitung dieser Daten schaffen - wir reden hier nur von einem Unterschied im Millisekundenbereich, was aber für einen Mikrocontroller schon eine beträchtliche Zeit ist.
Ich hoffe das die Tage veröffentlichen zu können, gleiches Schema verwendet Ahoy übrigens schon für `Autodiscover` für Homeassistant.

Erdbeerfranz, 2023-04-20:
Hallo, und vielen Dank für die Aufklärung, lumapu. Und sorry, wusste nicht, dass du an dem Thema schon dran bist. 
Wenn ich das richtig verstehe, dann programmierst du, lumapu, den "Code" für die Ahoy, oder? (Sorry für die Ausdruckweise bin Maschinenbauer^^)
Ich weiß nicht, ob das hier in die Runde passt, oder ob das separat besprochen werden soll / muss aber ich hätte noch eine Sache die z. m. für mich als sinnvoll erscheint, wenn diese denn umgesetzt werden könnte. Und zwar läuft bei mir den ganzen Tag, solange die Ahoy mit dem Wechselrichter kommuniziert das Display mit. An dem ESP wären ja noch Pins frei. Es wäre meiner Meinung nach eine gute Idee, wenn man hier einen Taster anbringen könnte, der ein Signal gibt, wann das Display eingeschaltet wird, am besten mit ausschaltverzögerten Timer sodass man den Taster drückt und das Display für z. B. 20 Sekunden leuchtet. Würde vielleicht über´s Jahr gesehen auch ein wenig Strom sparen. Denke programmiertechnisch sollte das doch möglich sein, oder + man bräuchte den Bildschrimschoner nicht da das Display nicht permanent was anzeigt. 
Wie gesagt, wenn der Beitrag hier unpassend ist, bitte verschieben.

beegee3, 2023-04-20:
@Erdbeerfranz beim ESP32 mit Touch Pins wäre das wahrscheinlich einfach. Aber bei ESP8266? Sind da noch passende Pins verfügbar?
Habe kein Display angeschlossen und bin daher vielleicht zu naiv: kann man nicht einfach einen Ein-/Aus-Schalter anbringen, der die Stromzufuhr zum Display unterbricht. Oder einen Taster, evtl. mit einer kleinen Schaltung, die nach dem Drücken zeitverzögert die Stromzufuhr wieder unterbricht?

Erdbeerfranz, 2023-04-20:
Das war nur so ein Gedanke meinerseits, ob das technisch geht d. h. die Pins, die noch verfügbar sind auch Eingänge sind, oder ob das Programmiertechnisch hinhaut das weiß ich leider nicht da bin ich raus. Was ich aber sagen kann ist, dass es nicht funktioniert, wenn man nur das 5V Kabel abklemmt, und dann wenn man das Display braucht wieder anklemmt, sozusagen einen Taster dazwischen klemmt. Dann bleibt der Bildschrim schwarz und nur, wenn alle Kabel dran sind, wird das Display beim neu-starten der DTU "aktiv"

Erdbeerfranz, 2023-04-20:
habe gerade mal nachgesehen, der RX Pin wäre noch unbelegt, laut Datenblatt müsste das ein Input sein, vll kann man den ja nutzen. Sollte ich falsch liegen, oder das ganze technisch nicht umsetzbar sein dann das ganze natürlich vergessen. Ich dachte nur wenn´s technisch geht könnte man die DTU ja mit einem Taster ausstatten und vielleicht ja Softwaremäßig erweitern sodass man auch einen Haken setzen kann in den Einstellungen, dass das Display per Taster aktiviert werden kann, man genügend Zeit hat die Werte abzulesen und somit weniger Strom verbraucht wird falls man das wünscht. Oder aber wenn nicht in den Einstellungen aktiviert, dann läuft das Display ganz normal immer mit.. War nur eine Idee, selbst umsetzten kann ich das nicht da ich keine Programmierkenntnisse habe und daher der Gedankenanstoß

beegee3, 2023-04-20:
> ... dass es nicht funktioniert, wenn man nur das 5V Kabel abklemmt ...

da bin ich raus, aber vielleicht weiß der Displayexperte @knickohr, ob sich **extern** etwas machen lässt?

knickohr, 2023-04-20:
OK, bevor wir hier weiter diskutieren, bitte ich euch, falls das wirklich gewünscht ist, einen neuen Issue aufzumachen. Oder wer etwas Programmierkenntnisse hat, eine PR 😉

Schauen wir uns erst mal den 8266 an und prüfen ob da tatsächlich noch Pins frei haben. Wir sollten jetzt nicht einfach blind sagen, ja da gibt es noch welche, wir müssen auch andere Konfigurationen berücksichtigen. Da wäre z.B. das Nokia Display, das nicht nur 2 zusätzliche Pins wie die OLAeDs braucht, die dürfen wir nicht belegen, ansonsten kommt der Nächste weinend daher das sein Nokia nicht mehr geht 😳

[image]

Das wäre die Beschaltung mit dem (worst case) Nokia-Display. Man sieht wunderbar das es einfach mehr Pins braucht, aber man erkennt auch das wir tatsächlich noch Pins frei haben 😎

D0 und A0 wären noch verfügbar. Damit könnte man was machen. Ich weiß jetzt nicht, ob man an den A0 auch einen Taster dran machen kann, es ist ein analoger Eingang, aber in meinem jugendlichen Leichtsinn würde ich behaupten es funktioniert.

Den D0 würde ich mal außen vor lassen, es gibt hin und wieder Probleme bei diversen 826her mit dem Reset, oder war’s der IRQ ? vom NRF, so das man notfalls da umschwenken könnte.

Zum Thema einfacher Schalter in die Versorgungsspannung des Displays :  Das funktioniert nicht !  Ganz einfach weil dann auch der Controller vom Display ausgeschaltet wird und beim erneuten Einschalten muß das Display also wieder initialisiert werden. Davon bekommt aber der 8266 nichts mit, so das es einfach dunkel bleibt.

Bei. ESP32 könnte man einen freien Touch-Eingang verwenden. Das muß natürlich auch frei konfigurierbar über die GUI sein !

So, wer macht jetzt einen Issue auf, oder besser einen PR mit Beispielcode ?  😇

Den Code dazu könnte man aus der Routine mit der Nachtabschaltung klauen oder anpassen.

Zu bedenken gibt es auch, das die Funktion im Setup der GUI optional aktiviert und deaktiviert werden sollte. Ich persönlich möchte nicht jedesmal einen Taster drücken um ein Display einzuschalten !

Antworten, Kommentare, usw. bitte in einen neuen Issue.

DonTermi, 2023-04-21:
> Ist schon wieder tot. Erstmal würde das nur ein kompletter Neustart wieder lösen.
>
> Pinging 192.168.100.85 with 32 bytes of data: Request timed out. Request timed out. Request timed out. Request [...]

Nach dem Update von 0.6.0 auf 0.6.9 hatte ich genau das selbe Probleme. Kam erst halbwegs wieder drauf nach dem ich mich mit dem Webinstaller Connected hatte und über Console & Logs RESET tätigte. Strom abziehen, Reset Taster am ESP8266 hat alles nichts geholfen.

Nach dem Downgrade auf 0.6.0 läuft das Wifi erstmal wieder stabil.

Erdbeerfranz, 2023-04-22:
@DonTermi, ich habe heute auch auf die 0.6.9 upgedatet und komme auch nicht mehr oder nur sehr schwer auf die DTU drauf und Daten über MQTT bekomme ich auch sehr unregelmäßig. Wie kann ich das ganze wieder auf 0.6.0 downgraden? Gibts dafür eine Anleitung? Besten Dank schon mal vorab

onemorename, 2023-04-24:
@beegee3  & @lumapu: Hab die DTU jetzt unters Dach verpflanzt wo die Inverter drauf liegen und die Sendeleistung runter gefahren auf Min / Low. Was auffällt, die TX retransmits sind nach wie vor knapp unter dem TX count, vielleicht ist in dem RadioModul ja der Wurm drin? Wie ihr schon geschrieben habt bleibt dem ESP während dem TX retransmit wenig Zeit WLan zu bedienen, grade in Verbindung mit dem MQTT Setting neue Werte sofort raus zu jagen könnte ich mir vorstellen dass das System hier ins Schwitzen kommt.

Hab dort -43 RSSI Wlan (Repeater neben dran), da sollte dieses mal alles bestens sein. Ich halt euch auf dem Laufenden

Sollen wir die Unterhaltung in den (neuen) WLAN Bug Thread moven?

closed 2023-07-23 by lumapu