lumapu/ahoy #884: Feature Request: Submit all data as one JSON string to MQTT

igi666, 2023-04-14:
Can you please add an option to (alternatively or additionally) send all data as one JSON string to MQTT?

Tasmota and others sensors I use also submit JSON strings so adding another Sensor using one topic per value seems unnecessary complicated. If it can be configured taht would be nice but I think parsing/filtering on the actual consumer of the string should be totally fine.

njumaen, 2023-04-15:
I support this request. Shelly does it the same way.

lumapu, 2023-04-15:
can you give me a sample JSON structure? I'm not a homeassistant user, so I don't have any experience

igi666, 2023-04-15:
Here's an example String from Tasmota/my powermeter:
{"Time":"2023-04-16T00:07:31","Main":{"counter_pos":2711.720,"counter_neg":1.021,"power":376}}

To map this to here I'd say it should look like this:
JSON:
```
{
  "ahoydtu": {
    "version": "0.6.0",
    "total": {
      "YieldTotal": "555.5",
      "YieldDay": "3500"
    },
    "Inverter1": {
      "available": "1",
      "ch0": {
        "YieldDay": {},
        "YieldTotal": {}
      }
    }
  }
}
```

JSON String:
`"{\"ahoydtu\":{\"version\":\"0.6.0\",\"total\":{\"YieldTotal\":\"555.5\",\"YieldDay\":\"3500\"},\"Inverter1\":{\"available\":\"1\",\"ch0\":{\"YieldDay\":{},\"YieldTotal\":{}}}}}"`

njumaen, 2023-04-16:
Example topic: myshelly/status/switch:0

{"id":0, "source":"WS_in", "output":true, "apower":6.7, "voltage":225.6, "current":0.031, "aenergy":{"total":4179.808,"by_minute":[112.433,112.433,111.935],"minute_ts":1681617719},"temperature":{"tC":35.6, "tF":96.1}}

I use apower (actual power) and minute_ts (it's timestamp), so please do not forget ch0/P_AC and the timestamps.

lumapu, 2023-07-23:
that's to heavy for small ESPs (8266). I don't see the requirement. Please open a new issue or reopen if this request still exists.
Background: Building large Strings like this is realy 'expensive' on an edge device.

closed 2023-07-23 by lumapu