lumapu/ahoy #885: NRF+ detected but no successful RX

seanco1, 2023-04-14:
Good evening,

Are the first gen MI-1200 units supported [serial number begins with 1060xxxxxxxx]? I have tested version 0.60 and 0.67 using (ESP32 with NRF24L01+) and (ESP8266 Wemos D1 V4 with NRF24L01+ and NRF24L01+ PLUS – PA LNA SMA antenna).

Seems the NRF24 units are detected. In the Ahoy software it shows -> "is connected" datarate -> "250 kbps". Tested the ESP32 and 8266 about 5 meters from two inverters but there are never any RX success packets. Tried power levels low up to high. the NRF24L01+ has a 100uf cap soldered to it and the NRF24L01+ PLUS – PA LNA SMA unit has a converter board with caps that takes 5V down to the required 3.3.

I see in the serial console when I add an inverter -> "I: added inverter  (3rd Gen) 1060xxxxxxxx"

tastendruecker123, 2023-04-14:
In the settings, are led1 and led0 set to off?

seanco1, 2023-04-14:
> In the settings, are led1 and led0 set to off?

Both LEDs are set to off currently. I did test only on the ESP32 since it has 2 LEDs, and it was 100% on because of no inverter connections.

tastendruecker123, 2023-04-14:
Got it. Can you take a screenshot of your pin config?

beegee3, 2023-04-14:
for the moment 1st gen MIs are not supported. There are several problems:

- 1st gen MIs only transmit with a datarate of 2MBps and Ahoy is programmed with 250kbps fixed.
- while transmitting the inverters use different channels. The timeslot for each channel before switching to the next channel at a datarate of 2MBps is currently unknown (even if one can assume that it is a multipe of 511 us).
- the 1st gen MI transmit commands are unknown. Maybe they are the same as with the 2nd gen MI.

If you are interested in testing, please feel free to use the appended special version I made for the ESP32 (based on version 0.6.5). This version uses a datarate of 2MBps and recognizes 1st gen MI correctly. 2nd gen MI commands are used and the default timeslot is set to 511us.

Because of your serial number starting with 1060 you did not see fields for 'Max Module Power (Wp)', "Module Name" and "Yield Total Correction [kWh]" in your inverter settings. Now they should be visible. If not, clear your browser cache an reload the settings page. You need to fill out these fields to complete the setup. Save and reboot.
Hopefully you now receive RX packets 🤞!

If you do not receive any RX success packets, I'm sorry, but it was worth a try. 

If it works, but you get a lot of missing frames you may change the timeslot to different value (best guess for a first try: 2*511 = 1022) by setting it on the Serial / Control page:
set "Power Limit Command" selection to "timeslot [us] RXstart [0-4] Datarate [1 = 2MBps, 2 = 250kbps]"
and put the number 102201 in the "Power Limit Value" field (1022 = timeslot, 0 = RXstart, 1 = Datarate).
You may change timeslot and RXstart (= a kind of channel offset) as you like, but datarate should always be 1.
At last klick on "SEND POWER LIMIT". You should immediatly see a confirmation with the new setting in the serial window above.
For testing it helps to set Settings->Inverter->Interval to a value of 5 or less seconds.

Happy testing and please give feedback!

[ahoy_065_special_esp32.zip](https://github.com/lumapu/ahoy/files/11237333/ahoy_065_special_esp32.zip)

seanco1, 2023-04-15:
> 

Thanks for this special ZIP file. I have loaded it and there is progress. Before I made the change using the Power Limit Command I got some rx success but many fragments. At least we receiving packets now.

I changed the "Power Limit Command" to 102201 as suggested and now I don't seem to get RX success (very limit amount) but I get stats on the home page. All 4 PV (with the names I set under inverter) and stats in the green section. I guess this is related to the timeslot / rxslot / data rate I just dont know what I'm doing there lol

nrf24l01+ is connected
Datarate 2 MBps
Power Level MIN
TX count 81
RX success 1
RX fail 10
RX no answer 70
RX fragments 23
TX retransmits 97

beegee3, 2023-04-15:
👍 you're receiving data and it seems to be coded as with 2nd gen MI !
For testing, please don't setup MQTT or a display but **use an interval value of 1 second** (Settings->Inverter->Interval).
This means that every second Ahoy tries to get data from the inverter. You will have to reboot before the change is accepted. Ahoy starts with the default timeslot of 511us and you will recognize a lot of transfer entries in the serial window on the Serial / Control page. Let it run for a while. This will fill your Radio statistics.
For explanation: only a lot of RX fail and RX no answer is bad. Don't worry about RX fragments. It's just showing that not all expected information was returned. The TX retransmits counter is misleading. Ahoy's MI program code needs more than one data transfer to put all information together. After the first successful transfer each other part is counted as a retransmit until all information is received. By that you will see a lot of TX retransmits in your statistics.

Now you may play with the special "Power Limit Command". As said Ahoy starts with a timeslot=511us, RXstart=0, Datarate=1, that is a value of 51101, which seems to be not too bad. Have a look at the serial window. By time you will get an impression, which value gives good results.
You may start by vary the timeslot in steps of 50, that is 550, 600, 650, ... I don't know where to stop, maybe 1500 or 2500.
Put it together with RXstart=0 and Datarate=1 to 55001, 60001 and so on. With a timeslot of 1000 and more you may vary the RXstart, i.e. try 1 as well. That is use values ending with 11 (100011, 105011, ...).
Each time you start with a new value the statistics is resetted. If you get the impression, you have a "good" value let it run a while (several minutes) and look at the statistics. You may improve a "good" value by variing in smaller steps. Each single microsecond may be decisive.
Btw.: If you you switch between the browser pages and can't remember the current value input, just send 0 as value and the current value is shown on the serial window.
Here a statistic from a 2nd gen MI to compare with:
TX count 4203 
RX success 3750
RX fail 16
RX no answer 436
RX fragments 16025
TX retransmits 8727

In a "normal" use you like to have actual values each minute or so. That is the inverter interval setting is something like 30 s.
At last it is possible that with 1st gen MI this needs to be shorter (maybe 10 s) to reach this goal.
Don't expect too much. As far as I know, there is no experience with 1st gen MI, you're the one who has to give the feedback😁 

Happy testing.

beegee3, 2023-04-15:
@rejoe2 @lumapu looks like it's possible to get the 1st gen MIs running! 😃

rejoe2, 2023-04-17:
> @rejoe2 @lumapu looks like it's possible to get the 1st gen MIs running! 😃

Yes, interesting! 
But still, one RX success in 81 tries is a very poor quote... 

@seanco1: Do you have any info about what has been received? Some production related data or "just" the firmware version info?
As we sometimes get very surprising results - would you mind trying with the "2" datarate as well?

(@beegee3 : I'm quite aware you are much deeper in all the details and have good reason to insist on "1". Unfortunately, I most likely will not have much time to spend on this the comming weeks, so I didn't look into the code changes by now).

seanco1, 2023-04-17:
> Yes, interesting! But still, one RX success in 81 tries is a very poor quote...
>
> @seanco1: Do you have any info about what has been received? Some production related data or "just" the firmware [...]

I can send some data tomorrow and show you the TX RX stats. I found that using 153331 "Power Limit Value" gives me the best success. Not great but stats do eventually come though. If I set the data rate to 250K then I don't receiving anything at all, seems the first gen units need 2M data rate.

This is after a couple minutes of using 153331 "Power Limit Value"
nrf24l01+ is connected
Datarate 2 MBps
Power Level LOW
TX count 53
RX success 16
RX fail 3
RX no answer 34
RX fragments 69
TX retransmits 129

Some other hardware info
mac 24:0A:C4:xx:xx:xx
sdk v4.4.4
cpu_freq 240
heap_free 191208
sketch_used 1102
wifi_rssi -41
ts_uptime 86501
esp_type ESP32
heap_total 308456
chip_revision 1
chip_model ESP32-D0WDQ6
chip_cores 2

Firmware running is 0.6.5

Couple lines of the serial output
16:52:14 I: (#0) sth. missing: Request Retransmit 0x39
16:52:14 39 pid: 39
16:52:14 I: TX 11B Ch3 | 39 41 50 18 18 81 30 83 52 39 71 
16:52:14 I: (#0) sth. missing: Request Retransmit 0x39
16:52:14 39 pid: 39
16:52:14 I: TX 11B Ch23 | 39 41 50 18 18 81 30 83 52 39 71 
16:52:14 *** time: 303 ***
16:52:14 I: (#0) got all msgs
16:52:22 I: (#0) resetPayload
16:52:22 I: (#0) prepareDevInformCmd 0x05
16:52:22 I: (#0) enqueCommand: 0x36
16:52:22 I: (#0) enqueCommand: 0x05
16:52:22 36 pid: 05
16:52:22 I: TX 11B Ch40 | 36 41 50 18 18 81 30 83 52 05 42 
16:52:22 I: (#0) nothing received
16:52:32 I: (#0) resetPayload
16:52:32 I: (#0) prepareDevInformCmd 0x05
16:52:32 I: (#0) enqueCommand: 0x36
16:52:32 I: (#0) enqueCommand: 0x05
16:52:32 36 pid: 05
16:52:32 I: TX 11B Ch61 | 36 41 50 18 18 81 30 83 52 05 42 
16:52:32 *** time: 7071 ***
16:52:32 I: (#0) next request is 0x37
16:52:32 37 pid: 37
16:52:32 I: TX 11B Ch75 | 37 41 50 18 18 81 30 83 52 37 71 
16:52:32 I: (#0) sth. missing: Request Retransmit 0x37
16:52:32 37 pid: 37
16:52:32 I: TX 11B Ch3 | 37 41 50 18 18 81 30 83 52 37 71 
16:52:32 I: (#0) sth. missing: Request Retransmit 0x37
16:52:32 37 pid: 37
16:52:32 I: TX 11B Ch23 | 37 41 50 18 18 81 30 83 52 37 71 
16:52:32 *** time: 721 ***
16:52:32 I: (#0) next request is 0x38
16:52:32 38 pid: 38
16:52:32 I: TX 11B Ch40 | 38 41 50 18 18 81 30 83 52 38 71 
16:52:32 *** time: 659 ***
16:52:32 I: (#0) sth. missing: Request Retransmit 0x39
16:52:32 39 pid: 39
16:52:32 I: TX 11B Ch61 | 39 41 50 18 18 81 30 83 52 39 71 
16:52:32 I: (#0) sth. missing: Request Retransmit 0x39
16:52:32 39 pid: 39
16:52:32 I: TX 11B Ch75 | 39 41 50 18 18 81 30 83 52 39 71 
16:52:33 I: (#0) sth. missing: Request Retransmit 0x39
16:52:33 39 pid: 39
16:52:33 I: TX 11B Ch3 | 39 41 50 18 18 81 30 83 52 39 71 
16:52:33 I: (#0) sth. missing: Request Retransmit 0x39
16:52:33 39 pid: 39
16:52:33 I: TX 11B Ch23 | 39 41 50 18 18 81 30 83 52 39 71 
16:52:34 I: (#0) sth. missing: Request Retransmit 0x39
16:52:34 39 pid: 39
16:52:34 I: TX 11B Ch40 | 39 41 50 18 18 81 30 83 52 39 71 
16:52:42 I: (#0) resetPayload
16:52:42 I: (#0) prepareDevInformCmd 0x05
16:52:42 I: (#0) enqueCommand: 0x36
16:52:42 I: (#0) enqueCommand: 0x05
16:52:42 36 pid: 05
16:52:42 I: TX 11B Ch61 | 36 41 50 18 18 81 30 83 52 05 42 
16:52:42 I: (#0) nothing received
16:52:52 I: (#0) resetPayload
16:52:52 I: (#0) prepareDevInformCmd 0x05
16:52:52 I: (#0) enqueCommand: 0x36
16:52:52 I: (#0) enqueCommand: 0x05
16:52:52 36 pid: 05
16:52:52 I: TX 11B Ch75 | 36 41 50 18 18 81 30 83 52 05 42 
16:52:52 I: (#0) nothing received
16:53:02 I: (#0) resetPayload
16:53:02 I: (#0) prepareDevInformCmd 0x05
16:53:02 I: (#0) enqueCommand: 0x36
16:53:02 I: (#0) enqueCommand: 0x05
16:53:02 36 pid: 05
16:53:02 I: TX 11B Ch3 | 36 41 50 18 18 81 30 83 52 05 42 
16:53:02 *** time: 9592 ***
16:53:02 I: (#0) next request is 0x37
16:53:02 37 pid: 37
16:53:02 I: TX 11B Ch23 | 37 41 50 18 18 81 30 83 52 37 71 
16:53:02 *** time: 512 ***
16:53:02 I: (#0) next request is 0x38
16:53:02 38 pid: 38
16:53:02 I: TX 11B Ch40 | 38 41 50 18 18 81 30 83 52 38 71 
16:53:02 *** time: 1319 ***
16:53:02 I: (#0) sth. missing: Request Retransmit 0x39
16:53:02 39 pid: 39
16:53:02 I: TX 11B Ch61 | 39 41 50 18 18 81 30 83 52 39 71 
16:53:02 I: (#0) sth. missing: Request Retransmit 0x39
16:53:02 39 pid: 39
16:53:02 I: TX 11B Ch75 | 39 41 50 18 18 81 30 83 52 39 71 
16:53:02 I: (#0) sth. missing: Request Retransmit 0x39
16:53:02 39 pid: 39
16:53:02 I: TX 11B Ch3 | 39 41 50 18 18 81 30 83 52 39 71 
16:53:02 *** time: 229 ***
16:53:02 I: (#0) got all msgs
16:53:12 I: (#0) resetPayload
16:53:12 I: (#0) prepareDevInformCmd 0x05
16:53:12 I: (#0) enqueCommand: 0x36
16:53:12 I: (#0) enqueCommand: 0x05
16:53:12 36 pid: 05
16:53:12 I: TX 11B Ch23 | 36 41 50 18 18 81 30 83 52 05 42 
16:53:12 I: (#0) nothing received

rejoe2, 2023-04-17:
OK, great! 
You seem to receive the entire production data and firmware data should also be available (as there are no requestst in the log)?

If you also enable "serial debug" (settings/system config) you might also see the RX data, which is helpfull to understand how the code works internally. 

Still a success rate of around 30% imo is rather poor, but nevertheless, this is a great starting point!

@beegee3 : Is there anything from my side to support atm?

beegee3, 2023-04-17:
first of all it's a great success! Until now, we knew nothing about 1st gen MIs. Now we have something to build on.
@seanco1 As I see you work with multiples of 511us but I wonder that RXstart = 3 gives the best results so far. Did you try values around 1533us like 1510us up to 1550us in steps of 5 ? Or (derived from 153331) shorter intervals like 102221 or 51111 ?
@rejoe2 I would suggest we still wait if @seanco1 is able to get better results before realizing fixed values for timeslot and RXstart. Maybe we'll know more by the end of the week. We need to switch the datarate between 250kbps and 2MBps (together with changing timeslot and RCstart) for a mixed setup with 1st gen MIs and 2nd/3rd gen inverter. This should be done right before sending a request in the `sendPacket` function. I'll think about something to implement this.
All other adaptions are easy, most of them I did before to create the special test version. So everything's fine atm.

closed 2023-07-23 by lumapu

rejoe2, 2023-10-04:
As atm we're working on inverter specific timings an/or PA-levels and so on, we may come back to this again, but atm. I'm not able to do testing myself for the next couple of days. How about the status, @seanco1?

@beegee3 : If you're interested in the recent heuristics code from the "all-in-one-proposal" originally by @Oberfritze, pls have a look at my repo; latest (unfortunately non working) revision is https://github.com/rejoe2/ahoy/pull/6/files.

beegee3, 2023-10-05:
@rejoe2 wenn du evtl. MIs der 1. Generation integrieren willst, denk' bitte daran, dass die nur auf 2MBps kommunizieren, d.h. man muss den NRF abhängig von der Inverter-Generation zwischen 250kBps und 2MBps umschalten. Insgesamt braucht man einiges an Erweiterung (Seriennummern der 1.Gen zulassen, also 4.Stelle = 0, eine NRF Umschaltfunktion, usw.). Dachte eigentlich, das Interesse ist zu gering, um sich die Mühe zu machen. Zudem kostet es Speicher, der eh schon knapp wird.
Warum funktioniert dein Repo eigentlich nicht? An den kleinen Bugs (z.B. `hmRadio` `rf24RxHMCh1` ist als 5x5 dimensioniert, hat aber nur 2 Spalten) kann es doch nicht liegen.

rejoe2, 2023-10-06:
> @rejoe2 wenn du evtl. MIs der 1. Generation integrieren willst, denk' bitte daran, dass die nur auf 2MBps kommunizieren, d.h. man muss den NRF abhängig von der Inverter-Generation zwischen 250kBps [...]

Denke auch, dass das Interesse eher nicht da ist. Andererseits ist der Mehraufwand _möglicherweise_ doch überschaubarer als evtl. gedacht, weil die aktuelle devel eh' vorsieht, das Inverter-Objekt bis auf sendCommand-Ebene runter zu übergeben (ggf. mit auf den Command entsprechend angepaßten timeouts etc.), und da ist es m.E. nur noch eine Kleinigkeit, auch noch die Datenrate umzustellen. Falls (!) der Rest nicht groß abweicht. UU. könnte man mit compiler-flags arbeiten und dafür z.B. das Display weglassen, dann sollte auf dem ESP32 wieder genug Platz sein; das opendtu-Board mit dem ESP32-S-irgendwas hat eh' genug Ressourcen...

>Warum funktioniert dein Repo eigentlich nicht? An den kleinen Bugs (z.B. `hmRadio` `rf24RxHMCh1` ist als 5x5 dimensioniert, hat aber nur 2 Spalten) kann es doch nicht liegen.

Ich weiß es im Moment nicht und kann grade auch nicht testen, bin erst nächste Woche wieder an der Hardware und habe den ESP mit meiner Version in eine Bootloop geschickt...

rejoe2, 2023-10-06:
@beegee3 : Danke auch für den Hinweis wegen der Array-Größe. Das ist noch der an der Stelle unverändert übernommene Original-Code von @Oberfritze aus "all-in-one". Werd's anpassen, wobei mir noch nicht klar ist, wie man das ganze am sinnvollsten für alle 3 3rd-gen-Inverter-Varianten ergänzen kann. Vielleicht reichen nur 2 Empfangskanäle auch für die 4ch-Geäte? 
Wollte ich eigentlich demnächst mal ausprobieren, aber das muss jetzt halt aus den o.g. Gründen warten...

alromh87, 2025-09-02:
Hope there's still interest and activity on this topic, I want to integrate 2 first generation MI-1200 Inverters, this are installed units that have only a partial serial number starting with 1060 and missing 3 characters. I'm trying to brute-force the identification by sending 0x02 command. I test it with a second generation unit and it works, but If I change to 2M I can't get any answer.

I am not sure its because of the differences in the firmware or selecting the channel.

I will try to build a sniffer to capture and identify packages.

Once I get the serial I will be more than happy to integrate the code to support first gen.

Any thoughts?

alromh87, 2025-09-02:
Thank you, glad to be here. 

I'm using python code running on a pi zero,  sending multiple times the 0x02 broadcast request and listening for the answer on different channels. 

I based my dinner script on this code I found on https://github.com/bri1234/Hoymiles-HM-RaspberryPI that I found on the Mikrocontroller thread that seems to contain a lot of the information of hm dtu.

https://www.mikrocontroller.net/topic/525778

I've been reading as much information as I could for the past few days, if you could point me to the original mi codebase would be much appreciated. 

I also looked into the pi code that's on the tools directory.

alromh87, 2025-09-02:
@rejoe2 
I think I found it, Is this the original MI codebase that runs on esp8266 that you mentioned earlier?

https://github.com/Ziyatoe/DTUsimMI-Hoymiles

It looks that it also implements a sniffer, I will give it a try.

rejoe2, 2025-09-02:
Note: for 2nd and 3rd Gen. inverters, the RX channel is relative to the tx channel, answers may be repeated on the previous or following channel. For details see hmRadio.hm in Ahoy codebase.

alromh87, 2025-09-03:
Success, my brute force method worked to find one of my partial serial, the one missing 3 digits, now I need to find the other inverters serial, hope its close enough for brute force to be feasible. Could be since they where bought together.

```
I used message 0x0f and getting a response on channel 3
[TRY] -> 106040903281                                                                                                                   
[C61][C1][C35][C35][C23][C35][C3][C35][C75][C6][C9][C3]Response : 8f 40 90 32 81 40 90 32 81 12 00 6e 00 01 05 fa 0d                    
  [FOUND] -> 106040903281                                                                                                               
Stored status: 200                                                                                                                      
Body:                                                                                                                                   
 Serial saved securely.
```

From this I know:
- Inverters answer to this message
- Channels are correct
- Data-rate is correct

I have a known inverter to make it work with ahoy.

Now I will
- Try the provided [ahoy_065_special_esp32.zip](https://github.com/lumapu/ahoy/files/11237333/ahoy_065_special_esp32.zip)
- Compile and test the ESP8266 DTUsimMI-Hoymiles version
- Make changes to latest ahoy to get first generartion MIs working

If I can't get the missing serial number I could:
- Build a sniffer and analyze data https://github.com/Ziyatoe/DTUsimMI-Hoymiles/blob/main/sniffer/NRF24_sniff_toe.ino
- Analyze this logs and break how DTU unit gets Serials in short time, probably a broadcast address, either inverters or DTU listening on a secondary pipe.

Note: to test the provided bin I had to erase flash and then flash with bootloader and partitions from 0.6.0

`esptool --port /dev/ttyUSB0 --chip esp32 --before default_reset --after hard_reset   write_flash --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader.bin   0x8000 partitions.bin   0x10000 ahoy_065_special_esp32.bin`

alromh87, 2025-09-04:
@beegee3 
I loaded the provided firmware, thank you, I can get data now, I wonder what should I set on "Yield Total Correction [kWh]"

Attached capture of the screen, this panels seem to be dying(that's why I'm looking into them) and it was late. 

[image]

Also log
[logAhoy2M.txt](https://github.com/user-attachments/files/22129543/logAhoy2M.txt)

I have been going trough the Chinese code and time window for 2M seems to be 2/3 of the 250K time window, since more data can be send in less time, attached relevant code.

```
void UsartNrf_SendLoop_SetPackageTimeOut(bool Stop)
...
            if((CurNetCmd == NET_INIT) || (CurNetCmd == NET_WAIT_DOWNLOAD_DAT) || (CurNetCmd == NET_WAIT_DOWNLOAD_PRO))
            {
                if(Uart_CurrentDtuBaud == Device_250K)
                {
                    LocalTime_TimeOut = LOCAL_TIME_100MS * 5;
                }
                else if(Uart_CurrentDtuBaud == Device_2M)
                {
                    LocalTime_TimeOut = LOCAL_TIME_100MS * 3;
                }
                else
                {
                    LocalTime_TimeOut = LOCAL_TIME_100MS * 3;
                }
            }
            else
            {
                if(Uart_CurSendMainCmd == BROADCAST)
                {
                    LocalTime_TimeOut = 3 * LOCAL_TIME_1S;
                }
                else if(Uart_CurSendMainCmd == DOWN_PRO)
                {
                    if(Uart_CurrentDtuBaud == Device_250K)
                    {
                        if((Uart_SendBuffer[10] & 0xf0) == 0x10)
                        {
                            LocalTime_TimeOut = LOCAL_TIME_100MS * 3;
                        }
                        else
                        {
                            LocalTime_TimeOut = LOCAL_TIME_100MS;
                        }
                    }
                    else if(Uart_CurrentDtuBaud == Device_2M)
                    {
                        if((Uart_SendBuffer[10] & 0xf0) == 0x10)
                        {
                            LocalTime_TimeOut = LOCAL_TIME_110MS;
                        }
                        else
                        {
                            LocalTime_TimeOut = LOCAL_TIME_55MS;
                        }
                    }
                }
                else
                {
                    if(Uart_CurrentDtuBaud == Device_250K)
                    {
                        LocalTime_TimeOut = LOCAL_TIME_100MS * 5;
                    }
                    else if(Uart_CurrentDtuBaud == Device_2M)
                    {
                        LocalTime_TimeOut = LOCAL_TIME_100MS * 3;
                    }
                    else
                    {
                        LocalTime_TimeOut = LOCAL_TIME_100MS * 3;
                    }
                }

```

Next step will be testing on latest firmware and change bands on the fly like DTUPro does.

alromh87, 2025-09-06:
I captured some transmission stats, 2M communication seems to less stable, maybe due to base channel hopping? Or faster responses get lost? If someone have any ideas, let me know.

[image]
[image]
[image]

I will analyze logs and make some test. For now I can at least have some info as of whats going on, It seems the problem is unit 1 and 2 are configured for a lower voltage profile than unit 3 causing them to turn off when there is high solar production, maybe due to wrong cable sizing. I will look into it.

rejoe2, 2025-09-06:
Great job!!!

Some short remarks, feel free to ask for details, if you need more background info: 
- Gen1 seems to really answer fast, so laking messages may be a result of to many sending repetitions/missing acks.
- The "undervoltage" response could be a result of wrong "interpretation" code. I just tried to "translate" the old message structure into the 3rd gen. version. But that's not tested in depth!
-The channel hopping basically seems to work, and the logic for the rx channel for the first answers also seem to be array index of tx+2. There's some drifting of the tx/rx channel pair for the first two units, so the channel array itself seems to be the same as for newer models. You may test, if results are better when changing direction for (tx-) hops - we had some (few, 2 or three!) 2nd gen units responding on other channels than expected, which may have been due to a slightly different "channel pipeline" (or not...).

rejoe2, 2025-09-06:
Additional remark: afair there's some command to change data rates. Perhaps you may try to switch these units to lower rates? At least you may see, if results then are better for these as well.

Btw: I never saw 100% success for 4ch MI before! (OK, to be honest: It's just a couple of logs we are taking into account *grin*)

stefan123t, 2025-09-08:
@alromh87 did you check the Excel Sheet I pointed out. There was an upload/download Grid Profile on some of the tabs when I quickly checked yesterday evening ?

Regarding the intial 0x1213 you may want to check out the code that was afair posted on the Github GIST by @noone2k here: 

https://gist.github.com/noone2k/0b3a116a6f35286abef7199b62a0777a

All the currently known Grid Profiles should be listed here, but there may be some gaps as not so many MI inverters are used with Ahoy/OpenDTU in general.

https://github.com/tbnobody/OpenDTU/wiki/Grid-Profile-Parser

alromh87, 2025-09-09:
@stefan123t Yes the excel was very useful to understand what is going on, unfortunately the only two grid commands are:

- 0A Download grid profile (Used to upload a grid profile)
- 10 Collect grid profile (I use this but only gives Code and Version)

So there is no command to get the complete grid profile I noticed 0A command addressing started at 0x10 so I tried sending lower values expecting to get the full profile back, with no luck. Using 0x10 triggers profile reception on the MI as expected, but without the hole file I am stuck. 

I checked the known Grid profiles with no luck also, profiles for Mexico haven't being reported.

I am getting a hoymiles DTU and plan to record Grid profile with it to share it, as well as identify and replicate discovery protocol.

With my modified firmware, 1st Gen units keep reporting alongside with 2nd Gen unit, just need to tweak the timings and channel hopping to get a more stable comm link, although it might also be because of 2MBPS having more trouble with walls.

One final note I think Support for 1st Gen MI would be a closer title for the Topic.

best regards

stefan123t, 2025-09-16:
@alromh87 felicitas - well done and congratulations !

We are all looking forward to your pull request. So you got the Gongfa 0x02 Command and the Grid Profile Update. With updated firmware you mean your version of the AhoyDTU, right ? Or did you also implement the firmware update for the Micro Inverter / Hoymiles Inverter by any chance ;)

alromh87, 2025-09-19:
@stefan123t I'm sorry for the delay and possibly rising higher expectations, I have posted the pull request to support 1st Gen MI units, changes are quite simple, just define data-rate based on generation and then change it accordingly before sending a message.

As for implementing Gongfa 0x02, grid profile update, I used python for this but I could implement those in ahoy too, here are some notes and ideas I would like to chat before coding.

**Gongfa 0x02 Command:**
Maybe add a search button in the Inverter section of the settings, to trigger scanning, then show a dialog with found units and the option to add them.

**Grid profile**
Correctly parse profile info for 1st Gen and show in the GUI
Where and how should the action of Grid Profile Update should be triggered, maybe even add a warning before proceeding.

**Firmware Update**
Unfortunately the units updated firmware automatically on connection with DTU, sometime before I started recording communications,  so i could not record the actual firmware, but it could be possible to record firmware from hoymiles DTUs, and from there replicate.

I think we could start separated github issues  for grid profile and firmware update to gather and better organize this information.