lumapu/ahoy #886: WT32-ETH01 support LAN Port in Settings

Matten-Matten, 2023-04-15:
## Hardware
  * [ ] ESP8266
  * [X ] ESP32
  * [ ] Raspberry Pi

Modelname: WT32-ETH01
Retailer URL: [WT32-ETH01](https://amzn.eu/d/49nQwHq)

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.0
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo, ist die Implementierung einer Konfiguration eines LAN Port möglich?
ähnlich wie bei [OpenDTU](https://github.com/tbnobody/OpenDTU/blob/master/docs/DeviceProfiles/wt32-eth01.json)

gruß
matten matten

Matten-Matten, 2023-04-15:
In openDTU sieht es in der WebUi wie folgt aus.

[image]

VitBeck, 2023-04-24:
Kann man die fehlende Reaktion auf diese Anfrage so interpretieren, dass es sonst kein Interesse von anderen Usern besteht und diese Anfrage nicht umgesetzt wird?

geronet1, 2023-04-24:
Ich brauch das auch demnächst, könnte auch den Code beisteuern da bei meiner Anlage kein WLan möglich ist.
Das dauert aber

peff74, 2023-04-24:
Also ich hätte auch Interesse, bringe mich gerade im PlatformIO um...
Bei uns is Nachts WLAN aus^^

roobbb, 2023-04-24:
I am also interested in. Alternatively connecting to a wiz5500/ wiz5100 could even do the job and I could keep the esp8266 based Wemos D1.

lumapu, 2023-04-25:
ist schon in der Pipeline, habe leider aktuell aber nicht allzu viel Zeit. Ich hoffe es sehr bald veröffentlichen zu können

MadMax-git, 2023-05-09:
Would be great!
Actually I'm working with WIFI for testing but in the planned installation LAN would be preferred :-)

lumapu, 2023-05-09:
hier die erste Version, ist auch im Branch `eth` als source verfügbar.

[230509_ahoy_0.6.12_eth.zip](https://github.com/lumapu/ahoy/files/11435523/230509_ahoy_0.6.12_eth.zip)

MadMax-git, 2023-05-09:
WT32-ETH01 ist bestellt :-)

Matten-Matten, 2023-05-09:
@lumapu,
danke für die erste vorab version, ich hab die bin via webinterface geupdatet, leider verbindet der esp sich nicht mit dem Netzwerk. weder WIFI noch Ethernet.

Ich muss mal schauen das ich den via serial neu flashe.
Gruß matten matten

MadMax-git, 2023-05-12:
So, der WT32-ETH01 ist gekommen.
Leider bei mir auch keine Verbindung: weder WLAN noch ETH.
Hab die (falsche) 0.5.12 und die (richtige) 0.6.12 probiert.

Mit der stable und der Dev 0.6.12 geht zumindest eine Verbindung mittels AP AHOY-DTU, allerdings schaffe ich es nicht mit meinem Netz zu verbinden (in dem Fall nicht wichtig, Ziel ist ja LAN aber wollte es mal "fallen lassen").
Es wird zwar gemeldet, dass die Einstellungen gespeichert wurden aber nach dem Neustart (auch schon "hart" probiert), trotzdem keine Verbindung...

Danke für die Mühe!

Kann ich sonst was tun?

Gruß, Joachim

lumapu, 2023-05-16:
ich habe es selbst noch nicht testen können und plane einen Aufbau mit W5500 an einem ESP32, evtl. gibt es hier Erkenntnisse die dir weiterhelfen

roobbb, 2023-05-18:
Hab soeben erfolgreich getestet. Source eth kompiliert für lolin32 lite. Zusammengesteckt mit einem W5500 gemäß config.h.
Zugriff via Ethernet klappt. Debug vom serial monitor schaut so aus:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
I: Initializing FS ..
I:  .. done
[    44][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
W: failed to load json, using default config
I: Settings valid: false
I: RF24 Amp Pwr: RF24_PA_I: LOW
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 3 (~ 2403 MHz)

Model			= nRF24L01+

RF Data Rate		= 250 KBPS

RF Power Amplifier	= PA_LOW

RF Low Noise Amplifier	= Enabled

CRC Length		= 16 bits

Address Length		= 5 bytes

Static Payload Length	= 32 bytes

Auto Retry Delay	= 1000 microseconds

Auto Retry Attempts	= 15 maximum

Packets lost on
    current channel	= 0

Retry attempts made for
    last transmission	= 0

Multicast		= Disabled

Custom ACK Payload	= Disabled

Dynamic Payloads	= Enabled

Auto Acknowledgment	= Enabled

Primary Mode		= RX

TX address		= 0xe7e7e7e7e7

pipe 0 (closed) bound	= 0xe7e7e7e7e7

pipe 1 ( open ) bound	= 0x0807988601

pipe 2 (closed) bound	= 0xc3

pipe 3 (closed) bound	= 0xc4

pipe 4 (closed) bound	= 0xc5

pipe 5 (closed) bound	= 0xc6

I: mEth setup...I: [ETH]: Register for events...
I: [ETH]: begin...
[ETH]: Got event...

ETH Started
done...
I: (#0) resetPayload
I: (#1) resetPayload
I: (#2) resetPayload
I: (#3) resetPayload
I: (#4) resetPayload
I: (#5) resetPayload
I: (#6) resetPayload
I: (#7) resetPayload
I: (#8) resetPayload
I: (#9) resetPayload
I: (#0) resetPayload
I: (#1) resetPayload
I: (#2) resetPayload
I: (#3) resetPayload
I: (#4) resetPayload
I: (#5) resetPayload
I: (#6) resetPayload
I: (#7) resetPayload
I: (#8) resetPayload
I: (#9) resetPayload
D: regularTickers
E (1219) gpio: gpio_install_isr_service(449): GPIO isr service already installed
[ETH]: Got event...
ETH Connected
[ETH]: Got event...
ETH MAC: FE:ED:DE:AD:BE:EF, IPv4: 192.168.0.110
FULL_DUPLEX, 100Mbps
D: onNetwork
D: regularTickers
D: updateNtpTime
ETH.linkUp()=down
ETH.localIP()=192.168.0.110
Go on? Yes...
D: updateNtpTime: checking udp "connection"...
D: updateNtpTime: About to (re)connect...
D: updateNtpTime: Connected...
D: updateNtpTime: prepare packet...
D: updateNtpTime: send packet...
D: updateNtpTime: about to handle ntp packet...
I: [NTP]: 2023-05-18 18:45:20 UTC
```
Ntp sync wollte noch nicht klappen. Tiefer habe ich bisher nicht getestet. 
Wenn ich den Code richtig verstehe, ist alles auf den Wiz5500 ausgelegt (https://github.com/khoih-prog/AsyncUDP_ESP32_W5500). Das WT32-ETH01-Board funktioniert damit nicht (?). 
Vielleicht hilft das weiter 😀

lumapu, 2023-05-19:
habe auch gesehen, dass das WT32 einen anderen LAN Chip verwendet, aber das kann man bestimmt auch integrieren.

lumapu, 2023-05-19:
> [    44][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
> W: failed to load json, using default config
I: Settings valid: false

das ist komisch, ich sehe es bei mir auch. Man kann keine Einstellungen speichern. Was ich noch nicht verstehe wie das mit Ethernet zusammenhängt

roobbb, 2023-05-19:
Im Framework vom esp32 kommt die ETH.h mit, welche sich um die LAN-Chips kümmert (per default LAN8720). Aber W5500 ist nicht dabei. Für den Esp8266 dürfte das deshalb generell nicht nützen. 
Ist wahrscheinlich einiges an Arbeit nötig, um die möglichen Konstellationen dann zu abstrahieren (Esp32 mit LAN8720, ESP32 mit W5500, Esp8622 mit W5500 etc.).

Wegen Settings hab ich an der settings.h ein paar Kleinigkeiten testweise auskommentiert, aber war anscheinend nix dabei, was die Ursache trifft.

In den Sourcen steckt enorm viel Arbeit drin. Ich steig da leider nur schwer durch. Sorry 😀

lumapu, 2023-05-27:
mal wieder eine neue Version (Änderungen aus dev03 branch gemerged). Um den Settings load failed Fehler zu umgehen habe ich gemerkt, dass man einfach zuerst das Release aufspielen kann, einmal mit dem Wifi verbindet und speichert. Dann per Update das ETH Package drauf und alles funktoniert 😀

[230527_ahoy_eth_esp32_0.6.15.bin.zip](https://github.com/lumapu/ahoy/files/11580569/230527_ahoy_eth_esp32_0.6.15.bin.zip)

MadMax-git, 2023-05-27:
Eben ausprobiert aber ist wohl nur der w5500 Chip "aktiviert"...
Der ist (wohl) im WT32-ETH01 nicht drin (?)... :-\

In der Console bekomme ich folgende Ausgabe:
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
E (472) esp_core_dump_flash: No core dum���ѥѥ���found!
E (472) esp_core_dump_flash: No core dump partition found!
I: Initializing FS ..
I:  .. done
I: Settings valid: true
I: RF24 Amp Pwr: RF24_PA_I: LOW
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 2 (~ 2402 MHz)
Model			= nRF24L01+
RF Data Rate		= 2 MBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 8 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 250 microseconds
Auto Retry Attempts	= 3 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Disabled
Auto Acknowledgment	= Enabled
Primary Mode		= TX
TX address		= 0xe7e7e7e7e7
pipe 0 ( open ) bound	= 0xe7e7e7e7e7
pipe 1 ( open ) bound	= 0xc2c2c2c2c2
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: mEth setup...I: [ETH]: Register for events...
I: [ETH]: begin...
E (1200) w5500.mac: w5500_reset(301): Read MR failed
E (1201) w5500.mac: emac_w5500_init(827): Reset w5500 failed
E (1201) esp_eth: esp_eth_driver_install(214): init mac failed
esp_eth_driver_install failed
done...
I: added inverter HM114183739056
I: (#0) resetPayload
I: (#1) resetPayload
I: (#2) resetPayload
I: (#3) resetPayload
I: (#4) resetPayload
I: (#5) resetPayload
I: (#6) resetPayload
I: (#7) resetPayload
I: (#8) resetPayload
I: (#9) resetPayload
I: (#0) resetPayload
I: (#1) resetPayload
I: (#2) resetPayload
I: (#3) resetPayload
I: (#4) resetPayload
I: (#5) resetPayload
I: (#6) resetPayload
I: (#7) resetPayload
I: (#8) resetPayload
I: (#9) resetPayload
W: WARNING! your NRF24 module can't be reached, check the wiring
E (1269) gpio: gpio_install_isr_service(449): GPIO isr service already installed
`

roobbb, 2023-06-05:
> Der ist (wohl) im WT32-ETH01 nicht drin (?)
> ...
> EDIT: Alternative: wie müsste ich denn ein W5500-Modul an einen ESP32 DEVKIT o.ä. (habe da ja noch welche) anschließen? 

Das WT32-ETH01-Board hat meist den LAN8720. Den wiz5500 m.W.n. generell nicht.
Ich hab nur das Lolin32-lite-Board - Anschluss müsste mit Devkit analog klappen (dürfte ja mehr GPIO rausgeführt haben, als mein lite-Dingens).

Folgende Pin-Map habe ich verwendet:

ESP32 (Lolin32 lite) | NRF24L01 | W5500
-- | -- | --
Vcc 3.3V | Vcc 3.3V | Vcc 3.3V
GND | GND | GND
GPIO04 | CE |  
GPIO05 | CS |  
GPIO16 | IRQ |  
GPIO18 | SCK |  
GPIO19 | MISO |  
GPIO23 | MOSI |  
GPIO12 |   | MISO
GPIO13 |   | MOSI
GPIO14 |   | SCLK
GPIO15 |   | SCS
GPIO22 |   | INT

[image]

MadMax-git, 2023-06-06:
@roobbb: DANKE!
Ich hab mal so W5500 bestellt, mal sehen...

xFvPx, 2023-06-21:
Ich habe auch mal ein ESP32,W5500 und NRF verbunden (PINS wie von roobbb beschrieben). 
Firmware 0.6.9 geflasht NRF erfolgreich geprüft.
Mit 230527_ahoy_eth_esp32_0.6.15 upgedatet.
Console zeigt

21:19:56.406 -> [ETH]: Got event...
21:19:56.406 -> ETH Connected
21:26:56.396 -> [ETH]: Got event...
21:26:56.396 -> ETH Disconnected
21:26:58.399 -> [ETH]: Got event...
21:26:58.399 -> ETH Connected

wenn ich das LAN Kabel stecke/ziehe/stecke.

Keine IP. Auch nicht im DHCP Server zu sehen.

xFvPx, 2023-06-22:
Update: Ich habe noch einmal in die config.h geschaut.
Firmware 0.6.9 geflasht, NRF CE von PIN 4 auf 17 geändert. Mit 230527_ahoy_eth_esp32_0.6.15 upgedatet. W5500 INT von PIN 22 auf 4 geändert.
16:07:06.965 -> [ETH]: Got event...
16:07:06.965 -> ETH MAC: FE:ED:DE:AD:BE:EF, IPv4: 192.168.176.90
16:07:06.965 -> FULL_DUPLEX, 100Mbps
16:07:06.965 -> ETH.linkUp()=down
16:07:06.965 -> ETH.localIP()=192.168.176.90
16:07:06.965 -> Go on? Yes...
16:07:06.965 -> I: [NTP]: 2023-06-22 14:07:06 UTC

Webserver ist über LAN erreichbar. Teste jetzt die Kommunikation zum WR.

roobbb, 2023-06-23:
@xFvPx : Danke für Deine Rückmeldung. Die config.h hatte ich auch konsultiert. Ich hab nochmal bei meinem Aufbau geschaut und mit Schrecken bemerkt, dass ich beim Malen von Schaubild + Map tatsächlich die Pins verwechselt habe 😱.
Sorry für die Verwirrung. Kann leider nicht sagen, wie ich das wieder geschafft habe 🙄.
Hab beides korrigiert, damit keiner mehr drüber stolpern muss. Hoffe, ich hab keinen neuen Bock drin ...

geronet1, 2023-06-23:
Also WT32-ETH01 geht noch nicht?
Aber W5500 geht? Dann bestell ich mir den mal.

roobbb, 2023-06-23:
Ich meine, der WT32-ETH01 klappt noch nicht, weil dazu noch Anpassungen im Code nötig wären. W5500 ist aber im Branch "eth" bereits vorgesehen. Im Versuchsaufbau hat das auch geklappt mit obiger Einschränkung:

>Um den Settings load failed Fehler zu umgehen habe ich gemerkt, dass man einfach zuerst das Release aufspielen kann, einmal mit dem Wifi verbindet und speichert. Dann per Update das ETH Package drauf und alles funktoniert

Zu tieferen oder längeren Tests bin ich leider nicht mehr gekommen. xFvPx ist da bereits weiter.

xFvPx, 2023-06-23:
WT32-ETH01 habe ich nicht getestet.
W5500 läuft  seit gestern. Bis jetzt absolut stabil.

eine Kleinigkeit:
die beiden MQTT states sind nicht ok:
[image]

MadMax-git, 2023-06-24:
So habe es nun auch hinbekommen...
Bei mir kam hinzu, dass das NRF-Modul für diesen Aufbau defekt war :-|
Langzeittest muss noch ein wenig warten... :-|
DANKE!! :-)

Gerd-Watzmann, 2023-07-27:
Einen ganz herzlichen Dank!!

Dank Euch bin ich jetzt einen ganzen Schritt weitergekommen, und für ahoyDTU nicht mehr auf WLAN angewiesen.  Das "Rattennest" als Versuchsaufbau funktioniert, und wird jetzt auf einer Lochrasterplatine ordentlich aufgebaut.

Eine kleine Anmerkung hätte ich aber noch:

Obwohl ich im Web-Interface immer eine statische IP-Adresse vergebe (mit DNS, submask und gateway) zieht sich der ESP32 beim Booten immer eine Adresse vom DHCP-Server. Das ist jetzt zwar kein Beinbruch, denn ich kann ihm ja routerseitig eine feste IP4-Adresse zuweisen. Aber offenbar wird die IP-Konfiguration von der Weboberfläche nicht für LAN übernommen. Dies nur als Hinweis - ich könnte den Code ja selber korrigieren, aber dafür reichen meine Fähigkeiten leider nicht ;-)

[image]

lumapu, 2023-07-27:
@Gerd-Watzmann danke für dein Feedback und Glückwunsch, dass alles soweit läuft.
Würdest du bitte bzgl. der Statischen IP und Ethernet einen eigenen Issue öffnen?

lumapu, 2023-08-04:
habe den Code mal wieder aktualisiert - leider aber noch nicht in Hardware testen können. Basiert auf der Version `0.7.23`.
@Gerd-Watzmann:  Sollte auch Issue #1049 fixen (Statische IP aus den Einstellungen wird jetzt übernommen)

[230804_ahoy_0.7.23_esp32_ethernet.zip](https://github.com/lumapu/ahoy/files/12265099/230804_ahoy_0.7.23_esp32_ethernet.zip)

MadMax-git, 2023-08-06:
Hmm, habe eben versucht es per Web-Interface aufzuspielen aber irgendwie klappt das nicht... :-\
Ich selektiere die bin aus dem ZIP und dann Update. Es erscheint kurz "upate started" aber irgendwie glaube ich nicht, dass es geklappt hat.
Sieht man die aktuelle FW? Bzw. steht unten rechts 0be01a9 :: 0.6.15
Also wohl (immer noch) 0.6.15
Auch ein Reboot (vor/nach dem Update) hat nicht geholfen...

Ich kann aber auch bei 0.6.15 bleiben, was ich brauche bekomme ich :-)

Gruß, Joachim

xFvPx, 2023-08-06:
Das Problem hatte ich auch. Ich konnte z.B. auch die Adresse des MQTT Broker nicht ändern. Ich bin dann einmal zurück auf die offizielle Main und dann auf die neue Eth.

Gerd-Watzmann, 2023-08-23:
Eine kleine Laien-Frage noch: 
wenn ich jetzt den letzten Stand inkl. Ethernet flashen möchte, welche .bin - Datei muss ich dann nehmen?
Besten Dank im Voraus!

lumapu, 2023-08-25:
die Ethernet bin

Matten-Matten, 2023-08-26:
Wird jetzt auch das WT32-ETH01 Board unterstützt oder nur das W5500 Modul?

Gruß
Matten Matten

Gerd-Watzmann, 2023-08-26:
> die Ethernet bin

Also 230817_ahoy_0.7.36_ba218ed_esp32_ethernet.bin 

Mein Problem ist, dass nach dem Flashen mit dieser .bin der ESP32 in einer Endlosschleife läuft und die untenstehenden Meldungen spuckt. Vorheriges Löschen des Speichers mit dem esptool hat leider auch nicht geholfen.

`

ELF file SHA256: 1efa8907fff07b12

E (1305) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
E (482) esp_core_dump_flash: No core dump..Ö…—•—•ΩπÅfound!
E (482) esp_core_dump_flash: No core dump partition found!
I: Initializing FS ..
I:  .. done
[    22][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
W: failed to load json, using default config
I: Settings valid: false
I: RF24 Amp Pwr: RF24_PA_I: LOW
I: Radio Config:
SPI Frequency..= 1 Mhz
Channel...= 3 (~ 2403 MHz)
Model...= nRF24L01+
RF Data Rate..= 250 KBPS
RF Power Amplifier.= PA_LOW
RF Low Noise Amplifier.= Enabled
CRC Length..= 16 bits
Address Length..= 5 bytes
Static Payload Length.= 32 bytes
Auto Retry Delay.= 1000 microseconds
Auto Retry Attempts.= 15 maximum
Packets lost on
    current channel.= 0
Retry attempts made for
    last transmission.= 0
Multicast..= Disabled
Custom ACK Payload.= Disabled
Dynamic Payloads.= Enabled
Auto Acknowledgment.= Enabled
Primary Mode..= RX
TX address..= 0xe7e7e7e7e7
pipe 0 (closed) bound.= 0xe7e7e7e7e7
pipe 1 ( open ) bound.= 0x3604958001
pipe 2 (closed) bound.= 0xc3
pipe 3 (closed) bound.= 0xc4
pipe 4 (closed) bound.= 0xc5
pipe 5 (closed) bound.= 0xc6
I: mEth setup...I: [ETH]: Register for events...
I: [ETH]: begin...

assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

Backtrace: 0x40083b7d:0x3ffb1eb0 0x4008c25d:0x3ffb1ed0 0x4009282d:0x3ffb1ef0 0x4008d26d:0x3ffb2020 0x4012f3b6:0x3ffb2060 0x401174bd:0x3ffb2080 0x401180b8:0x3ffb20c0 0x40116fe1:0x3ffb20e0 0x400f9246:0x3ffb2100 0x400ed4eb:0x3ffb2160 0x400eca59:0x3ffb2200 0x400ed790:0x3ffb2270 0x401056c6:0x3ffb2290

`

xFvPx, 2023-08-28:
Ich habe jetzt auch die
230817_ahoy_0.7.36_ba218ed_esp32_ethernet.bin.
Ich habe aber wie zuvor erst über den Webinstaller installiert und dann die ETH Variante über Webinterface/Update eingespielt.
Keine Endlosschleife.
I: [ETH]: begin...
[ETH]: Got event...
ETH Started
done...

Irgendwas stimmt bei mir aber nicht mit der Nachtschaltung:
[image]

Gerd-Watzmann, 2023-08-28:
> Ich habe jetzt auch die 230817_ahoy_0.7.36_ba218ed_esp32_ethernet.bin. Ich habe aber wie zuvor erst über den Webinstaller installiert und dann die ETH Variante über Webinterface/Update eingespielt. Keine Endlosschleife. I: [ETH]: begin... [ETH]: Got event... ETH Started done...
> 

Das wäre schonmal ein Unterschied, danke für die Info - ich habe es über esptool zu flashen versucht. 

Mit der 0.6.15 geht das, mit der 0.7.36 nicht.

Web-Installer ist bei mir keine Option, dafür müsste ich unter MacOS den Chrome (= Google ..) installieren.

xFvPx, 2023-08-29:
Ich habe auch noch einmal rauf und runter geflasht mit ESP Tool / Platform IO und Webinterface.
Hoffentlich habe ich dabei den Überblick behalten.
Dabei konnte ich die Loop Situation bestätigen. Ohne Vorversion fehlt hier anscheinend etwas.
Was aber wahrscheinlich für dich funktioniert ist: Erst mit ESP-Tool die "230817_ahoy_0.7.36_ba218ed_esp32.bin" flashen. Anschließend über das Webinterface auf die ETH updaten.

Dabei konnte ich folgendes beobachten:
WiFi Version:
[image]

Nur auf ETH upgedatet und nichts verstellt:
[image]

Gerd-Watzmann, 2023-09-07:
Ich habe jetzt erst mit ESP-Tool die 
"230817_ahoy_0.7.36_ba218ed_esp32.bin" geflasht, 
und dann über den Web-Installer die 
"230817_ahoy_0.7.36_ba218ed_esp32_ethernet.bin":

Ich bin dann wieder in der Endlosschleife, ohne Ethernet-Verbindung.

Die "230817_ahoy_0.7.36_ba218ed_esp32.bin" funktioniert über WLAN,
*aber* : ich komme nicht auf die Settings - es wird ein "Login" von mir verlangt.

Ich glaube, ich gebe auf ...

lumapu, 2023-09-12:
klar, hier muss ich auch endlich mal weitermachen 😊

azservice690, 2023-11-18:
Hallo zusammen, weis jemand ob es momentan eine Lösung gibt fur ein Modul 
WT32-ETH01 und das Funkmodul CMT2300A.
Ich habe grosse Interesse daran, da ich kein W-Lan in diese stelle habe.

lumapu, 2023-11-19:
sollte doch gehen, ich habs selbst nicht probiert. Aber rein theoretisch braucht man zwei SPI Schnittstellen, die hat der ESP32 mit HSPI und VSPI. 
Mit dem NRF geht es ja auch, und dieses Radio benötigt auch einen SPI Bus.
-> spricht nichts dagegen.

geronet1, 2023-11-19:
Habs jetzt nicht näher angeschaut, aber wenn CMT2003A und das NRF jeweils SPI haben warum nicht beide an einen Bus hängen mit zwei Slave_Select?

lumapu, 2023-11-19:
Ich glaube jeweils die libs beansruchen einen kompletten SPI, aber prinzipiell hast du natürlich recht. Selbst habe ich es noch nie ausprobiert.

hans341, 2023-11-19:
Hab grad festgestellt, daß das Update per Ethernet (!) von z.B. v0.8.7 auf v0.8.8 auch nicht nicht klappt. 
Gibt's da ne Chance? Das wäre echt super.

geronet1, 2023-11-26:
Ich habs jetzt auch hingekriegt nach dem Vorgehen von hans341, soweit funktioniert auch alles. Super Arbeit!

tjcooker, 2023-11-27:
Hallo,
die Ursache o.g. Reboot-Endlosschleife ist die Verwendung des unititialisierten Objekts `tmp` vom Typ `cfgSys_t` in der Funktion `settings::loadDefaults()` in der Datei `src/config/settings.h`. Dabei kann dann z.B. der Schalter `schedReboot` auf `true` gesetzt sein und es kommt zu dem Endlos-Reboot. Als Abhilfe kann (die unnötige) Zeile 371 (`memcpy(&mCfg.sys, &tmp, sizeof(cfgSys_t));`) auskommentiert werden, so dass `tmp` garnicht verwendet wird.
Bitte weitermachen mit der guten Arbeit an diesem tollen Projekt.
TJ

geronet1, 2023-11-28:
Ich hab das mal reingetippt: https://github.com/lumapu/ahoy/pull/1240/commits/8cccca1724879c59d866e8c5c4e5b79f16492c4c

Einen Bug hab ich noch: Bei "disable night communication" errechnet er den Sonnenaufgang von morgen, obwohl heute noch die Sonne scheint und startet deshalb nicht die Kommunikation. Beim testen später hat er dann einen Sonnenaufgang von 2120 angezeigt.. ich schau mal ob ich was finde.

geronet1, 2023-11-29:
Den "disable night communication" Bug hab ich nun auch gefunden: https://github.com/lumapu/ahoy/pull/1240

Fehler war daß er das NTP Paket sendet und mit "return true" im ETHERNET-Code sofort die Sonnenzeiten berechnet, obwohl keine Uhrzeit (0) vorhanden war. Folglich wartete er dann bis zum Sonnenaufgang am nächsten Tag.. deshalb muss er erst das NTP Paket empfangen und darf dann die Uhrzeiten berechnen.

hans341, 2023-11-29:
Das sieht ja aus als ob es mit grossen Schritten vorwärts geht. Sehr schön. Super Einsatz!
Leider muss ich mich bzgl. Github, Programmierung & Co. als kompletter Laie outen:
Bedeutet "ich hab's gefunden" bzw. "ich hab das reingetippt" erst mal noch nicht dass ich das jetzt anwenden kann?
Das muss jetzt ja sicher erst noch in einem Build eingebaut werden oder?

geronet1, 2023-12-08:
Ist schon drin: https://github.com/lumapu/ahoy/pull/1240

lumapu, 2023-12-08:
einfach hier downloaden: https://fw.ahoydtu.de
(gerade ist noch die dev `0.8.14` aktuell) 😀

hans341, 2023-12-09:
Hi, 
also ich weiss leider immer noch nicht genau, was ihr gefixt habt und was es in 0.8.14 gechaffft hat, aber ein Update 
von 0.8.14 auf 0.8.15
oder
von 0.8.15 auf 0.8.16
per Ethernet (!) kriege ich immer noch nicht hin. Es ist nach wie vor so, daß das Update nach wenigen Sekunden abbricht und ich Ahoy wieder auf die Update-Seite zurückfällt.

xFvPx, 2023-12-10:
Moin,
ich habe jetzt auch die 0.8.14 eingespielt.
Danke an geronet1. Keine Endlos reboot. "Night communication" sieht auch gut aus. 

Wie hans341 geschrieben hat funktioniert das Update auch bei mir leider nicht.
Auch das erstellen/editieren des Inverter funktioniert bei mir nicht. Ich habe dies mit der WLAN Variante erstellt und anschließend die Eth Variante drüber geflasht.
[image]

geronet1, 2023-12-10:
Editieren der Inverter und die Fehlermeldung beim Zeitsyncronisieren liegt irgendwo an der Verwendung der lib AsyncWebServer_ESP32_W5500 und der Ajax-Kommunikation, ich bin noch nicht dahinter gekommen warum das so ist.
Mit der 0.8.15 funktioniert nichtmal die Kommunikation mit dem WR, egal ob Wifi oder ETH Version.

lumapu, 2023-12-10:
Das liegt irgendwie an den Webserver, der den POST request als GET request interpretiert und dadurch dass falsche JSON zurück gibt.
NRF ist bis auf Ethernet gefixt

tjcooker, 2023-12-11:
Sorry, falls ich mich zu dumm beim Umgang mit Github anstelle, aber [hier](https://github.com/lumapu/ahoy/blob/development03/src/config/settings.h) ist der Bug mit dem unitialiserten Objekt `tmp`, welchen ich oben beschrieben habe, immer noch vorhanden, obwohl die Datei vor 2 Tagen geändert wurde.
In [diesem Patch](https://github.com/lumapu/ahoy/commit/8cccca1724879c59d866e8c5c4e5b79f16492c4c) von geronet1 ist der Fehler aber anscheinend behoben.
Also sorry nochmal, falls ich Github nicht verstehe, aber ich dachte, es gibt einen Branch im Repository in welchem alle Fixes gesammelt werden. Wo finde ich diesen?

geronet1, 2023-12-11:
Den config Bug hab ich jetzt anders gelöst, da waren bei nicht vorhandener Konfiguration die IP-Adressen nicht Null.
Ich finde den POST-Fehler nicht, in der lib erkennt er aber daß es ein POST Request ist..
Komischerweise geht ja "Save" mit den Einstellungen auch, das ist ebenso ein POST Request.

tjcooker, 2023-12-12:
Vielen Dank für die Aufklärung und entschuldige bitte meine Hartnäckigkeit: Wie komme ich an den neuesten Quellcode, welcher [diesen Patch](https://github.com/lumapu/ahoy/commit/8cccca1724879c59d866e8c5c4e5b79f16492c4c) enthält? Wenn ich den Code mittels `git clone -b development03 https://github.com/lumapu/ahoy/` auschecke, dann besteht weiterhin das Problem mit uninitialisierten Anfanswerten im Objekt `tmp` vom Typ `cfgSys_t`. Die fehlende Initialisierung ist natürlich auch der Grund, warum die IP-Adressen zufällige Werte enthalten.

[46 more replies]