lumapu/ahoy #887: Keine Verbindung zur NRF24l01+

Calle-Fly, 2023-04-16:
Hallo,
Ich bekomme leider keine Verbindung zur NRF24l01+ 
Kann es sein das ich nicht die + Version habe?
Ich habe alles so eingestellt und verkabelt wie auf den Bildern zu sehen.

## Hardware
    * [ ] ESP32 Rev 1 und ESP-WROOM-32
Modelname: ESP32-D0WDQ5 und 

# nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Power Stabilization:
no

### Version / Git SHA: 
Version: 0.6.0

[image]
[image]
[image]
[image]
[image]
[image]

Die Verdrahtung wie auf den Bild
USB Netzteil mit 1,5A

**ESP32 Rev1**
ESP32 Entwicklung Bord WiFi + Bluetooth Ultra-Low Power Verbrauch Dual Core ESP-32 ESP-32S ESP 32 Ähnliche ESP8266

Beschreibung:
Esp32 ist bereits integrierte Antenne und HF-Balun, Leistungs verstärker, rausch arme Verstärker, Filter und Power-Management-Modul. Die gesamte Lösung nimmt die geringste Menge an Leiterplatte fläche ein. Dieses Board wird mit 2,4 GHz Dual-Mode-Wi-Fi-und Bluetooth-Chips von tsmc 40nm Low-Power-Technologie, Leistung und HF-Eigenschaften am besten verwendet, was sicher und zuverlässig ist. und skalierbar für eine Vielzahl von Anwendungen.
Hohes Leistungs-Preis-Verhältnis
Kleines Volumen, leicht zu anderen Produkten eingebettet
Starke Funktion mit Unterstützung lwip Protokoll, Freertos
Unterstützt drei Modi: ap, sta und ap + sta
Unterstützung lua programm, leicht zu entwickeln
Entwicklung Bord: https://github.com/Nicholas3388/LuaNode
Klicken Sie auf den Link für weitere Details: http://espressif.com/en/products/hardware/esp32-devkitc/overview
https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md

ESP WROOM 32

Esp32 ESP-32 Development Board WLAN WLAN Bluetooth Dual Core CP2102 Filter Leistungs modul 2,4 GHz HF für Arduino NodemcuBeschreibungEsp32 ist bereits integrierte Antenne und HF-Balun, Leistungs verstärker, rausch arme Verstärker, Filter und Power-Management-Modul. Die gesamte Lösung nimmt die geringste Menge an Leiterplatte fläche ein.Dieses Board wird mit 2,4 GHz Dual-Mode-Wi-Fi-und Bluetooth-Chips von tsmc 40nm Low-Power-Technologie, Leistung und HF-Eigenschaften am besten verwendet, was sicher und zuverlässig ist. und skalierbar für eine Vielzahl von Anwendungen.Eigenschaften:Hohes Leistungs-Preis-VerhältnisKleines Volumen, leicht zu anderen Produkten eingebettetStarke Funktion mit Unterstützung lwip Protokoll, FreertosUnterstützt drei Modi: ap, sta und ap + staUnterstützung lua programm, leicht zu entwickeln

Nrf24l01 + Drahtlose Daten Übertragung Modul 2,4g/Die Nrf24l01 Upgrade-Version

Nrf24l01 + Drahtlose Daten Übertragung Modul 2,4g/Die Nrf24l01 Upgrade-VersionBeschreibungRF24L01 ist eine arbeit, in der 2,4-2,5 GHz weltweit ISM band single-chip-transceiver, wireless transceiver, darunter: die frequenz generator die verbesserte SchockBurstTM modus controller power verstärker kristall verstärker modulator-demodulator ausgang power kanal auswahl und protokoll set durch die spi-schnittstelle, um einen sehr niedrigen strom verbrauch, Senken strom verbrauch modus 12,3 mA Power-down modus und standby-modus, wenn in übertragen modus emission power 6dBm wenn strom verbrauch ist 9,0 mA akzeptanz modell.Ball zu open-ISM-band maximale 0dBm übertragen power, lizenz-ausgenommenes verwenden. Die open 100 meter!Unterstützung sechs kanäle von daten empfang(1) niedrige betriebs spannung: 1,9 ~~ 3,6 V low-spannung(2) high-rate: 2Mbps, luft übertragung zeit ist sehr kurz, stark reduziert die drahtlose übertragung von kollision phänomene (software einstellungen 1Mbps oder 2Mbps luft übertragung rate)(3)Multi-frequenz punkte: 125 frequenz punkte, zu erfüllen die multi-punkt kommunikation und frequenz hopping kommunikation bedürfnisse(4)ultra-kompakte: gebaut-in 2,4 GHz antenne kompakte 15x29mm (inklusive antenne)(5) Low power verbrauch: wenn in antwort modus kommunikation, schnelle luft übertragung und start-up zeit stark reduziert die strom verbrauch.(6) niedrigen anwendung kosten: NRF24L01 integriert alle RF protocol hohe-speed signal verarbeitung abschnitt, wie: automatisch senden verlorenen paketen und automatisch erzeugen eine bestätigen signal, Etc. die nRF24L01 Die spi-schnittstelle können nehmen vorteil der mikrocontroller hardware SPI port oder mikrocontroller I / O port zu simulieren die interne FIFO kann verwendet werden mit eine vielzahl von low-speed mikroprozessor interface, einfach zu bedienen low-kosten mikrocontroller.(7) erleichtern die entwicklung: die link schicht ist voll integrierte in die modul, sehr einfach zu entwickeln. Automatische verbreitung funktion, automatische erkennung und verbreitung von verloren pakete, die weiterverbreitung zeit und die anzahl der verbreitung können werden software-gesteuert automatische lagerung ist nicht erhalten die paket der antwort signal auto-antwort, nach dem empfang einer gültig daten, Das modul automatisch sendet die antwort festen frequenz erkennung signale ohne zusätzliche programmierung Carrier Detect-gebaut-in hardware CRC fehler erkennung und punkt-zu-multipoint kommunikation adresse steuer paket übertragung fehler zähler und träger erkennung Kann verwendet werden für frequenz hopping set kann auch set up sechs zu erhalten die kanal adresse, können haben eine wahl von open erhalten kanal standard die pin Dip 2,54 MM abstand schnittstellen für embedded Anwendungen.

mroenne2022, 2023-04-16:
Hallo @Calle-Fly  , leg mal IRQ auf GPIO 1 / G1 . Bei mir lief solch einer mal so im Ahoy , ich hab ihn dann mal neu geflast für OpenDTU, dann allerdings IRQ auf GPIO 16. Ich schlage vor zum Test mal den IRQ umzulegen.  
[image]

mroenne2022, 2023-04-16:
[image]

Hier das Schaubild zur Umsetzung GPIO auf den Port am ESP32

Calle-Fly, 2023-04-17:
Danke **mroenne2022** habe ich versucht, leider auch keine Verbindung.

Muss ich einen Wechselrichter haben um die Verbindung zur Nrf24l01 +zu sehen?

mroenne2022, 2023-04-17:
@Calle-Fly nein, den Wechselrichter brauchst du da überhaupt noch nicht. Das muss vorher alles perfekt laufen, erst danach macht es Sinn den WR einzutragen. Schade das es nicht so läuft bei dir,  verstehe ich nicht so ganz. Hast sicher die Verkabelung UND die Einstellung für den IRQ im Menü geändert? Wie gesagt, bei mir lief es so und jetzt umgebaut auf OpenDTU  -  läuft auch 🤷🏻

Calle-Fly, 2023-04-19:
Danke mroenne2022,
habe mal OpenDTU versucht das gleiche Problem.

Habe dann nochmal alle Verbindungen überprüft, eine Ader hatte einen Internen Bruch (G18) habe diese erneuert jetzt ist alles gut.

closed 2023-04-19 by Calle-Fly