lumapu/ahoy #888: Make day/night switch configurable

JRossbach, 2023-04-16:
It’s a nice feature to pause polling of the inverter during night. However - at least for my coordinates and eben due to zing offset of 60 minutes - this I quite early and Leads to miss few minutes/hours of power provided by the sun. 

IMHO keeping the „pause“ flag is satisfying but should/could be extended either by:
* more flexible offset
* Manual day/night switch time
* Flag that could be set via mqtt to indicate it‘s day/night

## Hardware
  * [ 
  * X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.0
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2023-04-17:
60 minutes are not enough ?

Where do you live ?

JRossbach, 2023-04-17:
Hi @knickohr, I'm living next to Frankfurt in the centre of Hessen state.

When configuring the night pause, the polling stops at approx. 5:45PM, while still having sun on the balcony.

knickohr, 2023-04-17:
25 Minuten reiche aus. Zumindest für wahnsinnige 150km Entfernung etwas weiter sücllich 😂

closed 2023-04-17 by lumapu