lumapu/ahoy #889: Using MQTT to set limit or restart inverter only works for default topic "inverter" and not for longer topics

anorak-47, 2023-04-17:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

12a3d50

### Build & Flash Method

ESP Tools (flash)

### Setup

MQTT brocker was added.
Topic is "holog/client/pv/iv/hoy"

### Debug Serial Log output

```shell
09:23:02 I: MQTT got topic: holog/client/pv/iv/hoy/ctrl/limit/0
09:23:02 I: json: {"val":75,"path":"client","cmd":"pv","id":0}
```

### Error description

Setting limit or restarting inverter using MQTT does not work if using a TOPIC prefix path longer than one element.

Affected commands:
<TOPIC>/ctrl/limit/<INVERTER_ID>
<TOPIC>/ctrl/limit/<INVERTER_ID>

My topic is holog/client/pv/iv/hoy. 
Publishing "holog/client/pv/iv/hoy/ctrl/limit/0 75" results in the following JSON output (I compiled my own version with this debug output enabled): 

09:23:02 I: MQTT got topic: holog/client/pv/iv/hoy/ctrl/limit/0
09:23:02 I: json: {"val":75,"path":"client","cmd":"pv","id":0}

As you can see, the "cmd" is wrong.
Code seems to work only with the default topic "inverter".

anorak-47, 2023-04-17:
Quick fix in pubMqtt.h:
Change line 

const char *p = topic;

to

const char *p = topic + strlen(mCfgMqtt->topic);

juepi, 2023-06-16:
Seems this issue is still unfixed in the current development release (0.7.4), does the mentioned quick fix have any side effects?

mslookup, 2023-06-24:
I closed my Bug Report #1004, because the contents are nearly the same.

I can add the information that indeed other topics than the default topic "inverter" are working, but as soon as you start nesting the prefix, it stops working.

I don't know whether @anorak-47 solution is also working for me, but i will try it out (but will need some time, never compiled something by myself...).

closed 2023-08-05 by lumapu