lumapu/ahoy #891: Inverter wird nicht gefunden oder Meldung: is available but no data was received until now

stiech82, 2023-04-18:
## Hardware
  * [ ] ESP8266
  * [* ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ *] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.0
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich habe jetzt auf AhoyDTU 0.6.0 umgestellt und plötzlich wird entweder der Inverter nicht gefunden oder es kommt die Meldung das keine Daten empfangen werden.

Egal welche von den zwei Meldungen. Ich kann den Wechselrichter über "Serial/Control" steuern, allerdings bekomme ich keine Daten zurück.

Mit den vorherigen Versionen von AhoyDTU hat es super funktioniert und die Einstellungen wurden 1:1 übernommen.

Meine Einstellungen:
Pinout
CS -GPIO5
CE -GPIO4
IRQ -GPIO16
SCLK -GPIO18
MOSI -GPIO23
MISO -GPIO19
LED0 -off / default
LED1 -off / default

Radio (NRF24L01+)
Hier ein Ausschnitt aus der Console:
12:27:44 I: resetPayload: id: 0
12:27:44 I: (#0) enqueCommand: 0x0b
12:27:44 I: (#0) prepareDevInformCmd 0x0b
12:27:44 I: TX 27B Ch3 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 a0 00 00 00 00 00 00 00 00 41 e1 c6 
12:27:44 W: (#0) Frame 2 missing: Request Retransmit
12:27:44 I: TX 11B Ch23 | 15 83 13 07 71 85 76 67 00 82 e5 
12:27:45 I: resetPayload: id: 0
12:27:45 I: (#0) enqueCommand: 0x0b
12:27:45 I: (#0) prepareDevInformCmd 0x0b
12:27:45 I: TX 27B Ch40 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 a1 00 00 00 00 00 00 00 00 d1 ec 5a 
12:27:45 W: (#0) Frame 1 missing: Request Retransmit
12:27:45 I: TX 11B Ch61 | 15 83 13 07 71 85 76 67 00 81 e6 
12:27:46 I: resetPayload: id: 0
12:27:46 I: (#0) enqueCommand: 0x0b
12:27:46 I: (#0) prepareDevInformCmd 0x0b
12:27:46 I: TX 27B Ch75 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 a2 00 00 00 00 00 00 00 00 21 f8 bd 
12:27:46 W: (#0) Frame 2 missing: Request Retransmit
12:27:46 I: TX 11B Ch3 | 15 83 13 07 71 85 76 67 00 82 e5 
12:27:47 I: resetPayload: id: 0
12:27:47 I: (#0) enqueCommand: 0x0b
12:27:47 I: (#0) prepareDevInformCmd 0x0b
12:27:47 I: TX 27B Ch23 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 a3 00 00 00 00 00 00 00 00 b1 f5 21 
12:27:47 W: (#0) Frame 1 missing: Request Retransmit
12:27:47 I: TX 11B Ch40 | 15 83 13 07 71 85 76 67 00 81 e6 
12:27:49 I: resetPayload: id: 0
12:27:49 I: (#0) enqueCommand: 0x0b
12:27:49 I: (#0) prepareDevInformCmd 0x0b
12:27:49 I: TX 27B Ch61 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 a5 00 00 00 00 00 00 00 00 11 de ac 
12:27:49 W: (#0) Frame 2 missing: Request Retransmit
12:27:49 I: TX 11B Ch75 | 15 83 13 07 71 85 76 67 00 82 e5 
12:27:50 I: resetPayload: id: 0
12:27:50 I: (#0) enqueCommand: 0x0b
12:27:50 I: (#0) prepareDevInformCmd 0x0b
12:27:50 I: TX 27B Ch3 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 a6 00 00 00 00 00 00 00 00 e1 ca 4b 
12:27:50 W: (#0) Frame 1 missing: Request Retransmit
12:27:50 I: TX 11B Ch23 | 15 83 13 07 71 85 76 67 00 81 e6 
12:27:51 I: resetPayload: id: 0
12:27:51 I: (#0) enqueCommand: 0x0b
12:27:51 I: (#0) prepareDevInformCmd 0x0b
12:27:51 I: TX 27B Ch40 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 a7 00 00 00 00 00 00 00 00 71 c7 d7 
12:27:51 W: (#0) Frame 2 missing: Request Retransmit
12:27:51 I: TX 11B Ch61 | 15 83 13 07 71 85 76 67 00 82 e5 
12:27:52 I: resetPayload: id: 0
12:27:52 I: (#0) enqueCommand: 0x0b
12:27:52 I: (#0) prepareDevInformCmd 0x0b
12:27:52 I: TX 27B Ch75 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 a8 00 00 00 00 00 00 00 00 81 86 69 
12:27:52 W: (#0) Frame 1 missing: Request Retransmit
12:27:52 I: TX 11B Ch3 | 15 83 13 07 71 85 76 67 00 81 e6 
12:27:54 I: resetPayload: id: 0
12:27:54 I: (#0) enqueCommand: 0x0b
12:27:54 I: (#0) prepareDevInformCmd 0x0b
12:27:54 I: TX 27B Ch23 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 aa 00 00 00 00 00 00 00 00 e1 9f 12 
12:27:54 W: (#0) Frame 2 missing: Request Retransmit
12:27:54 I: TX 11B Ch40 | 15 83 13 07 71 85 76 67 00 82 e5 
12:27:55 I: resetPayload: id: 0
12:27:55 I: (#0) enqueCommand: 0x0b
12:27:55 I: (#0) prepareDevInformCmd 0x0b
12:27:55 I: TX 27B Ch61 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 ab 00 00 00 00 00 00 00 00 71 92 8e 
12:27:55 W: (#0) Frame 1 missing: Request Retransmit
12:27:55 I: TX 11B Ch75 | 15 83 13 07 71 85 76 67 00 81 e6 
12:27:56 I: resetPayload: id: 0
12:27:56 I: (#0) enqueCommand: 0x0b
12:27:56 I: (#0) prepareDevInformCmd 0x0b
12:27:56 I: TX 27B Ch3 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3e 70 ac 00 00 00 00 00 00 00 00 41 b4 9f 
12:27:56 W: (#0) Frame 2 missing: Request Retransmit
12:27:56 I: TX 11B Ch23 | 15 83 13 07 71 85 76 67 00 82 e5

tqma1, 2023-04-18:
Versuche doch mal die Development Version 0.6.7

lumapu, 2023-04-18:
und leider auch prüfen, ob evtl. versehentlich durch die Umstellung die LEDs auf GPIO0 stehen, falls ja einfach wieder auf `OFF` stellen und speichern

stiech82, 2023-04-19:
> und leider auch prüfen, ob evtl. versehentlich durch die Umstellung die LEDs auf GPIO0 stehen, falls ja einfach wieder auf `OFF` stellen und speichern

Wie bereits erwähnt sind das meine Einstellungen:

CS -GPIO5
CE -GPIO4
IRQ -GPIO16
SCLK -GPIO18
MOSI -GPIO23
MISO -GPIO19
LED0 -off / default
LED1 -off / default

Ich habe das direkt copy/paste so rauskopiert. Die Einstellungen sind definitiv so.

stiech82, 2023-04-19:
> Versuche doch mal die Development Version 0.6.7

Tatsächlich funktioniert es mit dieser Version. Zumindest besser als mit 0.6.0. Danke

stiech82, 2023-04-19:
Leider funktioniert meine Nulleinspeisung nicht mehr.

Braucht es andere Befehle als bei v0.5.66?
Ich schicke per Node-Red den Befehl an Hoymiles/ctrl/limit_persistent_absolute/0

In der Console von Ahoy-DTU erscheint das wenn der Befehl von Node-Red über MQTT kommt:
07:26:30 I: (#0) resetPayload
07:26:30 I: (#0) enqueCommand: 0x0b
07:26:30 I: (#0) prepareDevInformCmd 0x0b
07:26:30 15 pid: 80
07:26:30 I: TX 27B Ch61 | 15 83 13 07 71 85 76 67 00 80 0b 00 64 3f 7b 86 00 00 00 63 00 00 00 00 31 e2 fa 
07:26:30 W: (#0) Frame 2 missing: Request Retransmit
07:26:30 15 pid: 82
07:26:30 I: TX 11B Ch75 | 15 83 13 07 71 85 76 67 00 82 e5 
07:26:30 W: (#0) Frame 2 missing: Request Retransmit
07:26:30 15 pid: 82
07:26:30 I: TX 11B Ch3 | 15 83 13 07 71 85 76 67 00 82 e5 
07:26:30 W: (#0) Frame 3 missing: Request Retransmit
07:26:30 15 pid: 83
07:26:30 I: TX 11B Ch23 | 15 83 13 07 71 85 76 67 00 83 e4 
07:26:31 W: (#0) Frame 3 missing: Request Retransmit
07:26:31 15 pid: 83
07:26:31 I: TX 11B Ch40 | 15 83 13 07 71 85 76 67 00 83 e4 
07:26:31 W: (#0) Frame 4 missing: Request Retransmit
07:26:31 15 pid: 84
07:26:31 I: TX 11B Ch61 | 15 83 13 07 71 85 76 67 00 84 e3 
07:26:35 I: (#0) resetPayload

beegee3, 2023-04-19:
ja, die Limit Anweisungen haben sich geändert, s. [User_Manual](https://github.com/lumapu/ahoy/blob/main/User_Manual.md)

closed 2023-07-23 by lumapu