lumapu/ahoy #892: [Bug]

knickohr, 2023-04-19:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.7

### Github Hash

12a3d50

### Build & Flash Method

ESP Tools (flash)

### Setup

Kann ich so nicht beschreiben, es handelt sich um das PCB von Markus, also der ESP32S3. Tut aber nichts zur Apache, tritt bei allen ESPs auf.

### Error description

Betrifft nur DTUs mit mehreren Invertern !

Beim Start der WRs werden die YieldDay Werte der einzelnen inverter offenbar erst dann genullt. Anschließend werden die Werte aller WRs zusammen gerechnet als einen einzigen Wert YieldDay total über MQTT übertragen.

Soweit ja alles OK. Wenn jetzt aber ein oder mehr Inverter noch nicht aufgewacht ist, dann werden offenbar alte Werte die sich noch in irgendeiner Variable befinden ebenfallsaufsummiert und übertragen.

Die Werte der einzelnen Inverter müssen vor der Übertragung genullt werden, egal ob die Inverter bereits aufgewacht sind oder noch schlafen.

[image]

beegee3, 2023-04-19:
das ist merkwürdig. Bei der Nullung um Mitternacht werden die Variablen mit den YieldDay Werten bei **allen** Invertern auf 0 gesetzt (Schleife in `app::tickMidnight`). Und genau diese Variablen werden für den YieldDay Total aufsummiert (`sendIvData` in `pupMqtt.h`). YieldDay Werte von **vor** Mitternacht sollten nirgendwo mehr gespeichert sein, also auch nicht aufaddiert werden können. Kannst du anhand der YieldDay Werte der einzelnen Inverter **vor** Mitternacht nachvollziehen, das deren Summe (bis auf den ersten aufgewachten) das ergibt, was du morgens beim YieldDay Total beobachtet hast?
Oder spielt dir der erste aufgewachte Inverter einen Streich und meldet (fälschlicherweise) diesen Wert? Denn beim Aufwachen wird von Ahoy aktiv nichts genullt, sondern nur das rausgegeben, was der Inverter meldet.

beegee3, 2023-04-19:
@lumapu oder könnte `doCalculations` in `hmInverter.h` der Übeltäter sein? Das rechnet doch den Inverter YieldDay Wert aus den Panel YieldDay Werten zusammen. Und die wurden nicht genullt. Es ist schwierig nachzuvollziehen, wann diese Funktion aufgerufen wird. Bin mir nicht sicher, ob das immer erst passiert, wenn ein Inverter Daten liefert und damit auch aktuelle Panel Werte vorliegen. Evtl. wäre es das einfachste (und sicherste), in der `zeroYieldDay` Funktion die YieldDay Werte **aller** Channels zu nullen. Wenn ich mir die Funktion ansehe, scheint das auch so geplant zu sein (Schleife über die Channel). Allerdings läuft die Schleife zu kurz (< statt <=) und in der Schleife steht `CH0` statt `ch`.
Modifiziert wäre es dann:
```
        void zeroYieldDay(Inverter<> *iv) {
            DPRINTLN(DBG_DEBUG, F("zeroYieldDay"));
            record_t<> *rec = iv->getRecordStruct(RealTimeRunData_Debug);
            uint8_t pos;
            for(uint8_t ch = 0; ch <= iv->channels; ch++) {
                pos = iv->getPosByChFld(ch, FLD_YD, rec);
                iv->setValue(pos, rec, 0.0f);
            }
        }
```

knickohr, 2023-04-19:
Ich teste heute Nacht mal genauer. Hab eben den Flow angepaßt um zu sehen wo die Werte herkommen, welcher Inverter der Übeltäter ist.

Was ich bis jetzt sehe, das die Wert genau minus dem Yield des ersten aufgewachten Inverters ist.

lumapu, 2023-04-19:
@knickohr: Erinnerung

lumapu, 2023-04-19:
@beegee3 ich glaube du hast mal wieder ins schwarze getroffen. Sieht für mich auch stark nach dem Fehler aus. @knickohr: ich veröffentliche gleich mal eine neue Version mit dem Fix.

knickohr, 2023-04-19:
Wieso finde ich nur immer so eine Sch… ? 😅🤪

knickohr, 2023-04-21:
Rückmeldung :  Hat heute Morgen alles prima funktioniert 👍

commit b6531180 (2023-04-19): 0.6.8
* fix #892 `zeroYieldDay` loop was not applied to all channels