lumapu/ahoy #893: MQTT empfängt keine Daten [Bug]

linus2k3, 2023-04-19:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.0

### Github Hash

9c7d7a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

[image]
[image]

### Debug Serial Log output

```shell
connect to network 'root' ...
I: added inverter HM114181867150
I: resetPayload: id: 0
I: resetPayload: id: 1
I: resetPayload: id: 2
I: resetPayload: id: 3
I: resetPayload: id: 4
I: resetPayload: id: 5
I: resetPayload: id: 6
I: resetPayload: id: 7
I: resetPayload: id: 8
I: resetPayload: id: 9
I: (#0) resetPayload
I: (#1) resetPayload
I: (#2) resetPayload
I: (#3) resetPayload
I: (#4) resetPayload
I: (#5) resetPayload
I: (#6) resetPayload
I: (#7) resetPayload
I: (#8) resetPayload
I: (#9) resetPayload
scanning APs with SSID root
reconnect in 8 seconds
reconnect in 7 seconds
reconnect in 6 seconds
[... 174 lines cut ...]
I: (#0) prepareDevInformCmd 0x01
I: TX 27B Ch40 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 5C B6 00 00 00 00 00                                                                                                                                         00 00 00 57 52 0F
I: (#0) nothing received
I: resetPayload: id: 0
I: (#0) prepareDevInformCmd 0x0B
I: TX 27B Ch61 | 15 81 86 71 50 81 21 42 94 80 0B 00 64 40 5C C0 00 00 00 00 00                                                                                                                                         00 00 00 3F 16 5F
I: (#0) nothing received
I: resetPayload: id: 0
I: (#0) enqueCommand: 0x01
I: (#0) enqueCommand: 0x0B
I: (#0) prepareDevInformCmd 0x01
I: TX 27B Ch75 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 5C CA 00 00 00 00 00 00 00 00 95 62 81
I: (#0) nothing received
```

### Error description

[image]

Es spielt keine Rolle, ob der HM-700 in Reichweite ist oder nicht. Er empfängt keine MQTT Nachrichten.

MQTT String:
[image]

Problem besteht heute, seit Update auf 0.6.0 über Web Installer

lumapu, 2023-04-19:
habe gerade die `0.6.7` released, bitte nochmal damit prüfen. Geht auch wieder über den Webinstaller. Und dann braucht man natürlich erst wieder ☀
Prüfe bitte mal die LED Einstellungen unter der Rubrik System, die sollten auf `OFF` stehen.

linus2k3, 2023-04-19:
Gerade installiert, gleiches Verhalten. 

[image]

lumapu, 2023-04-19:
der Titel ist falsch oder, es werden allgemein keine Daten empfangen.
Pinout sieht gut aus, hängt dein Inverter am Akku?

linus2k3, 2023-04-19:
Es werden keine Daten vom NodeRed (Venus OS) empfangen. Nodered kann aber auf die Daten vom Inverter empfangen und ins VRM Portal schicken. Ja der Inverter hängt am Akku, war bis heute kein Problem.

linus2k3, 2023-04-19:
Im Live wurden sie vorhin angezeigt. Jetzt liegt der ESP8266 neben mir. Auf dem Weg zum Gartenhaus ist mittlerweile ein See vom Regen... Wie meinst du das mit dem Gerätenamen? Eigentlich müsste ja über Topic und ID alles eindeutig sein?

lumapu, 2023-04-19:
Ja in früheren Versionen hat Ahoy automatisch (ab Werk) immer `Ahoy-DTU` oder sehr ähnlich geheißen. Ab `0.6.0` heißt Ahoy dann zB. `Ahoy-DTU-A5-12-B1`. Das wurde eingebaut, da es häufiger vorkommt, dass 2 oder mehr AhoyDTU in einem Netzwerk sind und damit dann zB. ein gesetztes Limit von allen AhoyDTUs verarbeitet wurde - da sich alle angesprochen fühlten. Jetzt ist jede DTU Einzigartig vom Namen

lumapu, 2023-04-19:
> Im Live wurden sie vorhin angezeigt. Jetzt liegt der ESP8266 neben mir. Auf dem Weg zum Gartenhaus ist mittlerweile ein See vom Regen... Wie meinst du das mit dem Gerätenamen? Eigentlich müsste ja über Topic und ID alles eindeutig sein?

auch ohne Wechselrichter published Ahoy beim Start zB. die Version und die Heap-Fragmentation (beim ESP8266). Wenn das geht, werden natürlich auch die Live-Daten übertragen.

linus2k3, 2023-04-19:
00:03:42 I: (#0) enqueued cmd failed/timeout
00:03:42 I: (#0) resetPayload
00:03:42 I: (#0) Requesting Inv SN 114181867150
00:03:42 I: (#0) enqueCommand: 0x01
00:03:42 I: (#0) enqueCommand: 0x0b
00:03:42 I: (#0) prepareDevInformCmd 0x01
00:03:42 15 pid: 80
00:03:42 I: TX 27B Ch40 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 65 3e 00 00 00 00 00 00 00 00 81 84 be 
00:03:42 I: (#0) nothing received
00:03:52 I: (#0) enqueued cmd failed/timeout
00:03:52 I: (#0) resetPayload
00:03:52 I: (#0) Requesting Inv SN 114181867150
00:03:52 I: (#0) prepareDevInformCmd 0x0b
00:03:52 15 pid: 80
00:03:52 I: TX 27B Ch61 | 15 81 86 71 50 81 21 42 94 80 0b 00 64 40 65 48 00 00 00 00 00 00 00 00 e9 c0 ee 
00:03:52 I: (#0) nothing received
00:04:02 I: (#0) enqueued cmd failed/timeout
00:04:02 I: (#0) resetPayload
00:04:02 I: (#0) Requesting Inv SN 114181867150
00:04:02 I: (#0) enqueCommand: 0x01
00:04:02 I: (#0) enqueCommand: 0x0b
00:04:02 I: (#0) prepareDevInformCmd 0x01
00:04:02 15 pid: 80
00:04:02 I: TX 27B Ch75 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 65 52 00 00 00 00 00 00 00 00 83 79 2d 
00:04:02 I: (#0) nothing received
00:04:12 I: (#0) enqueued cmd failed/timeout
00:04:12 I: (#0) resetPayload
00:04:12 I: (#0) Requesting Inv SN 114181867150
00:04:12 I: (#0) prepareDevInformCmd 0x0b
00:04:12 15 pid: 80
00:04:12 I: TX 27B Ch3 | 15 81 86 71 50 81 21 42 94 80 0b 00 64 40 65 5c 00 00 00 00 00 00 00 00 e9 3f 05 
00:04:12 I: (#0) nothing received
00:04:22 I: (#0) enqueued cmd failed/timeout
00:04:22 I: (#0) resetPayload
00:04:22 I: (#0) Requesting Inv SN 114181867150
00:04:22 I: (#0) enqueCommand: 0x01
00:04:22 I: (#0) enqueCommand: 0x0b
00:04:22 I: (#0) prepareDevInformCmd 0x01
00:04:22 15 pid: 80
00:04:22 I: TX 27B Ch23 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 65 66 00 00 00 00 00 00 00 00 42 1f be 
00:04:22 I: (#0) nothing received
00:04:32 I: (#0) enqueued cmd failed/timeout
00:04:32 I: (#0) resetPayload
00:04:32 I: (#0) Requesting Inv SN 114181867150
00:04:32 I: (#0) prepareDevInformCmd 0x0b
00:04:32 15 pid: 80
00:04:32 I: TX 27B Ch40 | 15 81 86 71 50 81 21 42 94 80 0b 00 64 40 65 70 00 00 00 00 00 00 00 00 28 f3 24 
00:04:32 I: (#0) nothing received
00:04:42 I: (#0) enqueued cmd failed/timeout
00:04:42 I: (#0) resetPayload
00:04:42 I: (#0) Requesting Inv SN 114181867150
00:04:42 I: (#0) enqueCommand: 0x01
00:04:42 I: (#0) enqueCommand: 0x0b
00:04:42 I: (#0) prepareDevInformCmd 0x01
00:04:42 15 pid: 80
00:04:42 I: TX 27B Ch61 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 65 7a 00 00 00 00 00 00 00 00 82 87 fa 
00:04:42 I: (#0) nothing received
00:04:52 I: (#0) enqueued cmd failed/timeout
00:04:52 I: (#0) resetPayload
00:04:52 I: (#0) Requesting Inv SN 114181867150
00:04:52 I: (#0) prepareDevInformCmd 0x0b
00:04:52 15 pid: 80
00:04:52 I: TX 27B Ch75 | 15 81 86 71 50 81 21 42 94 80 0b 00 64 40 65 84 00 00 00 00 00 00 00 00 ec c5 22 
00:04:52 I: (#0) nothing received
00:05:02 I: (#0) enqueued cmd failed/timeout
00:05:02 I: (#0) resetPayload
00:05:02 I: (#0) Requesting Inv SN 114181867150
00:05:02 I: (#0) enqueCommand: 0x01
00:05:02 I: (#0) enqueCommand: 0x0b
00:05:02 I: (#0) prepareDevInformCmd 0x01
00:05:02 15 pid: 80
00:05:02 I: TX 27B Ch3 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 65 8e 00 00 00 00 00 00 00 00 46 b1 fc 
00:05:02 I: (#0) nothing received
00:05:12 I: (#0) enqueued cmd failed/timeout
00:05:12 I: (#0) resetPayload
00:05:12 I: (#0) Requesting Inv SN 114181867150
00:05:12 I: (#0) prepareDevInformCmd 0x0b
00:05:12 15 pid: 80
00:05:12 I: TX 27B Ch23 | 15 81 86 71 50 81 21 42 94 80 0b 00 64 40 65 98 00 00 00 00 00 00 00 00 2c 5d 66 
00:05:12 I: (#0) nothing received
00:05:22 I: (#0) enqueued cmd failed/timeout
00:05:22 I: (#0) resetPayload
00:05:22 I: (#0) Requesting Inv SN 114181867150
00:05:22 I: (#0) enqueCommand: 0x01
00:05:22 I: (#0) enqueCommand: 0x0b
00:05:22 I: (#0) prepareDevInformCmd 0x01
00:05:22 15 pid: 80
00:05:22 I: TX 27B Ch40 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 65 a2 00 00 00 00 00 00 00 00 87 7d dd 
00:05:22 I: (#0) nothing received
00:05:32 I: (#0) enqueued cmd failed/timeout
00:05:32 I: (#0) resetPayload
00:05:32 I: (#0) Requesting Inv SN 114181867150
00:05:32 I: (#0) prepareDevInformCmd 0x0b
00:05:32 15 pid: 80
00:05:32 I: TX 27B Ch61 | 15 81 86 71 50 81 21 42 94 80 0b 00 64 40 65 ac 00 00 00 00 00 00 00 00 ed 3b f5 
00:05:32 I: (#0) nothing received
00:05:42 I: (#0) enqueued cmd failed/timeout
00:05:42 I: (#0) resetPayload
00:05:42 I: (#0) Requesting Inv SN 114181867150
00:05:42 I: (#0) enqueCommand: 0x01
00:05:42 I: (#0) enqueCommand: 0x0b
00:05:42 I: (#0) prepareDevInformCmd 0x01
00:05:42 15 pid: 80
00:05:42 I: TX 27B Ch75 | 15 81 86 71 50 81 21 42 94 80 01 00 64 40 65 b6 00 00 00 00 00 00 00 00 87 82 36 
00:05:42 I: (#0) nothing received

Okay, ich ändere mal den Namen.

linus2k3, 2023-04-19:
Okay, Name ist eindeutig... 

[image]
[image]

evtl. hier noch der Fehler?!

linus2k3, 2023-04-20:
Was noch zu sagen ist: Mit 0.5.66 hat alles funktioniert. Die einzige Änderung, die ich jetzt vorgenommen hab, das W beim Limit anzuhängen.

beegee3, 2023-04-20:
1. wie @lumapu bereits gesagt hat: seit Version `0.6.0` wird immer ein Teil der MAC-Adresse angehängt (die letzten 3. Hex-Werte) Unabhängig von dem Namen, den du vergibst. In deinem Fall also **Gartenhaus-DTU-_XXYYZZ_**. Die Werte von **_XXYYZZ_** kannst du deiner MAC Angabe in der Systemübersicht entnehmen.
2. laut deinen Logs antwortet dein Inverter nicht ("nothing received"), entsprechend auch die WebUI Anzeige "Inverter #0: HM-700 (v0) is not yet available". Solange keine Verbindung zum Inverter besteht, können keine Werte übertragen und kann auch kein Limit gesetzt werden.
3. das Topic für das Limit setzen ist falsch: statt der Seriennummer ist die ID des Inverters anzugeben. Das ist die Nummer, die in der Übersicht ist ("Inverter #_x_", da werden die Inverter beginnend mit 0 einfach hochgezählt). In deinem Fall lautet das Topic also **inverter/ctrl/limit/0** (s. auch das Beispiel im [User Manual](https://github.com/lumapu/ahoy/blob/main/User_Manual.md) )

Harald-U, 2023-04-20:
@linus2k3: Was Du in diesem [Screenshot](https://github.com/lumapu/ahoy/issues/893#issuecomment-1515452563 ) zeigst, ist die Definition Deines MQTT-Brokers in NodeRED, die hat mit Ahoy-DTU nichts zu tun. Der IP-Netzwerkname Deiner DTU ist für MQTT auch uninteressant. Interessanter ist die Definition der MQTT-In-Node selbst. Zum Beispiel:

Meine MQTT-Definition in Ahoy:
[image]

Und meine MQTT-In-Node in NodeRED:
[image]

Ich habe mein Basis-Topic "solar" genannt, mein Inverter ist "HM-400" und mit dieser Definition bekomme ich alle Topics aller Ebenen darunter.

linus2k3, 2023-04-25:
Danke Beegee3! Ich war n …. :( läuft, außer inverter turn off/on. Hatte noch keine Zeit den MQTT Request zu prüfen. Aktuell läufts aufm ESP8266 recht instabil, aber ich kann wieder regeln, danke nochmal!

Oskar70, 2023-07-08:
Habe eine funktioniernde AHOY-DTU (0.5.66) (ESP8266 Wemos D1, NFR24 mit Antenne). Habe eine 2.Ahoy-DTU nach selbem Muster mit gleichen Teilen aber mit 0.6.9 gebaut und bekomme auch wie bei linus2k3 pid: 80, nothing received. Nach dem Lesen diverser Hinweise habe ich folgende Verbindungsvarianten probiert:
a) D3 -- IRQ und D4 -- CE, so auch in System Config der AHOY Oberfläche
b) D4 -- IRQ und D3 -- CE, so auch in System Config der AHOY Oberfläche
c) D4 -- IRQ und D1 -- CE, so auch in System Config der AHOY Oberfläche
d) D1 -- IRQ und D4 -- CE, so auch inSystem Config der AHOY Oberfläche
Nur  im Fall c) war die LED vom Wemos die ganze Zeit an (ist auch bei meiner funktionierended AHOY-DTU so). Bei keiner Version wurden Daten empfangen wurden und es kam immer:
14:58:01 I: (#0) resetPayload
14:58:01 I: (#0) enqueCommand: 0x01
14:58:01 I: (#0) enqueCommand: 0x0b
14:58:01 I: (#0) prepareDevInformCmd 0x01
14:58:01 15 pid: 80
14:58:01 I: TX 27B Ch61 | 15 82 93 76 66 84 92 15 92 80 01 00 64 a9 79 79 00 00 00 00 00 00 00 00 87 2b 65 
14:58:01 I: (#0) nothing received
.
.
.
Ich habe eine 3. AHOY_DTU gebaut, und damit die selben Probleme. 
Ich habe einen Hinweis gefunden, die Bauteile auf dem NRF24 abzuschirmen. Gemacht, nichts gebracht. 
Ich habe mir in der Zwischenzeit 2 ESP32 Wroom Boards (38 Pin's) bestellt und hoffe, dass es damit funktioniert. Habe mir leider keinen NRF24L01+ ohne Antenne bestellt und hoffe, dass dies nicht das eigentliche Problem ist. Downgrade auf 0.5.66 hat auch nicht geholfen.

Hat jemand noch einen Tipp was ich noch probieren könnte?

lumapu, 2023-07-23:
klingt nach einem NRF Problem für mich. Was sagt die Oberfläche von Ahoy auf der index Seite? Wird das NRF Modul erkannt?
Es gab mal eine Zeit, da waren sehr viele defekte / fehlerhafte NRF Module auf dem Markt.

Oskar70, 2023-07-24:
Danke für die Antwort. Habe, nachdem ich neue ESP's gekauft hatte (an denen es nicht lag, neue NRF's bei Makershop gekauft und nun geht's.VGWolfgang--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

closed 2023-07-24 by lumapu

golamic, 2023-07-29:
hallo,

ich habe eine Ahoydtu aufgebaut mit ESP8266 mit Antenne und Display,   Vers. 6.9... und einen HM1500 mit 4 Modulen.
Alles funktioniert bestens, über das Diplay kann ich ohne weitere Himfsmittel direkt die aktuelle Leistung, den Tages- und den Gesamtertrag ablesen. Auch über Mqtt  mit entsprechender Software kann ich mir von überall alle relevanten Daten anzeigen lassen.
NUR... von Zeit zu Zeit ... bekomme ich keine Anzeige wenn ich direkt auf die DTU via Fritzbox zugreifen möchte. Da steht dann...
"every second the values are updated"  und es ist nicht möglich sich z.B einzuloggen. Aber hin und wieder, ohne was zu ändern!  bekomme ich eine ordentliche Anzeige der Daten und habe auch vollen Zugriff...
Meinem Nachbar geht es genauso... mal ja mal nein, nur hat der keine LCD Anzeige.... daran kann es also nicht liegen.
Was ist da los?

lumapu, 2023-07-30:
Bitte einen neuen issue aufmachen, dieser hier ist bereits geschlossen.

Zum Problem: was steht unter /system im Feld heap fragmentation?

Habt ihr einen Kondensator verbaut und das PowerLevel auf min oder low stehen?

lumapu, 2023-07-31:
stimmt, Kondensator ist einfach eine Standardfrage von uns 😉

golamic, 2023-07-31:
mein neuer Issue    ...   #1053--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.