lumapu/ahoy #894: Wrong values when inverter sleeps

Gangang61, 2023-04-21:
Github Hash: 9c7d7a0

When solar power is switched off, inverter go to sleep immediately. There is no chance to send new values. DTU will continue publishing last received values, which of course are wrong. Could that behavior be changed?

lumapu, 2023-04-21:
can you describe it more detailed? Maybe in German because we're 99% German here.

Why does the inverter switch off? No sun or manually?
Did you tried to use the settings which are provided by Ahoy to reset values?

bobyzipper, 2023-04-21:
I can try to explain, as I had the same "issue". Gangang61 have system with battery between solars and inverter. In normal system solar to inverter, inverter status is changing between 0-1-2 consequently and all values (incl. channel values are updated acordingly). In systems with battery, when it is empty system shuts the "sun" immediately off and inverter is switching from status 2 to status 0 without to pass trough status 1. On this way DTU is not sending channels data updates and MQTT broker keeps last values till inverter is retuned back online. This means all night system reports full production despite of that the inverter is offline.  
@Gangang61 I found a very simple solution for me, maybe it will work for you also. In Home Assistant make an automation script which reset all power values when inverter status is going to 0, like that:

alias: Ahoy DTU Values reset
description: reset some values after comm is down
trigger:
  - platform: numeric_state
    entity_id: sensor.inverter_status
    below: 1
condition: []
action:
  - service: mqtt.publish
    data:
      topic: inverter/Hoym600/ch2/P_DC
      payload: "0"
      retain: false
      qos: 0
  - service: mqtt.publish
    data:
      topic: inverter/Hoym600/ch1/P_DC
      payload: "0"
      retain: false
      qos: 0
  - service: mqtt.publish
    data:
      topic: inverter/Hoym600/ch1/I_DC
      payload: "0"
      qos: 0
      retain: false
  - service: mqtt.publish
    data:
      topic: inverter/Hoym600/ch2/I_DC
      payload: "0"
      qos: 0
      retain: false
  - service: mqtt.publish
    data:
      topic: inverter/Hoym600/ch0/P_AC
      payload: "0"
      qos: 0
      retain: false
mode: single

@lumapu  Bitte deine Projekt nicht unterschätzen. Es ist schon weltweit beliebt :-)

Gangang61, 2023-04-21:
@bobyzipper:
You are completely right, I operate the HM600 out of a battery. As I don´t use HomeAssistant your proposal doesn´t work for me.
I will find a work around in my system, I just need more time to investigate. My system is running just for 2 days now :-)

knickohr, 2023-04-21:
A Solar inverter is NOT a battery inverter 😉

So ist doesn‘t work correctly as it is NOT designed for it.

if you use MQTT with HA or Node-Red, it’s easy to reset this values manually when the inverter is going down.

closed 2023-07-23 by lumapu