lumapu/ahoy #895: [Bug] Reset values and YieldDay at midnight

TobiScience, 2023-04-21:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nothing

### Error description

The daily data (total and both inverters get a reset at 0:00 in the interface) but only the total data get an mqtt update at 0:00

https://i.imgur.com/tAeuITn.png (the red circle)

settings:
Reset values and YieldDay at midnight [x]
Reset values when inverter polling pauses at sunset [ ]
Reset values when inverter status is 'not available' [ ]

thanks for the great project

lumapu, 2023-04-25:
Does it mean that both of the inverters are publishing a value different from zero at midnight? When is their last update?

TobiScience, 2023-04-25:
No there is no mqtt update for the inverters between nighttime. 

For example yesterday the last mqtt update was around 20:45 from the inverter and the first update comes when the wake up 6:35. Only the total gets a reset and mqtt publish at midnight. 

I think the inverters should get an update at midnight since ahoy resets both inverters and total at midnight.

ottelo9, 2023-05-13:
Exactly the same problem here. I discussed this problem in Discord already.

lumapu, 2023-05-16:
I'm sorry, I didn't get the problem. From the implementation side the inverter data should become zero from the inverter itself because of slowly reduced light in the evening. The yield is one of the fields which doesn't decrease while turning off and need a manual reset it you need this value day by day.
What would be the expected behavior compared to the implementation?

I'm also fine if you answer in German.

TobiScience, 2023-05-16:
Hey,
um 24 Uhr werden in der Oberfläche die Tageswerte auf 0 gesetzt.
Gleichzeitig wird dies auch per Mqtt verschickt, dort allerdings nur Total, die mqtt für die Inverter kommt erst morgens. 

Besser wäre wenn um 24 Uhr für Total und alle Inverter ein Mqtt mit 0 kWh verschickt wird. Sonst sieht das so komisch aus wie auf dem Screenshot aus Grafana.

lumapu, 2023-08-05:
bitte nochmal mit der aktuellen Release `0.7.25` testen, da dieser Issue etwas aus dem Fokus gerückt ist. Danke.

TobiScience, 2023-08-07:
Soweit ich das sehe funktioniert das mit dem Night reset, :)

allerdings ist jetzt Total, also das Summieren von mehrern Invertern nicht mehr korrekt :/

[image]

lumapu, 2023-08-07:
man müsste quasi eine Art 'timeout' festlegen, ab wann YieldDay Werte verschickt werden dürfen, damit würde man auch auf den x-ten inverter warten.

TobiScience, 2023-08-07:
Müsste man nicht eigentlich nur verhindern das total per mqtt verschickt wird wenn ein Inverter noch auf 0 steht, weil eigentlich kann ein Invertier ja nicht auf 0 stehen wenn er läuft. Der Zähler geht ja nur rauf,

lumapu, 2023-08-08:
guter Hinweis!

lumapu, 2024-08-16:
maybe resolved in the meantime, check the new Release 0.8.140 😉

closed 2024-08-16 by lumapu