lumapu/ahoy #898: MQTT is not connected

Duesendaniel, 2023-04-22:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT Broker hinzugefügt

### Error description

'MQTT is not connected' tritt immer weider auf, diesmal nach 1 Tag uptime nach update auf 0.6.9. Keine Fehlermeldung in der Console. Im  ioBroker nur ' Client connection closed: timeout'. Nach reboot läuft es wieder. Vorher hatte ich die 0.6.0 und 0.6.4 auf ESP32  D1 mini laufen und nach 4 - 5 Tagen dann das Problem.

teletapi, 2023-04-22:
Hab ein ähnliches Problem FW 6.8 und 6.9 ESP32 nach dem abendlichen abschalten der DTU und am nächsten tag wieder einschalten "kein mqtt connect" nach mehrfachem Neustart geht es dann wieder aber wieder nur bis zum nächsten ein/ausschalten dann wieder das gleiche Spiel

Duesendaniel, 2023-04-22:
Das Problem hatte ich schon bei der 0.5.8X mit ESP 32 und ESP 8266. Mal ist es nach 1 Tag uptime oder 3-5 Tages uptime passiert. Nach Reboot oder MPR war dann sofort für die nächsten Tage alles ok.

teletapi, 2023-04-22:
Das passiert nicht solange es online ist, das passiert nur wenn ich der DTU den Stecker ziehe und morgens wieder einschalte dann kommt einfach kein mqtt. ich muss dann jenachdem 2-4 mal neu strom drauf geben irgend wann ist mqtt wieder connected und bleibt es auch den ganzen tag über

Duesendaniel, 2023-04-23:
Wenige Stunden nach dem gestrigen Update auf 0.6.9 war die Verbindung wieder weg.

StefanTR96, 2023-04-25:
Ebenfalls das selbe Problem seit dem Update auf 0.6.9

Jumperm, 2023-04-25:
Vielleicht bringt es etwas Licht ins dunkle: ich hab die Fritzbox als lokalen NTP.
8266er mit 100uF. Sauber geflashed mit esphome flasher 
https://github.com/esphome/esphome-flasher

derTillus, 2023-04-26:
Ich habe das oben genannte Problem seit 0.5.66 und es ist leider mit dem Update auf 0.6.9 unverändert.

Jumperm, 2023-05-02:
Besteht mit 0.6.12 leider immer noch.

exo1987, 2023-07-02:
Seitdem ich "Pause polling inverters during night" deaktiviert habe, bleibt MQTT verbunden.

lumapu, 2023-07-03:
sehr interessant, kann aber leider nicht erklären warum das scheinbar zusammenhängt

Jumperm, 2023-07-05:
Ich habe auf 7.6. gewechselt und nur den mqtt herausgenommen. Läuft seit einem Tag mit RestAPI ohne Einschränkung.

lumapu, 2023-08-05:
bitte das neue Release `0.725` testen und Feedback geben, danke.

closed 2024-01-02 by lumapu