lumapu/ahoy #899: MQTT - HomeAssistant - und dann?!

KoLSMS, 2023-04-22:
Hallo Gemeinde,

Stück für Stück komme ich voran, mein großes "Ziel": Die Auswertung meines BKW mit Grafana :)

AhoyDTU läuft stabil, sendet per MQTT Daten an mein Homeassistent, genau jetzt hänge ich.

Die AhoyDTU wird mir in Homeassistent angezeigt, weiter komme ich nicht, daher die Frage: Wie geht es nun weiter?!

Ich habe auf Homeassistent bereits InfluxDB installiert, NodeRed, wenn ich alles richtig gelesen habe, schaufelt mir NodeRed die Daten von MQTT in die InfluxDB, nur wie?!

Ich bin dankbar für jeden Tipp, vielen Dank.

Josy68, 2023-04-22:
Hallo! Du bist hier falsch. Frag die Home Assistent Community Lg

rmayergfx, 2023-04-24:
@KoLSMS 
Auch wenn du hier im falschen Forum bezüglich dieser Fragen unterwegs bist, hier ein hilfreicher Link:
https://daniel-strohbach.de/home-assistant-influxdb-grafana-fuer-n00bs

nexulm, 2023-04-24:
Könnte dann so aussehen. ;-)
[image]
---
Bei mir sieht es derzeit in der HA configuration.yaml
#--- InfluxDB ---
influxdb:
#    host: localhost
    host: <IP>
    port: <PORT>
    database: <database_name>
    username: <database_username>
    password: <database_password>
    max_retries: 3
    default_measurement: state
    include:
       domains:
        - sensor

-----
Zusätzlich musst du dann das Grafana AddOn installieren und ein wenig Zeit mit der Konfiguration eines Dashboards verbringen.
-----
Ich habe mittlerweile meine HA-Installation (auf einem Raspi 4) entlastet und die influxDB + Grafana auf einen externen Server (hier: Synology NAS) umgezogen.

Viel Erfolg mit deinem BK und der Config. ;-)

KoLSMS, 2023-04-25:
Wow, so "schwer" war das ganze ja dann doch nicht, vielen Dank @rmayergfx @nexulm für die Hinweise...

Nachdem ich alle Entitäten in der InfluxDB hatte, hat mich das dann doch gestört irgendwie :-)

Daher schreibe ich jetzt nur noch die Entitäten von der AhoyDTU in die InfluxDB, das ganze sieht dann in der configuration.yaml so aus:

#--- InfluxDB ---
influxdb:
    host: localhost
    port: 8086
    database: DeinDatenbankname
    username: DeinBenutzername
    password: DeinPasswort
    max_retries: 3
    default_measurement: state
    include:
       entities:
        - sensor.hoymiles_alarm_mes_id
        - sensor.hoymiles_ch1_irradiation
        - sensor.hoymiles_ch1_i_dc
        - sensor.hoymiles_ch1_p_dc
        - sensor.hoymiles_ch1_u_dc
        - sensor.hoymiles_ch1_yieldday
        - sensor.hoymiles_ch1_yieldtotal
        - sensor.hoymiles_ch2_irradiation
        - sensor.hoymiles_ch2_i_dc
        - sensor.hoymiles_ch2_p_dc
        - sensor.hoymiles_ch2_u_dc
        - sensor.hoymiles_ch2_yieldday
        - sensor.hoymiles_ch2_yieldtotal
        - sensor.hoymiles_efficiency
        - sensor.hoymiles_f_ac
        - sensor.hoymiles_i_ac
        - sensor.hoymiles_pf_ac
        - sensor.hoymiles_p_ac
        - sensor.hoymiles_p_dc
        - sensor.hoymiles_q_ac
        - sensor.hoymiles_temp
        - sensor.hoymiles_u_ac
        - sensor.hoymiles_yieldday
        - sensor.hoymiles_yieldtotal

Als nächstes, sobald ich wieder mehr Zeit habe, knöpfe ich mir Grafana vor :-)

Danke nochmals!

PS: Vielleicht kann das ja jemand gebrauchen was ich gepostet habe für die configuration.yaml ?!

Gruß...

closed 2023-04-25 by KoLSMS