lumapu/ahoy #900: [Bug]

ncqgm, 2023-04-22:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

was already installed

### Setup

- admin password set for all pages except "live"

### Error description

- downloaded release 0.6.9
- installed via OTA
- reboot, connect to local wlan as station
- set admin password for all pages except "live"
- wrote down password locally (so, should not be the issue)
- saved settings,reboot

I can now
- view "live" page
- pull up "login" page via menu
- pull up "login" page manually via /login

However, when I enter the password (or an empty password as suggested in some
other issues) and click on the [Login] button "nothing happens". Nothing means that
the login page is reloaded with the password field cleared out and the UI will not let
me access any other pages.

Any chance to reset the password without re-flashing ? Via serial console or REST API ?

Thanks,
Karsten

ncqgm, 2023-04-22:
Sorry for not adding a bug title:

"Login page not functional on 0.6.9/esp32"

ncqgm, 2023-04-23:
I have been able to erase-reflash my device so it got
un-partially-bricked which takes away the immediate problem.

However, I suppose, the issue still stands: the "Login" page does not log in
(at least for me, tested on Firefox Linux , Firefix Windows, Firefox ESR, Chromium, Bing).

Thanks,
Karsten

lumapu, 2023-04-25:
never heard about before. From my sight it's not possible to brick an ESP.
Did reflash help or not?

lumapu, 2023-04-28:
have you checked the "erase" checkbox during installation. Maybe there is some corrupt JSON-settings file on your ESP which leads to the login issue

ncqgm, 2023-04-29:
Ich hatte mich wohl etwas unklar ausgedrückt: nach dem neuflashen (mit erase) ist wieder alles "normal", sowohl unter 0.6.0 als auch 0.6.9. Ich komme auf alle Seiten der Weboberfläche. Allerdings habe ich bislang den Passwortschutz der Seiten nicht wieder aktiviert. Insofern läuft alles wieder normal -- abgesehen davon, daß es auch bei mir Probleme mit der instabilen Wifi-Verbindung gibt (auch ohne mqtt).

lumapu, 2023-05-03:
das heißt den Issue hier können wir schließen und diskutieren in #901 bzgl. Wifi weiter?

ncqgm, 2023-05-04:
Schwierig zu sagen: nach dem Flashen habe ich wieder Zugriff auf alle Seiten (weil durch das Flashen der Seitenschutz
entfernt wurde, wozu ich überhaupt flashen wollte, nachdem ich das Login-Problem bekommen hatte). Seither habe ich
den Seitenschutz nicht erneut aktiviert, sodaß ich nicht sagen kann, ob das (login-)Problem noch besteht. Dessen Symptome waren ja (s.o.), daß man die login-Seite aufrufen konnte, diese aber nicht zu einer Authentifizierung führte. Ich werde mal versuchen, eine einzelne Seite zu schützen (sodaß ich weiter ohne Passwort auf die Setupseite zum wieder ausschalten kann) und berichte später darüber. Es wurde damals auch kein Logout-Link im Menü angezeigt, auch wenn ich (nach augenscheinlich nicht funktionierendem Login) parallel in einem anderen Tab die Ahoy-Oberfläche erneut aufrief.

ncqgm, 2023-05-06:
Ich habe getestet: Nur eine Seite als zu schützen markiert. Das Problem trat nicht erneut auf. Initial war die geschützte Seite nicht im Menü. Ich konnte mich einloggen, danach konnte ich die geschützte Seite wieder übers Menu aufrufen. Das Logout-Link war dann auch da und ausloggen ging. Danach war die geschützte Seite auch nicht mehr erreichbar.

Man könnte wohl das Issue vorerst schließen.

closed 2023-05-06 by ncqgm