lumapu/ahoy #904: FeatureRequest: HMS Support + Parallele Nutzung von NRF24L01+ und CMT2300A

MiniOh, 2023-04-23:
Hallo, 

ich weiß, is gibt bereits einige Themen dazu.
Dennoch hier die Frage ob es auch mit Ahoy, ähnlich wie beim OpenDTU Projekt möglich sein wird, auf einem ESP32 die beiden Antennen NRF24L01+ und CMT2300A parallel zu betreiben?

Bei mir ist noch die Beschaffung eines Inverters offen, und ich würde gerne einen HMS 1800 nutzen, sollte in Zukunft die Nutzung einer DTU mit beiden Antennen möglich sein.

Danke schon mal vorab, für die hoffentlich positive Rückmeldung.

Beste Grüße

lumapu, 2023-04-25:
ja genau, das ist schön integriert, schaue einfach später nach einem Branch `HMS`

MiniOh, 2023-04-25:
Oh, super.
Wusste gar nicht dass das schon soweit ist.

Da ich eh nur ESP32 nutze, sollte das ja für mich passen.

Gibts eine PIN Belegung für die zweite Antenne?

lumapu, 2023-04-25:
ja ist eine Testversion: https://github.com/lumapu/ahoy/tree/hms

MiniOh, 2023-04-25:
Sorry für dei wahrscheinlich blöde Frage.
Wie komme ich in der HMS Branch zu einer dem .bin File?
Unter Actions gibts die Branch nicht, um das fertge File herunterzuladen, oder?

MiniOh, 2023-04-25:
> ja ist eine Testversion: https://github.com/lumapu/ahoy/tree/hms
> 
> Pinbelegung ist relativ frei wählbar, ich habe hier leider keine Erfahrung und auch noch nicht getestet, ob beides gleichzeitig geht.

Das Testen kann ich gerne übernehmen, sobald die Antenne ankommt.
Der Inverter sollte diese Woche noch kommen.

lumapu, 2023-04-25:
hier ist die bin gebaut aus dem aktuellen Stand:
Getestet habe ich es jetzt über Wochen hinweg mit einem HMS 2000 4T, ein weiterer Test mit 3x HMS 2000 4T wurde auch schon gemacht.

[230425_ahoy_0.6.10_HMS_esp32.bin.zip](https://github.com/lumapu/ahoy/files/11326812/230425_ahoy_0.6.10_HMS_esp32.bin.zip)

MiniOh, 2023-04-26:
Super.
Besten Dank.
Dann werde ich auch mal etwas damit spielen.

Gibts die Antennen auch irgendwo in Deutschland zu kaufen, oder kommen die aktuell alle noch aus China?

lumapu, 2023-04-26:
meinst du Funkmodule oder Antennen?

MiniOh, 2023-04-26:
Funkmodule, sorry.

lumapu, 2023-04-27:
melde dich beim Kontakt auf Ahoydtu.de, ich hab was da

MiniOh, 2023-04-28:
> hier ist die bin gebaut aus dem aktuellen Stand: Getestet habe ich es jetzt über Wochen hinweg mit einem HMS 2000 4T, ein weiterer Test mit 3x HMS 2000 4T wurde auch schon gemacht. [...]

Hallo, 

ich hatte heute mal die 0.6.10 HMS getestet.
Grundconfig war ok. Ich habe dann den NRF24L01+ aktivert.
Danach habe ich in der Config CMT2300A aktivert (ohne dass ein Funkmodul verkabelt ist)
Anschließend war die DTU nicht mehr per IP erreichbar.

Auch seriell erhalte ich folgende Meldug:
"Serial port is not ready. Close any other application using it and try again."

Ne Idee?

MiniOh, 2023-04-28:
OK, ich kann es noch etwas detailierter eingrenzen.

Ich habe nochmal mit einem ESP32 DEVKIT1 geteset.

Grundkonfig OK. 

NRF24 verkabelt, CMT2300A noch nicht vorhanden.

Test 1:
Nur NRF24 aktiviert und Pins konfiguriert.
>>Reboot, alles OK, **Meldung NRF24 is connected.**

Test 2:
NRF24 aktiviert und Pins konfiguriert und zusätzlich das Häkchen bei "CMT2300A Enable"
Pins des CMT2300A auf **OFF / default**
>>Reboot, DTU nicht mehr erreichbar, auch nicht seriell. Nur über Halten des Boot Buttons wieder flashbar.

Test 3:
NRF24 aktiviert und Pins konfiguriert und zusätzlich das Häkchen bei "CMT2300A Enable"
Pins des CMT2300A auf **12, 13, 14 (beliebig)**
>> Reboot OK, DTU noch erreichbar, allerdings ist mit zusätzlich aktivem CMT2300A, die NRF24 nicht mehr ereichbar.
Meldung "your NRF24 module can't be reached, check the wiring and pinout"

Hilft das evtl. bei der weiteren Entwicklung der HMS Branch?
Oder ist generall kein Parallelbetrieb von HM und HMS Invertern geplant?

Danke schon mal vorab.

mc-0815, 2023-05-10:
Hi, 

vermutlich für euch eine dumme Frage, aber wo finde ich eine Wiring Schaubild für eine ESP32 oder ESP8266 und dem Funkmodul CMT2300A? Habe heute einen HMS-1600 und möchte gern loslegen.

Vielen Dank,

lumapu, 2023-05-10:
Bei mir hängt das Modul am VSPI (SCK und MOSI)

Daraus ergibt sich als feste Pins, ESP zu CMT:
D23 - SDIO
D18 - SCK

und frei wählbar:
D17 - CSB 
D26 - FCSB
D34 - GPIO3

lumapu, 2023-05-10:
> Gibts bereits eine Version nach der 0.6.10 für HMS?
> Kannst du nochmal die .bin bereitstellen?

Ja kann ich, allerdings hat sich da bisher nicht wirklich was verändert.

lumapu, 2023-05-10:
> Oder ist generall kein Parallelbetrieb von HM und HMS Invertern geplant?

Doch ist schon geplant, genau dieses Setup habe ich hier auch an Wechselrichtern. Plan ist Mi, HM und HMS zu unterstützen.

mc-0815, 2023-05-11:
> Bei mir hängt das Modul am VSPI (SCK und MOSI)
> 
> Daraus ergibt sich als feste Pins, ESP zu CMT: D23 - SDIO D18 - SCK
> 
> und frei wählbar: D17 - CSB D26 - FCSB D34 - GPIO3

Vielen Dank.

MiniOh, 2023-05-11:
> > Oder ist generall kein Parallelbetrieb von HM und HMS Invertern geplant?
> 
> Doch ist schon geplant, genau dieses Setup habe ich hier auch an Wechselrichtern. Plan ist Mi, HM und HMS zu unterstützen.

Guten Morgen, 

vielen Dank für die Info und die tolle Hilfe.
Wie auch schon per Mail kommuniziert, funktioniert bereits mit der 0.6.10 HMS, der Parallelbetrieb von HM und HMS.

[image]

lumapu, 2023-05-11:
> Wie auch schon per Mail kommuniziert, funktioniert bereits mit der 0.6.10 HMS, der Parallelbetrieb von HM und HMS.

würdest du dafür mit uns noch dein Pinout teilen?

MiniOh, 2023-05-11:
> > Wie auch schon per Mail kommuniziert, funktioniert bereits mit der 0.6.10 HMS, der Parallelbetrieb von HM und HMS.
> 
> würdest du dafür mit uns noch dein Pinout teilen?

Hallo, ja, natürlich.

Der CMT ist auf dein Board gelötet, der NRF manuell au die Pins des ESP32 gelötet.

[image]

Argafal, 2023-05-12:
Hier noch ein pull request, der die gleichzeitige Nutzung von NRF und CMT auch auf dem Fusion v2 Board (und anderen S3/C3/S2 basierten Boards) ermoeglicht: https://github.com/lumapu/ahoy/pull/935

Fusion v2 pinout Einstellungen:

[image]

commandermatz, 2023-05-13:
> hier ist die bin gebaut aus dem aktuellen Stand: Getestet habe ich es jetzt über Wochen hinweg mit einem HMS 2000 4T, ein weiterer Test mit 3x HMS 2000 4T wurde auch schon gemacht. [...]

Diese Version läuft nicht auf einem 8266?

MiniOh, 2023-05-13:
Nein

MiniOh, 2023-05-26:
Nur mal zur Info.
Bei mit läuft die HMS Version 0.6.10 seit 2 Wochen fehlerfrei mit 6 HM und 2 HMS Invertern.

@lumapu 
Gibts bereits eine neuere .bin?

LeoWinterDE, 2023-05-26:
Nur zum Verständnis, sind die HMS und HMT (#233) von der Kommunikation her gleich?

lumapu, 2023-05-26:
@LeoWinterDE Ja beide verwenden ein 868MHz Protokoll und benötigen den Funkchip CMT2300A

@MiniOh sehr cool, freut mich dass es so gut läuft.

hier ist die Version, die auf den neuesten Sourcen basiert `0.6.15`:
[230527_ahoy_esp32_hms_hmt_0.6.15.bin.zip](https://github.com/lumapu/ahoy/files/11580498/230527_ahoy_esp32_hms_hmt_0.6.15.bin.zip)

MiniOh, 2023-05-30:
@lumapu 

Es gibt doch noch etwas, was mit dem HMS Invertern nicht ganz passt.
Einer der beiden HMS, liefert abends beim "Offline gehen" Zero Yield Day Werte, wodurch dann auch die Total Yield Day nicht mehr passt.

Hast du eine Idee an was das liegen könnte?

[image]

Es ist der "letzte" Inverter der Offline ging, welcher dann 0 bei Yield Day liefert, falls das hilft.

[image]

Olli

lumapu, 2023-05-30:
ich denke es liegt an wechselnden Lichtverhältnissen in der Dämmerung. Der Wechselrichter ist schon mal aus und hat alles vergessen, dann bekommt er nochmal genug Saft um noch eine Nachricht zu schicken, leider durch den vorherigen reset mit falschen Informationen.
Wir könnten eine Unterdrückung von diesen Nachrichten einbauen, sie wären leicht erkennbar.

MiniOh, 2023-05-31:
Ich weiß was du meinst.
Das ist bei mir bisher nur einmal passiert bei extrem dunkler Bewölkung vor der Dämmerung.
Aber keinesfalls unter "normalen" Verhältnissen.

In den letzten Monaten kam das nie vor.
Lediglich jetzt an 3 aufeinanderfolgenden Tagen, und auch nur mit den HMS.

MiniOh, 2023-05-31:
@lumapu 

Hey, 

heute ist das nicht passiert. Ich bin testweise aber wieder zurück zur HMS 0.6.10
Die letzten beiden Tage, also das Verhalten auftrat, war die HMS 0.6.15 installiert.
Zufall, oder hat sich da etwas diesbezüglich verändert?

Beste Grüße

lumapu, 2023-06-01:
@MiniOh mich wundert, dass dein NRF Pinout geht, da du HSPI und VSPI gemischt hast. Hier mein Vorschlag:
[image]

MiniOh, 2023-06-02:
Danke für den Hinweis.
Bisher klappt das einwandfrei, zumindest mit der 0.6.10.

jho59, 2023-07-14:
> Bei mir hängt das Modul am VSPI (SCK und MOSI)
> 
> Daraus ergibt sich als feste Pins, ESP zu CMT: D23 - SDIO D18 - SCK
> 
> und frei wählbar: D17 - CSB D26 - FCSB D34 - GPIO3

Ich bin etwas überrascht ...
Da der ESP32 sowohl Daten zum Funkmodul senden als auch von dort empfangen muss, müssten doch sowohl MOSI als auch MISO verbunden sein? Das ist bei Verwendung eines CMT2300A aber nicht der Fall. Wie bekommt der ESP32 in dem Fall Daten? Und was ist mit einem SPI-ChipSelect? Wenn der CMT2300A an VSPI angeschlossen ist, wäre das der GPIO5 ...
GPIO34 fungiert in dem Fall als IRQ? 

Grüße, Jürgen

schluss123, 2023-07-20:
Hi Lumapu,
Wäre es möglich, alle PinOUT Konfigurationen frei definierbar zu machen? Auf meiner PCB sind die festen Pins, ESP zu CMT:
D23 - SDIO und D18 - SCK als direkte Leiterbahn zum NRF24L01+ ausgestaltet,  OpenDTU läuft damit, Ahoy leider nicht mehr. Ich kann die GPIOs 18 und 23 NICHT zum CMT2300A routen.
Danke.

lumapu, 2023-07-22:
@jho59 

> Wie bekommt der ESP32 in dem Fall Daten? 

Der Mosi Pin wird ganz besonders eingestellt in diesem Fall. Das CMT Modul verwendet den Pin bidirektional, war für uns auch neu und eine gewisse Zeit ein Blocker während der Entwicklung.

lumapu, 2023-07-22:
@schluss123 werde ich mir anschauen. Hast du einen ESP32 mit 30 oder 38 Pins?

schluss123, 2023-07-22:
> @schluss123 werde ich mir anschauen. Hast du einen ESP32 mit 30 oder 38 Pins?

@lumapu : Ich nutze meistens die 38poligen ESPs, gelegentlich auch die 30poligen im Evo-Board. 38 bevorzugt. Danke Dir.

Jakobi88, 2023-07-27:
@lumapu Ähnliches Problem bei mir - SCK liegt auf dem HSPI GPIO 12 und SDIO liegt ebenfalls auf dem HSPI GPIO 14. Dito 38 Pin ESP32.

lumapu, 2023-07-27:
@schluss123 @Jakobi88 macht bitte einen neuen Issue dafür auf - so geht die Info verloren, da dieser Issue beim nächsten Release zugemacht wird. (wegen `fixed dev`)

schluss123, 2023-07-28:
added at #1046

closed 2023-08-05 by lumapu

Agerter, 2023-08-17:
Hallo welche Antenne kann ich für den CMT2300A Funkchip verwenden geht da nur diese große Antenne oder reicht es auch ein kleine Antenne mit IPX Anschluss und einem Kabel zu verwenden

jho59, 2023-08-17:
@Agerter:
Zunächst einmal sollte die Antenne für 800 - 900 MHz geeignet sein. Eine WLAN-Antenne für 2,4 GHz hätte vermutlich höhere Verluste. "Groß" und "klein" sind bei Antennen irrelevant, da geht es darum, welchen Antennengewinn in dBi der Anbieter angibt.  Antennengewinn bedeutet aber auch, dass die Antennen eine gewisse Richtwirkung hat. Man muss sie dann evtl. auf den Wechselrichter ausrichten.

Der IPX-Anschluss ist für max. 30 Steckvorgänge spezifiziert. Es spricht nichts dagegen eine 800 MHz-Antenne mit Kabel und IPX-Stecker zu verwenden. Nur sollte man die Antenne idealerweise 1mal anschließen und es dann so lassen. Eine Antenne mit SMA-Steckverbindung und eine Knickstelle zum Ausrichten ist da praktischer.

schluss123, 2023-08-17:
Geeignet und Verluste sind zwar schöne Worte, aber vielleicht nicht ganz die Antwort, die der Ersteller hatte:
der CMT2300A wird bei einer Mittenfrequenz von 850 MHz betrieben, das ergibt eine Wellenlänge von ca. 35cm. Die verwendete Antenne sollte dann ein viertel, halbes oder ganzes der Wellenlänge sein, um "optimal" an die Frequenz angepasst zu sein. Also sollte die Drahtlänge der Antenne : 35 oder 17,5 oder 8,75 cm lang sein. Einen wirklichen Antennengewinn hat keine dieser Antennen, da sie sehr wenig Richtwirkung haben und bei dBi immer gegen einen isotropen Strahler definiert wird.
Eine WLAN-Antenne mit einer Mittenfrequenz von 2.450 MHz hat eine Wellenlänge von 12,2 cm.
Also am besten einen IPX-Verbinder auf SMA-Einbaubuchse fest verbauen und dann einen Antennendrahtstummen mit SMA-Stecker bauen.

schluss123, 2023-08-17:
@lumapu Du hast Recht, es gibt noch einen Verkürzungsfaktor, aber bei den Reichweiten, die wir für die WRs brauchen, müsste auch eine nasse Spaghetti als Antenne reichen.

jho59, 2023-08-17:
Probiere es einfach aus und berichte dann, wie es funktioniert. 

Ich habe einen ESP32 nur mit CM2300A laufen und eine Antenne für die 860 MHz am Funkmodul. Entfernung zum Wechselrichter ca. 3,5 m durch eine 30 cm dicke Ytongwand bzw. durch das Fenster. 
 RX success 984, RX no answer 3, RX fragments 2946 - ich hätte da ein anderes Verhältnis erwartet.

58dahl, 2023-09-05:
Moin,
ich hab mal eine blöde Frage, im Setup des AHOY-DTU ist unten der IRQ Pin zu definieren. Wenn ich mein CMT2300 Breakboard ansehe ist da aber kein IRQ Pin vorhanden. Steh ich auf dem Schlauch oder welcher Pin am CMT ist gemeint?
Kann mir das jemand beantworten?
Vielen Dank
Pit

58dahl, 2023-09-06:
> ja das ist korrekt GPIO3 ist der IRQ (ich schreibe es zukünftig ins Setup)

Hi Likas, danke für die Aufklärung, das wird dann klarer für die Leute die nicht so tief in der Materie stecken.
Fürs Setup wäre das eine gute Idee evtl. GPIO3(IRQ) zu schreiben. Dann sollte es klar sein.
Lieben Dank und Gruß
Pit

MWagnerOE5AOO, 2023-09-09:
Hi zusammen, 
Nur zur Ergänzung zum Thema Antenne. Die Koaxleitung vom Board weg abisolieren und auf den Schirm und den Innenleiter je ein Stück Draht mit 7,8cm anlöten. So entsteht ein Dipol mit einer Resonanzfrequenz von 850Mhz. An der Luft ist der Verkürzungsfaktor 0,98. Der geöffnete Dipol liegt mit der Impedanz etwas neben den 50Ohm der Module. Daher passen die längen auch nicht genau.   
[image]
[image]
Soll die Frequenz bei 886 liegen die beiden Drähte auf 7,6cm kürzen.
viel Spaß und Erfolg beim Nachbauen
Michael