lumapu/ahoy #905: FeatureRequest: NRF24L01+ Reset button (manually)

nexulm, 2023-04-24:
Hi,

due to the fact that after updating my Ahoy today from 0.6.0 to 0.6.9 Ahoy reports the whole time that the NRF24L01 isn't connected or could not initiated. I've tried which short and long PowerOn-Reset but nothing helps at the beginning. For me it seems to be a timing issue while ESP8266 in combination with the NRF24L01+ is starting. After power off for round about 3-5min the issue was solved so far. Ahoy is working fine now, but for me it would be great if the UI will get a reset button for the NRF24L01+ to solve such an issue much easier and quicker. ;-)

lumapu, 2023-04-25:
don't understand the behavior. Normally a reset (power cylce) of 5 seconds is enough. The NRF is initialized every time Ahoy boots up.

nexulm, 2023-04-26:
Not sure why the power up was so unstable regarding resetting the NRF24L01+!?!
But today I've received suddenly such behaviour. :-(
[image]
---
Nevertheless I'll check the wiring these days as well (after the weekend).

closed 2023-07-23 by lumapu