lumapu/ahoy #906: NRF communication success rate with V0.6.9

reserve85, 2023-04-25:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hello,

I updated to version 0.6.9 a few days ago and the NRF communication has become worse. 

Here is the comparison between V0.6.0 and V0.6.9:

V0.6.0:
[image]

V0.6.9:
[image]

There is a difference in the success rate of 87% with version 0.6.0 and 73% with version 0.6.9.
I know there are only 100 data packets, but this increases linearly. Just recorded 100 to make the screenshot.

knickohr, 2023-04-25:
Only 100 TX ?  This is not enough. Please make such tests with at least 1000 TX or min. 1 hour by 10s interval.

reserve85, 2023-04-25:
Yeah i already did that yesterday, just captured no pic to show here...
i´ll post an update here later that day.

reserve85, 2023-04-25:
V0.6.0:
[image]
V0.6.9:
[image]

rejoe2, 2023-04-25:
Interersting. 
You're the first user documenting lower success ratios with the changed TX/RX logics. 

@beegee3 Any idea about that? 
And where do the differences between TX count and RX (success + fail + no answer) come from? In my setup, there's always only a very small difference (typically 1 or 2) between these numbers, most likely from (unanswered?) firmware version requests...

@reserve85 Can you please provide also info about your interval and the nRF used? ( A picture of the later may be helpfull).

reserve85, 2023-04-25:
Ok, i think i basically got it: zero export script was still running in the background (thought i turned it off but it was still active). Without running it it the background, the success rate is about 80%+ with 0.6.9.

But Anyway, with WebAPI commands in the background the Version 0.6.0 works better for me.

rejoe2, 2023-04-25:
Still, I'd state 80% to be a rather poor success rate.
But at least, there's some hint where the differences come from...

reserve85, 2023-04-25:
@rejoe2: inverval is 7 and max retries 5.
here are the pictures you requested:
[image]
[image]
[image]
[image]

knickohr, 2023-04-25:
Maybe the power supply ?

Looks like you‘re using a external device and soldered a USB plug. Have you tried standard USB supply ?  Maybe an additional capacitor at the 5V line helps ?

reserve85, 2023-04-25:
it´s this one: https://www.ebay.de/itm/142861629445
nothing soldered, i just cutted away the isolatian near the plug, that´s why you can see the red cable.
But if you want i can try another power supply.

I also tried a lot of alignment regarding the antenna, i get the best results if it´s straight out of the case (not 45° and not standing).

beegee3, 2023-04-25:
@reserve85 is not the only one, @onemorename has problems with retransmits, see https://github.com/lumapu/ahoy/issues/901#issuecomment-1521618199
It doesn't look like it's the same problem, but it's also related to radio transmission.

🤔 between 0.6.0 and 0.6.9 the method for receiving data in `hmRadio` changed. But I don't think that this is the problem. I also looked into my statistics and observed almost twice as much 'RX no answer' as compared with dev. 0.6.x versions where the new method was already implemented. A comparison of data taken on different days is always difficult due to different weather conditions and influence by other signals (wifi, bluetooth, ...). So I will do some tests tomorrow to see if changes between 0.6.9 and dev. versions before make a difference.

beegee3, 2023-04-26:
first a comment on my observed change in 'RX no answer': I changed the position of my ESP from above to below the desk. That could be the explanation. Sorry for that 😢 
Nevertheless here are the results of the tests:
with version 0.6.4 the method for receiving data changed. Since then it has stayed that way. So the tests compare versions 0.6.0, 0.6.4, 0.6.8 and 0.6.9. The inverter interval was set to 1s to get statistics with at least 1000 TX counts in short time. Since the number of transfers is the sum of TX count + TX retransmits one has to divide by this sum to get the right percentages for the RX values and TX retransmits.
Of course all tests were done without any power limit commands.

In a first step MQTT was disabled.
version|v 0.6.0|v 0.6.4|v 0.6.8|v 0.6.9
---|---|---|---|---
TX count|1001|1000|1000|1002
RX success|572|708|716|712
RX fail|40|20|17|19
RX no answer|388|271|267|271
RX fragments|2012|2239|2283|2253
TX retransmits|346|453|478|490
 | | | |
TX count+retrans.|1347|1453|1478|1492
 | | | |
% RX success|42,5%|48,7%|48,4%|47,7%
% RX no answer|28,8%|18,7%|18,1%|18,2%
RX fragments per TX|1,49|1,54|1,54|1,51
% TX retransmits|25,7%|31,2%|32,3%|32,8%

it is obvious that the change of method brought success and that there is no difference between 0.6.4 up to 0.6.9 !
It is clear that receiving more answers raises the probability of retransmits, so no wonder you see it here.

In a second step MQTT was enabled with an interval 0s (immediate publish).
version|v 0.6.0|v 0.6.4|v 0.6.8|v 0.6.9
---|---|---|---|---
TX count|1047|1000|1000|1007
RX success|303|414|438|542
RX fail|77|36|116|101
RX no answer|666|549|445|364
RX fragments|1303|1368|1661|1914
TX retransmits|477|397|867|995
 | | | |
TX count+retrans.|1524|1397|1867|2002
 | | | |
% RX success|19,9%|29,6%|23,5%|27,1%
% RX no answer|43,7%|39,3%|23,8%|18,2%
RX fragments per TX|0,85|0,98|0,89|0,96
% TX retransmits|31,3%|28,4%|46,4%|49,7%

**with MQTT enabled, each version in each category gets worse!** To my opinion the asynchronous MQTT data transfer is not only responsible for strange behavior mentioned in other issues but as well for 'bad' NRF transfer. Perhaps pending MQTT messages delay the NRF channel switching so that answers of the inverter are sent on a different chanel and not received. That raises 'TX retransmits' and 'RX no answer'.
But if you compare the versions within the table, 0.6.9 seems to be the best. That shows that the development of MQTT is on the right way. As far as I know from his comments, @lumapu ist still working on it. Let's wait and see ...

At last I liked to see if a fixed MQTT interval far away from the 1 second inverter interval is able to resolve the problem. By setting the MQTT interval to 10s it's going better, but still bad compared with the values of the first table.
Here only statistics for versions 0.6.4 and 0.6.9 were created.
version|v 0.6.4|v 0.6.9
---|---|---
TX count|1000|1009
RX success|522|583
RX fail|73|51
RX no answer|405|375
RX fragments|1855|1944
TX retransmits|741|564
 | | 
TX count+retrans.|1741|1573
 | | 
% RX success|30,0%|37,1%
% RX no answer|23,3%|23,8%
RX fragments per TX|1,07|1,24
% TX retransmits|42,6%|35,9%

knickohr, 2023-04-26:
I‘m wonderung why do you have so much fails 🤔

I‘m running 0.6.8 since days without any RX fail. ESP32, MQTT on, 5s interval, power level high, 5 inverters.

[image]

onemorename, 2023-04-26:
Mine is now online for two and a half days, no issues anymore on 0.6.9 since:

Moved the DTU closer to the inverters and changed Radio from Max to Low
Dtu is next to a repeater, improved from -68 RSSI to -41
MQTT is still set to 0
Raised Inverter Polling from 5s interval to 10s

[Image.png]

All fine now and perfectly stable!

[90DCA1DE-F3C1-49E8-8504-214369C8F5BF]<https://user-images.githubusercontent.com/63048210/234691558-24206633-e7f8-45dd-bb57-ea3e3a6a6293.jpeg>

onemorename, 2023-04-26:
Mine is now online for two and a half days, no issues anymore on 0.6.9 since:

Moved the DTU closer to the inverters and changed Radio from Max to Low
Dtu is next to a repeater, improved from -68 RSSI to -41
MQTT is still set to 0
Raised Inverter Polling from 5s interval to 10s

[image]

All fine now and perfectly stable!

lumapu, 2023-04-26:
@beegee3 thank you for that good and detailed testing. I can't imagine that MQTT generates that kind of problems. You're right I'm still working on a asynchronous publish of the values.
Funny thing: We work every time around the same topic: WiFi, heap and now MqTT.

I'm nearly finished with my improvements, I only need a timeframe during daylight to test it.

beegee3, 2023-04-26:
@knickohr: yes, the number of fails belongs to the new ESP position. Close to the router, surrounded by other electronics there is a lot of signal noise. I'm not happy with that, but for some reason the only possibility at the moment. On the other hand your statistics is always excellent. Hopefully it stays like this🤞

@onemorename: your statistics is fine too. Your very low 'RX no answer' (2,6%) is the explanation for the high 'TX retransmits' ("receiving more answers raises the probability of retransmits"). No time to worry 👍

@lumapu: nice to hear that. Can't wait to see the next version and, of course I hope for the best too (and will test it 😁)

reserve85, 2023-04-27:
Thanks for your tests, unfortunately for some reason it behaves a little differently for me. I can not understand why it is getting worse at my inverter, the only thing I changed is the firmware. However, I can live with it, so everything is fine.

beegee3, 2023-04-28:
**tests with new version 0.6.11**

first of all: I had a lot of problems with MQTT publish.
After turning on MQTT and rebooting the publish got stuck in an infinite loop of "ch0/LastAlarmCode 1".
Rebooted a second time. Now it got stuck in a loop of "ch0/active_PowerLimit 100".
The next reboot led to a loop of publishing the inverter hardware info.
Only after several reboots it finally published the desired actual values.
Each time I changed something in the settings it ran into the same problems again.
While the publishing got stuck, serial output and live view showed the actual values.
So the problem lies somewhere in the publish queue (@lumapu sorry, I haven't got enough into the new code to see where).

Now to the tests:
Since it's a new day and the ESP position slightly changed the results cannot be compared with the statistics stated before.
The first tests use an inverter interval 1s. Starting without MQTT, then with MQTT interval set to 0s.
As you can see, the results ran worse, the same as with the old versions. Also the webUI was laggy and sometimes the live view displayed total values even if only one inverter was used. In one case three of the brown boxes were shown (!)
But with a MQTT interval set to 10s the results are on the same level as without using MQTT!!! It looks like the polling interval 1s is too fast when publishing directly. That's why I tested as well with polling intervals of 2s and 5s while MQTT interval set to 0s. And, surprise (or not?), with 2s it went much better compared to 1s. From time to time the webUI was still laggy.
But the best results of all tests are given with 5s! And (with very few exceptions) a smooth webUI. 5s or more are more realistic than the 1s or 2s used for the tests, so that's fine.

polling interval|1s|1s|1s|2s|5s
---|---|---|---|---|---
MQTT interval|no MQTT|0s|10s|0s|0s
|||||
TX count|1000|1001|1000|1001|1052
RX success|821|478|817|743|962
RX fail|21|77|24|13|0
RX no answer|158|445|158|245|90
RX fragments|2562|1671|2541|2256|2882
TX retransmits|478|613|496|585|482
|||||
TX count+retrans.|1478|1614|1496|1586|1534
|||||
% RX success|55,5%|29,6%|54,6%|46,8%|62,7%
% RX no answer|10,7%|27,6%|10,6%|15,4%|5,9%
RX fragments per TX|1,73|1,04|1,70|1,42|1,88
% TX retransmits|32,3%|38,0%|33,2%|36,9%|31,4%
---
... after 3 hours with inverter interval set to 5s and MQTT interval set to 0s the good statistics values remain.

Two things I've noticed:
1. while publishing, the free heap temporarily decreased by about 25000. No problem for the ESP32, but is is ok for the ESP8266?
2. frequently clicking on the webUI turns it laggy again. This may not be the standard situation, but perhaps it is possible (and easy) to put the webUI demands and responses in a queue as well?

All in all seems to me the new version still needs some fixes, but is on the right way.

beegee3, 2023-04-29:
addendum: with version 0.6.12 the problem with MQTT infinite loops is solved (for me). Everything is running fine now!

lumapu, 2023-05-03:
@beegee3 hast du die Messungen nochmal mit der `0.6.12` gemacht? Wäre interessant inwieweit MqTT jetzt noch die Successrate reduziert / beeinflusst.

beegee3, 2023-05-04:
@lumapu der Unterschied zwischen 0.6.11 und 0.6.12 ist ja nur die Fehlerbehebung der MQTT infinite loops, d.h. die Messungen zur 0.6.11 können so auch für die 0.6.12 genommen werden. Mit der 0.6.12 habe ich daher die Messungen nicht nochmals durchgeführt, sondern das inverter polling interval auf 5s mit MQTT interval 0s. Die Statistikwerte entsprechen denen der 0.6.11 (Tabelle oben, rechte Spalte). Die Umstellung der MQTT publish Methode ist daher aus meiner Sicht ein voller Erfolg!

Nur das oben angesprochene temporäre Absinken des free heap könnte m.E. beim ESP8266 Probleme bereiten. Das in anderen Issues angesprochenen MQTT disconnected könnte eine Folge davon sein. Habe den Verdacht, dass das free heap Absinken durch häufiges Klicken in der WebUI ausgelöst wurde, also nicht in den MQTT Funktionen zu suchen ist. Wenn die Inverter Abfragen zu Retransmits führen, bleibt dem ESP wenig Zeit, andere Aufgaben zu erledigen, wodurch es zu Hängern bei der WebUI kommen kann und der free heap belastet wird (im Prinzip analog zu dem, was vor 0.6.11 auch das MQTT Problem war).
Eigentlich sollte man die Retransmits zeitlich verzögern, d.h. TX packets erst dann senden, wenn sonst nichts anliegt (WebUI, Api, MQTT). Ist aber wahrscheinlich nicht so einfach umzusetzen. Doch ein vernünftiges Inverter Intervall hilft ja auch. Unter 5 Sekunden sind für den Produktivbetrieb generell nicht zu empfehlen, bei schlechtem NRF Empfang (-> viele Retransmits) lieber mehr (insbesondere in Kombination mit ESP8266). Das ist natürlich auch beim Limitieren zu berücksichtigen. Vielleicht kann man das als Empfehlungen formulieren und in den User Guide aufnehmen.

knickohr, 2023-05-04:
Mein nicht sehr, auch nicht konstruktiver Kommentar dazu :
Empfehlungen werden nicht gelesen und verdrängt, sogar ganz bewußt ignoriert 😒 Siehe Elko und Power Level 🫣
Es vergeht kein Tag in der Community ohne das wieder ein Anrufer daher kommt und mit den unendlich alten Fragen nervt 😢

lumapu, 2023-05-04:
@beegee3 ok, verstanden. Die Werte sind ja die besten aus der ganzen Tabelle 😄.
Finde toll, dass die neue MqTT publish Methode für ankommt, evtl. ist es dadurch sogar lesbarer geworden. Die Änderung hatte ich eigentlich für den ESP8266 gemacht, interessant, dass auch der ESP32 davon so stark profitiert.
Bzgl. Heap habe ich seit der Änderung die besten Werte seit dem es Ahoy gibt, habe keinen signifikanten Einbruch gesehen.
Ich bin seit einiger Zeit dran den Kern (App, hmPayload, hmInerter usw. bei zu bauen. Ich erhoffe mir dadurch noch bessere heap Werte und vor allem ein echtes Abfrageintervall.

@knickohr dokumentieren bringt auch nichts oder, ich weiß, das AhoyDTU.de total hinterherhinkt.

knickohr, 2023-05-04:
@lumapu 
Es sollte halt in der Doku das optional raus und bei den Bildchen prinzipiell eingezeichnet sein.

Egal ob es nun wirklich gebraucht wird oder nicht. Ein 30 Cent Elko macht das Kraut nicht fett 😉

knickohr, 2023-05-06:
So Leute, bin seit 35 Stunden auf der 0.6.12.

Neben dem Problem mit dem Nullen, siehe #927, habe ich plötzlich wieder RX fails. Gleiches Setup, gleicher Standart 😵 Ich beobachte das mal weiter, kann gut sein das das schwere Gewitter gestern Abend mir da einen Streich spielt.

ZsZs73, 2023-05-07:
Hello,
I am also facing with an issue of low number of updates coming from the NRF with v0.6.9 release build.
I found that the issue ceases if I disable the display (SSH1106 connected to D1 and D2)
Please see the following screenshot. At 11:55 I've disabled the display.
[image]
MQTT is set to 15 sec update interval.

closed 2023-07-23 by lumapu