lumapu/ahoy #907: Thingspeak

HeinKlocktein, 2023-04-25:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.9_
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Is it possible to implement a direct connection to thinkspeak e.g. with MQTT?

Regards, from north germany,

Michael

rmayergfx, 2023-04-25:
Hallo Michael,

thingspeak nicht thinkspeak. 
Dazu gibt es bereits Anleitungen: https://www.instructables.com/Uploading-Data-to-ThingSpeak-With-MQTT/

lumapu, 2023-04-25:
siehe auch #185 und #410

HeinKlocktein, 2023-04-25:
@lumapu: Danke für die Links zu den Diskussionen. Leider letztendlich nicht zielführend bzw. die threats sind m.E. ergebnislos.  
@rmayergfx Dein Link verweist auf die Verwendung von MQTT und Thingspeak generell bzw. unter Verwendung eines Programms. Das scheint veraltet zu sein, da es bei Thingspeak eine Schnittstelle über mqtt3.thingspeak.com gibt.
Wenn man eingeloggt ist kann über https://thingspeak.com/devices/mqtt der Channel freigeschaltet werden.   

Ahoy-DTU habe ich versucht wie folgt zu konfigurieren:

Broker / Server IP: mqtt3.thingspeak.com
Port: 1883
Username (optional): Ogw4LB4fAwXXX --> Spezieller Thingspeak MQTT Username  
Password (optional): xxxxxxxxxxxxxxx --> Spezielles Thingspeak MQTT Passwort
Topic: channels/1243952/publish

Die Serial Console zeigt diese Meldung: 
 I: MQTT disconnected, reason: malformed credentials

Die Anmeldedaten habe ich mehrfach überprüft. Auch hab ich sowohl mit subscribe als auch mit publish mein Glück probiert. 

Gruß

Michael

rmayergfx, 2023-04-26:
@HeinKlocktein 
Überleg dir ob du wirklich die Daten in der Cloud haben möchtest. Warum nicht lokal auf einem Raspi oder Docker, da geht nichts ins WWW und du bist mit deinen Daten sicher, egal ob der Dienstleister in ein paar Jahren den Server abschaltet oder umzieht. 

Meine IOT sind in einem eigenen VLAN oder Zugriff ins Internet und auch die Datenbanken liegen lokal. Es gibt dazu ja schon einige Lösungen. 
z.B. https://solaranzeige.de/phpBB3/solaranzeige.php, Home Assistant...

HeinKlocktein, 2023-04-26:
@rmayergfx: Danke für deine Hinweise. 
Kannst Du mir bei meiner Frage helfen? 
I: MQTT disconnected, reason: malformed credentials
Ist mein Topic, bzw. wie ich das mache, so ok?

Gruß

Michael

rmayergfx, 2023-04-26:
Nein, kann ich nicht da ich kein ThinkSpeak benutze. Frag mal @lumapu was sich hinter der Fehlermeldung verbirgt.

Lade dir doch mal den http://mqtt-explorer.com/ und prüfe damit ob du überhaupt an das System kommst.

lumapu, 2023-04-28:
ich kann das auch nicht erklären. Irgendwas stimmt mit den Logindaten bei MqTT nicht.

HeinKlocktein, 2023-04-30:
@rmayergfx Der MQTT-Explorer funktioniert mit Thinkspeak nicht. Zumindest in der aktuellen Version nicht. 
MQTTX funktioniert (https://mqttx.app/).
@lumapu zur Info: 
Im Netz gibt es einige Diskussionen dazu. Denke es hängt an der verwendeten Lib und/oder an Thinkspeak Besonderheiten.  
--> Mit MQTTX muss man erst eine Verbindung mit Thinkspeak etablieren und....

Ich schließe das Thema mit dem Ergebnis das es so z.Zt. nicht umsetzbar ist.

closed 2023-04-30 by HeinKlocktein