lumapu/ahoy #908: "powerdelta"

grericht, 2023-04-25:
Hallo,
bei Tasmota gibt es für mqtt ja die möglichkeit per Powerdelta sich häufiger/schneller bei Änderungen informieren zu lassen. Gibt es die Möglichkeit bei der AhoyDTU auch? Der Code hat doch sicher eine gewisse Nähe?

lumapu, 2023-04-25:
die Daten werden immer im gleichen Interval ausgelesen. Dh. es gibt keine 'schnelleren' Daten. Wenn du einen ESP32 verwendest, sollte ein Abfrageinterval von 2-5 Sekunden drin sein.

grericht, 2023-04-25:
Ich will ja auch keine schnelleren Daten. Ich will sie nur "bei Bedarf schneller senden".
Ich bekomme aktuell die Daten alle 5sekunden vom Inverter. Aktuell gebe ich sie alle 10 Sekunden per mqtt eiter, da mein Tasmota SMI es auch nicht schneller schafft. Wenn es aber eine Schwankung gibt (Wolke zieht vor die SOnne) gibt es bei Tasmota die Funktion, die Daten bei "deutlicher" Änderung zum letzten Wert per powerdelta sofort weiter zu geben. Das macht hier doch auch SInn. Man könnte den Datenverkehr im Netz deutlich einschränken und nur noch alle 30Sekunden senden und dafür bei Änderungen über 10% sofort senden.
Die Funktion gibt es aber nicht?

offtopic: leider wurde die bei dem SMI Tasmota Feature gesperrt. Weil wohl die Annahme ist, dass die SM nur den Zählerstand führen. Meiner hat aber halt auch den aktuellen Verbrauch. und den Sekündlich. Aber ich bekomme den nur alle 10Sekunden per mqtt gesendet.

rmayergfx, 2023-04-26:
Alle 10sec ? Was möchtest du denn damit erreichen. Ganz ehrlich, was bringt es dir den Wert in so kurzen Intervallen zu erhalten. Wenn er dann noch vllt. in eine Datenbank geht wächst diese innerhalb kürzester Zeit unnötig an.

Möchtest du damit etwas in Echtzeit steuern?

grericht, 2023-04-26:
klar will ich damit im besten fall in echtzeit steuern?! Es ist für mich die einzige Möglichkeit herauszubekommen wieviel Solarenergie ich aktuell produziere und damit will ich natürlich auch sachen steuern. ich wüsste auch nicht warum 10sekunden da zu schnell sein sollten. und auch die datenbank kann werte alle 10sekunden mehr als gut vertragen. manche loggen sekündlich!

und ganz grunlegend fühle ich mich immernoch sehr missverstanden. ich will gar nicht häufiger die daten. ich will sie nur BEI BEDARF schneller! wenn der tag keien wechselnden bewölkungen hat, dann sollte die produktion langsam ansteigen und dann langsam wieder sinken. da reicht mir auch einmal pro minute der wert. aber wenn eine wolke vor die sonne zieht, will ich den wert halt nicht erst 30 sekunden später. auch 10 sekunden können da sehr viel sein. häufiger will cih aber nicht einstellen, da ich dann viele sehr änliche werte bekomme.
powerdelta ist dafür perfekt. daten alle 60 sekunden und bei änderungen über 10% sofort. ich verstehe gar nicht warum das hier nicht verstanden wird? nochmal: ich will eigentlich viel seltener daten in die datenbank schreiben. aber dafür halt vor allem werte die anders sind als der wert vorher.

rmayergfx, 2023-04-27:
Ich habe es schon verstanden was du gerne möchtest, aus meiner Sicht wäre so eine Delta Abfrage aber auch wiederum nur sinnvoll, wenn man einen Schwellwert definieren kann, d.h. die Meldung geht erst raus wenn z.B. 50% Änderung gegenüber dem vorhergehenden Wert stattfindet. Nicht immer macht es doch auch Sinn im Sekundentakt Änderungen vorzunehmen. Was soll denn das Ziel sein? Möchtest du in Echtzeit gegen den Stromzähler steuern?

grericht, 2023-04-27:
ich glaube die powerdelta funktion ist nicht verstanden.
UNABHÄNGIG vom teleperiod zeitraum (also der zeitraum alle wieviel sekunden die aktuellen daten per mqtt geschickt werden) kann ein powerdelta festgelegt werden. dieses kann entweder in numerischer form oder in prozentgröße angegeben werden. powerdelta 1-100 = prozentuale änderung
powerdelta 101 - unendlich = änderung um einen gewissen wert seit der letzten meldung + 100

heißt es gibt genau die gesuchte möglichkeit: teleperiod eher selten (zB nur einmal pro minute) und bei änderung im mehr als 1 watt oder mehr als 5% sofort die daten ins netzwerk. das funktioniert mit tasmota perfekt.

grericht, 2023-04-27:
blöd ist auch bei ahoy dtu, dass dehr viele daten jedes mal übertragen werden. den tagesertrag zb bräuchte ich viel seltener. bzw. richtig oft brauche ich nur die watt auf der wechselspannungsseite. dort hätte ich gerne jede sekunde den wert

rmayergfx, 2023-04-27:
Du darfst eines nicht vergessen. Die Daten werden vom WR zur Verfügung gestellt. 
In AhoyDTU stellst du dann das Abfrageintervall ein, per MQTT has du dann auch sofort die Werte. Was du dir wünschst wäre eine abgespeckte Abfrage, der dir z.B. nur die für dich Relevanten Werte aus dem WR ausliest und weitergibt.
Und da stellt sich dann die Frage, ob das überhaupt mit der Firmware des WR möglich ist, oder es dort nur ein "Alles oder Nichts gibt".
Daher verstehe ich nicht, wie willst du das Umsetzen. AhoyDTU kann doch gar nicht wissen ob sich ein Wert geändert hat oder nicht, sondern nur wenn der WR abgefragt wurde. Also müsstest du Prinzipiell jede Sekunde den WR abfragen und dann entscheiden was nun geschehen soll, aber da die Werte AhoyDTU nun vorliegen hat, kannst du sie auch per MQTT auslesen und mit einem Script z.B. in HA oder was auch immer weiterverarbeiten, du musst dann halt auch nur den(die) Werte abfragen die für dich sinnvoll sind.

grericht, 2023-04-27:
das habe ich bisher anders verstanden. ich stelle doch das abfrage und das mqtt sende intervall getrennt ein. ich könnte also zb jede sekunde vom wr abfragen und alle 30sekunden per mqtt senden+powerdelta bei zb 5% änderung. dann hätte ich die werte alle 30 sekunden oder wenn änderung nach maximal 1 sekunde. genau wie gewünscht. jetzt muss ich irgendwelche kompromisse eingehen

rmayergfx, 2023-04-27:
Und welchen Wert hast du im MQTT Broker eingestellt? Ich sehe hier wirklich keinen Anwendungsfall das ich unbedingt mit einem Delta arbeiten müsste. Die Systeme haben doch sowieso eine gewisse Trägheit. 

Schreib doch mal genau was du überhaupt machen möchtest, vllt. gibt es ja einen ganz anderen Weg der schneller und effizienter ist.

grericht, 2023-05-11:
also nochmal (und ich verstehe immer noch nicht wie man das nicht verstehen kann) das szenario:
ich lese meinen digitalen stromzähler mit einem ttl lesekopf an einem esp8266 mit tasmota aus. Ausgelesen werdne aktuelelr Verbrauch aller Phasen einzeln, gesamt und der Zählerstand aller phasen einzeln und gesamt. alle werte können (das ist tasmota eigen) in einem einstellbaren rythmus gesendet werden. Das schnellste sind 10Sekunden. Normalerweise bietet Tasmota noch die powerdelta variante. Dann kann je NACH EINSTELLBARER ÄNDERUNG (entweder nach veränderten W oder %) auch schneller/sofort der wert geschickt werden. Dieser ist (dummerweise) genau für die Smart Meter Interface Routine ausgeschaltet. Das häufigste sind deswegen 10Sekunden. AUßER: man kann auch für einzelne Werte festlegen, dass sie sofort nach Erhalt geschickt werden. Das habe ich für den aktuellen Gesamtverbrauch so eingestellt. Dieser wird also etwa sekündlich per mqtt der rest alle 10sekunden gesendet. Diese Werte lese ich in Home Assistant aus und verwende sie für andere Enditäten und Automatisierungen. Ich weiß also sekündlich ob und wieviel ich gerade ins Netz speise oder aus dem Netz beziehe. 

Außerdem habe ich für mein Balkonkraftwerk mit hoymiles wr eine ahoy dtu. Diese ist aktuell so eingestellt, dass sie ich glaube alle 2 sekunden vom wr die aktuellen werte bekommt und ich glaube alle 10 sekunden per mqtt schickt. Für viele Automatisiereungen und berechnungen brauche ich nun die aktuelle Produktion und den Bezug.
So möchte ich zum Beispiel gerne wissen wieviel aktuell im Haus Verbrauche. Das ist aktuell die Summe von Bezug und Produktion. Leider ist der eine Wert sekundlich und der andere Wert mit 10 Sekunden sehr selten und damit sehr oft ungenau. dass heißt, dass mein errechneter RealVerbrauch teilweise negativ ist, weil der Bezug schon bei -300W ist aber die Produktion im HomeAssitant noch bei 100W steht, weil es bis gerade bewölkt war. 

Nun gibt es leider keine option entweder nur den aktuellen Produktionswert häufiger zu schicken (ok, ich könnte den häufiger einzeln anfragen glaube ich) oder bei genau dem wert einzustellen, dass dieser häufiger/schneller gesendet wird, wenn er sich deutlich ändert.

Es ist jetzt nicht so, dass dies für Automatisierungen großen einfluss hat, dass mal etwas erst 10sekunden später losgeht. Für die Statistik ist es aber sehr ungünstig/unschön wenn dort steht, dass ich zB einen Minimal Verbrauch an dem Tag im Negativen bereich hatte. Ich verbrauche ja zu jedem Zeitpunkt mind. 100W in meinem Haus.

Also entweder wäre es gut den P-DC Wert häufiger/schneller schicken zu lassen oder ioch muss ihn halt wilrich häufiger anfragen per mqtt. das sit danna ber der doppelte traffic.

grericht, 2023-05-11:
auf die tasmotaoptopn powerdelta zurückzugreifen erscheint mir immernoch als super praktikabel.

dtuuser, 2023-05-11:
Ich lese meinen Zähler mit einem Tasmota jede Sekunde per Webapi Aufruf ab, und meine HM600 per Webapi alle 20 Sekunden. Verarbeitung auf einem Windowsrechner. Dann wird bei Bedarf innerhalb von 10 Sekunden der WR über API gedrosselt. 
Nach der Drosselung findet die nächste Drosselung erst nach 60 Sekunden statt. So komme ich aktuell an einem Sonnentag auf maximal 300Watt Einspeisung. Seit Juni 2022 habe ich so bis jetzt 400KWh produziert, aber nur 31KWh eingespeist.
An einem Sonnentag komme ich aber auf bis zu 180 Änderungen.
Da ich kein mqtt nutze, habe ich aktuell Laufzeiten von bis zu 21 Tagen

grericht, 2023-05-15:
"Wir haben das alle schon längst verstanden, aber ..." scheint nicht so: "Selbst wenn du mal einen Smartmeter bzw. digitalen Zähler eingebaut bekommst" ich habe längst einen digitalen Smart-Meter. Ich lese genau einen Wert dort sekündlich aus: meinen aktuellen Bezug/Einspeisung.
Die anderen Werte interessieren reichen mir alle 30Sekunden.
Diese Geschichte von zu viel Datentrafic bei häufigerer Abfrage:
1. reden wir hier von Daten im Byte oder kB Bereich. ich nehme nicht war, dass mein Home-Net hier ürgendwie überlastet ist
2. (noch wichtiger) ich möchte weniger Daten ins Netz! Ich möchte genau einen Wert des Hoymiles WR/der AhoyDTU häufiger im Netz haben. Und zwar die aktuelle Produktion. Die gerne Sekündlich oder noch besser bei Änderungen über 1W (deswegen die powerdelta Anfrage - bei Tasmota geht das). Alle anderen Werte, die ich aktuell auch alle 10Sekunden einspeise reichen mir locker auch alle 30sek. INsgesamt würd eich also massiv Traffic sparen.

Und jetzt nochmal ganz konkret: Gibt es irgendwlche Möglichkeiten einzelne mqtt Werte häufiger als andere zu senden?

lumapu, 2023-05-18:
nein die Daten per MqTT sind immer komplett, da diese auch nur komplett von Wechselrichter abgefragt werden können.

grericht, 2023-05-19:
ok. jetzt sind wir quasi beim anfang. der ist zustand ist geklärt. und nun habe ich viele beiträge damit verbracht zu erklären, dass mir das feature, dass einzelne daten häufiger per mqtt gesendet werden könnten als andere sehr gut gefallen würde. ich meine es sogar ganz gut beschrieben zu haben warum das so ist. und ich glaube, dass es grundsätzlich möglich ist. bei tasmota ist es sogar möglich bedingungen festzulegen wie schnell einzelne werte gesendet werden sollen.
mir wurde hier bisher erklärt, dass es dafür keinen bedarf gibt - das weise ich zurück! ich bräuchte den aktuellen power-wert deutlich häufiger als die anderen werte. ich glaube für andere wäre dieses feature auch interessant.
oder mir wurder erklärt, dass dies zuviel traffic verusachen würde, was ich zurückweise, weil es mir darum geht WENIGER traffic zu produzieren. wenn ich nur einen wert häufig brauche, kann ich mir entweder alel werte häufig senden und zuviel traffic haben oder alle werte selten und die Anforderung nicht erfüllen. Ich will genau einen von 24 Werten so oft wie möglich und den Rest alle 60 sekunden statt jetzt alle 2 oder wie viele heir einstellen alle 10sekunden. Ich behaupte, dass die meisten nur P_AC so häufig brauchen.

wenn nun ein kleines bischen verständlich geworden ist worum es geht: kann man hoffen, dass dieses Feature irgendwann kommt?

grericht, 2023-05-19:
> nein die Daten per MqTT sind immer komplett, da diese auch nur komplett von Wechselrichter abgefragt werden können.

auch wenn ihr euch da gegenseitig daumen hoch gebt: die daten vom inverter zu bekommen und per mqtt zu senden sind doch zwei komplett getrennte dinge. der esp8266 bekommt die daten vom inverter (alle) und es ist technisch programmierbar (und aktuell auch einstellbar) wie oft er sie per mqtt weitergibt. beides sind doch mit sicherheit unterschiedliche skripte und es sollte technisch möglich sein in dass mqtt senden skript reinzuschreiben, dass der eine wert sooft und der andere wert sooft gesendet wird. welche werte wie oft gesendet werden könnte in der weboberfläche oder per console einstellbar sein.

rmayergfx, 2023-05-19:
Ich verstehe deinen Standpunkt durchaus, es bringt aber rein gar nichts hier angesäuert zu reagieren. Geh in die Sourcen, schau es dir an und compiliere es selbst wenn es so immens wichtig für dich ist. Niemand sonst benötigt so etwas spezielles und es ergibt m.E. auch keinen Sinn so etwas zu machen, da die Systeme unabhängig davon eine gewisse Trägheit haben, bevor sie auf die Änderungen reagieren. 
Aus technischer Sicht ist das absolut sinnbefreit, du würdest ein System benötigen das die Daten vom WR in Echtzeit ohne Verzögerung weiterreicht, sinnvollerweise verdrahtet mit kurzen Wegen, damit auch keine Latzenz oder unnötiger Traffic auf irgendeinem Netz stattfindet. Gerade bei WLAN hast du bedingt durch das Protokoll und den Overhead immer eine gewisse Größe, egal ob du nun 1 oder 2 Werte übertragst. Hast du noch Störungen durch Fremdeinflüssse, werden die Pakete vllt. auch noch mehrfach übertragen und dann sind die Laufzeiten auch schon wieder höher. Ich möchte persönlich nichts im Netzwerk haben, was mir sekündlich Traffic verursacht.  Das mag auch nicht jeder Router.

btw. mit den nächsten Generationen der WR wird das sich u.U. sowieso komplett erledigt haben, da die eingesetzen Firmware Releases und Hardware z.B. per Default nur noch alle 15 min die Daten in die Cloud schicken.

grericht, 2023-05-23:
"Aus technischer Sicht ist das absolut sinnbefreit, du würdest ein System benötigen das die Daten vom WR in Echtzeit ohne Verzögerung weiterreicht"
genau das geht übrigens mit Tasmota bei aktiviertem Smart Meter Interface. Hier kann ich im Script 16 zur "Genauigkeit" draufschlagen und schon wird dieser EINE Wert sofort nach Erhalt per MQTT gesendet. Bedeutet konkret, dass ich halt den aktuellen Verbrauch laut Smart Meter fast Sekundengenau im System verarbeiten kann. Da fände ich es schon schön den Solarproduktionswert vom Hoymiles auch sofort weiterzuleiten und nicht versucht zu sein alle fast 25Werte so oft wie möglich zu senden. 
Ich halte s für ein Gerücht, dass ich der einzige bin, der das nutzen würde. Es wäre eine Reduktion von Traffic.
Auch deine Anderen Sachen halte ich für Gerüchte. Fast alle die HomeAssistant oder ioBroker nutzen wollen die Daten möglichst genau und damit auch Automatisierungen nutzen. Da macht 15Min gar keinen Sinn. Zumindest für die aktuelle Produktion. Der Rest kann gerne nur alle 15Min gesendet werden. würde mir sogar einmal am Tag reichen zu wissen wieviel ich insgesamt produziert habe.
Die Anderen Dinge sidn einfach irgendwelche Ängste und Theorien die sich nicht praktisch nachweisen lassen. Mein System muss nicht in Echtzeit arbeiten. Macht es auch nicht. Aber ich hab aktuell fast alle Werte alle 2 Sekunden. Und würde einfach gerne 23Werte nur zB einmal pro Minute schicken. Wenn mal ein Wert übersprungen wird passiert genau gar nichts. Wüsste nicht warum ich das reduzieren sollte.

Aber gut, die ANtwort, dass es technisch möglich ist aber du mir nicht helfen magst hätte als Post 2 gut getan. Evtl werde ich die Frage nochmal in English neu stellen und schauen ob andere den P_AC Wert auch häufiger/Alle anderen Werte seltener senden wollen und sich jemand der Problemstellung annehmen mag.

rmayergfx, 2023-05-25:
Willst du mit deinem Anliegen nun das Repository in allen Sprachen zumüllen? Ganz schlechte Idee.
@lumapu hat dir oben bereits längst geantwortet was möglich ist und was nicht.  Rate mal welche Sprache er spricht und woher er kommt....

grericht, 2023-05-25:
nein. aber dieser thread ist leider aus verschiedenen gründen von anfang an sehr am Thema vorbei gegangen. Es gab am ANfang die Aussage gibts nicht und ist dann immer mehr zu würde gehen aber braucht niemand geworden. diese Aussage kommt von genau 2 Menschen. Und der Grund weswegen es nicht gebraucht wird ist, dass es zuviel Traffic produzieren würde und dieser ist obsolet.

Ich würde also gerne genau noch ein Thread eröffnen und fragen ob die Möglichkeit TRAFFIC ZU SPAREN indem alle Werte seltener und nur einzelne Werte häufiger gesendet werden können Interesse genießt und wenn ja ob er umsetzbar ist. Ich glaube es gibt Bedarf und für Leute am Code ist dieser einfach umsetzbar. Hier wird leider beides nicht deutlich und nachdem ihr versucht habt mir lange zu erklären, dass ich erstmal nen Smart Meter brauch, den ich lange hab, ist hier halt nichts mehr zu "gewinnen".

lumapu, 2023-05-25:
Ich fasse den feature-request mal zusammen:
Du willst bei einem fix eingestellten MqTT intervall (verschieden von 0) eine weitere Option haben, die es erlaubt prozentual festzulegen, dass schon früher gesendet wird. 
Es wird schnell komplex, daher die Frage an @grericht:
Was soll Ahoy machen, wenn die Abweichung nur für einen Inverter zutrifft, darf dann trotzdem alles gesendet werden - das wäre das einfachste Szenario.

brcommander, 2023-08-25:
Moin zusammen, bin gerade über diesen thread gestolpert, weil ich auch auf der Suche nach einer passenden Lösung bin. 
Ich hätte gerne im Prinzip auch die Option mit PowerDelta.
Meine Konstellation:
PV-Anlage liefert an einen SolarEdge Wechselrichter. Gleichzeitig habe ich noch ein Batteriesystem, das ebenfalls an einem (eigenen) Wechselrichter hängt.
Da es leider so ist, dass ich zwar den Batterie-Wechselrichter und die Batterie per Modbus auslesen kann, nicht aber den "großen" Wechselrichter, der den PV-Strom vom Dach in Wechselstrom wandelt, bin ich für den Hausverbrauch auf den berechneten Wert aus der SolarEdge Monitoring Cloud angewiesen. Und das ist... naaaaajaaaaa.
Aus diesem Grund habe ich mir nun einen Shelly Pro 3EM an die 3 Phasen des Haupt-WR geklemmt. Und der liefert mir (per Home-Assistant Integration) alle 3 Sekunden entsprechende Werte.

Weiterhin habe ich einen Hichi Lesekopf am EVU-Hauszähler. Darin werkelt ein ESP8266, was uns zu dem genannten Problem führt, dass der maximale Wert für Teleperiod auf 10 begrenzt ist, auch wenn der Lesekopf mind. alle 3 Sekunden aktuelle Werte liefert (kann man ja in der Tasmota Weboberfläche sehen)

Um nun meinen tatsächlichen Hausverbrauch (aktuelle Leistung) lokal zu berechnen, brauche ich folgende Daten:
1. PV-Produktion NACH WR (damit ich auch wirklich die vorhandene Leistung ohne Wechselverluste habe) -> Shelly Pro 3EM
2. Die Daten vom Batterie-WR (per Modbus): negativer Wert, wenn die Batterie geladen wird, entsprechend Positiv, wenn ich "Produktion" aus der Batterie habe
3. Netz-Leistung, abgelesen per Lesekopf am Hauszähler: Negativ bei Einspeisung, positiv bei Bezug (= auch eine Art Produktion)

--> 1 + 2 + 3 = mein Nettoverbrauch im Haus

und hier ergibt sich das Problem: Wenn jetzt 1 und 2 im 3-Sekunden-Takt kommen, 3 (Netz) aber nur alle 10sek, dann ist die Berechnung nur alle 30 sek. richtig..

Soweit so gut / schlecht. Und ja, die Frage nach dem Sinn von so vielen Datenpunkten ist natürlich berechtigt. Dazu sei aber folgendes gesagt:
1. Wenn man die Datenpunkte in einer InfluxDB speichert, dann kann man die später mitteln und damit weniger Datenpunkte aufberwahren (Stichwort "Retention Policy")
2. Die Menge an Daten, die übers Netzwerk müssen hält sich in Grenzen (gefühlt macht da jeder PC, Smartphone etc. deutlich mehr Telemetrie, ohne dass ich das aber jetzt im Einzelfall nachgeprüft hätte).
3. gibt es durchaus Anwendungen, für die korrekte und einigermaßen zeitnahe Informationen nötig sind. An der Stelle möchte jetzt keine (erneute) Diskussion auf machen, ob das wirklich so ist etc. pp. (z.B. zum Abschalten von sensiblen Geräten, wenn die Netzfrequenz einen gewissen Schwellenwert übersteigt...) Sagen wir einfach: auch ich möchte die Daten schneller im Home Assistant haben. Punkt.

Ja, ich bräuchte nicht in jeder Übermittlung alle bereitgestellten Daten wie SSID, Energie (kWh)... das würde tatsächlich auf deutlich seltener reichen. Aber gut, da kann man ja die Datenaufzeichnung nur bei Änderungen machen.

Aber: Im Fall des IR-Lesekopfs stehen die Daten ja relativ zeitnah bereit, sodass ich nicht erst abwarten muss, bis alle Sensoren abgearbeitet sind, was laut @arendst durchaus auch mal 4-5sek dauern kann und somit eine 3sek Teleperiod sinnfrei wäre.

Die Funktion `PowerDelta ` klingt nach einer passenden Option für mich. Nur leider gibt es sie in der Tasmota-Variante für den 8266 im Standard nicht. Vielleicht kann man sich ja eine custom-Variante zusammenbauen, wo das wieder verfügbar ist, dafür andere Sachen wie Domoticz etc. wegfallen.

Was ich übrigens auch noch gelesen habe: es gibt ja die [Regeln](https://tasmota.github.io/docs/Rules/) in Tasmota. Leider habe ich das noch nicht geschafft, die zum Laufen zu kriegen, aber damit könnte es funktionieren. Für Hinweise bin ich dankbar!

grericht, 2023-09-26:
> Ich fasse den feature-request mal zusammen: Du willst bei einem fix eingestellten MqTT intervall (verschieden von 0) eine weitere Option haben, die es erlaubt prozentual festzulegen, dass schon [...]

Ja klar darf dann alles gesendet werden. Immernoch besser als alles ganz oft zu senden oder die daten meist zu spät zu bekommen. Ideal ist es aber nicht, da dann das ganze Paket wohl trotzdem sehr oft gesendet wird.
Bei Tasmota kann man aber in der Console einstellen welche Daten bei welchem Powerdelta gesendet werden. Es ist sogar möglich einige Daten häufiger als andere und trotzdem zu anderen Zeiten als das ganze Paket zu senden. Das wäre natürlich ideal. Wenn selber bestimmt werden kann welche Daten es wie oft braucht. Ich zb nutze für die Momentanauswertung nur die p_ac werte der einzelnen Panele von zwei verschiedenen WR. Bekomme aber den ganzen Datensatz beider Inverter alle 2 Sekunden, da ich diese 4 Daten so oft brauche/möchte.

lumapu, 2023-09-27:
aber es ist im Endeffekt doch nur Netzwerktraffik oder? Jedes Feld einzeln einstellbar zu machen, macht das ganze Konstrukt doch nur unnötig kompliziert.

grericht, 2023-09-27:
ja ist nur netzwerktraffic.
der widerrum soll ja von HA verarbeitet werden. und da macht es schon einen unterschied ob er oft alle daten bekommt oder nur alle 10min alle daten und die aktuelle produktion (in der regel 1-5 werte) häufiger. es muss ja auf dem server auch verarbeitet werden, gespeichert werden und nach ich glaube 11tagen auch wieder gelöscht werden.
für mich würde bleiben: besser als jetzt aber ich könnte mir vorstellen, dass es umsetzbar ist entweder irgendwo einen hacken zu machen bei 
☑"neben dem üblichen senderythmus zusätzlich bei:" 
☑xxx "%änderung" oder ☑xxx"W Änderung"
 ☑"alle werte", ☑"p-ac aller panele einzeln", ☑"p-ac gesamt" senden

ich will damit nicht sagen, dass es einfach ist. Ich könnte es aktuell nicht umsetzen! ich wäre sehr dankbar und hätte großen respekt vor der leistung. Ich will nicht mitteilen, dass ich denke, dass diese funktion sehr sinnvoll wäre.

grericht, 2023-09-27:
Bei tasmota funktioniert das über befehle in der console. Weiß aber nicht ob das bei ahoydtu auch umsetzbar ist.
das hätte den scharm, dass die meisten (die die funktion wohl auch nicht brauchen) das menü nicht unübersichtlicher wird aber die funktion trotzdem zur verfügung steht. und wie gesagt: bei tasmota kann ich per konsolenbefehl für jeden wert einzeln festlegen ob er extra getriggert werden soll, wenn er entweder eine prozentuale oder bestimmte wertänderung erfährt. ich finde das nahezu perfekt gelöst.

stefan123t, 2023-12-21:
Es geht also bei diesem Feature Request darum die Daten die regelmäßig per MQTT übertragen werden in zwei Klassen einzuteilen:

* **wichtige Daten**, wie `P_AC` sobald sich die Werte des Wechselrichters entsprechend der u.a. PowerDelta Werte (*prozentual* / *absolut*) geändert haben.
* **allgemeine Daten**, wie `SSID`, `YieldTotal` und `YieldDaily` Energie (kWh), etc. die nur alle xx Minuten gesendet werden brauchen.

Hier ist das Feature / Command "PowerDelta<x>" bei Tasmota beschrieben:
**Power Monitoring - PowerDelta**
https://tasmota.github.io/docs/Commands/#power-monitoring

> **PowerDelta<x>**
> Set maximum delta of phase a in energy monitoring devices to report on active power load change while the power is ON. 
> PowerDelta will not report when the power turns off. 
> `0` = disable reporting on power change
> `1..100` = set reporting on percentage power change to send an MQTT telemetry message
> `101..32000` = set reporting on absolute power change to send an MQTT telemetry message (offset by 100, e.g., 101=1W, 207=107W)

Da wir (bisher) keine Scripting Sprache wie Tasmota verwenden, ist es auch möglich eine derartigen Schwellwert (prozentual / absolut) auch ohne den bei Tasmota verwendeten Offset von 100 zu implementieren.

grericht, 2024-04-08:
Ich wollte nur nochmal fragen, ob das Thema lösbar ist und es ein Interesse dafür gibt? Ich halte es immernoch für total sinnvoll nur die P-AC Werte so oft wie möglich zu publishen und den Rest (viel) seltener. Ich werde es aber leider nicht umsetzen können glaube ich.

CosmicDuck, 2024-10-28:
Hallo, ich habe mehrere Hichi Wlan Köpfe die meine Stromzähler auslesen. Einen wollte ich auch dazu verwenden um eine 0-Einspeisung mit meinem Victron Multi-Plus hin zu bekommen. Die 10 s waren einfach zu lang und das System hat sich dann sich immer einschwingen müssen.
Das Hichi-Tasmota hat aber eine Lösung dafür. Man kann Werteänderungen sofort als MQTT Nachricht versenden, und das nur für die Werte wo es tatsächlich gebraucht wird. Damit kann ich die Teleperiode zurück auf 60 s stellen und sende nur noch die aktuelle Leistung zum Iobroker der dann den Victron mit dem Wert versorgt.
Hier der Link dazu: [MQTT sendet zu oft oder zu selten](https://sites.google.com/view/hichi-lesekopf/faq) 

PS: Eine Lösung mit den PowerDelta habe ich nicht hin bekommen. Immer wenn ich einen Befehl versuche in der Konsole auszuführen wird der mit Unknown Command beantwortet. Ich denke das das einfach nicht im Hichi eingebaut ist oder ich was vollkommen missverstanden habe.

grericht, 2024-10-28:
es scheint hier leider, von denen den eine umsetzung möglich wäre, niemensch an einer lösung interessiert, da scheinbar das problem nicht gesehen wird. ich finde das sehr schade. für eine gute stuerung des hausverbrauches ist eine schnelle mitteilung der aktuellen produktion wichtig. die ganzen restlichen werte reichen viel seltener.

CosmicDuck, 2024-10-28:
Ja, zumindest bei den Hichi Leseköpfen (und kompatiblen) scheint das zu gehen. Bei mir sieht es im Moment nach der Anpassung des Skriptes so aus:

`11:27:45.049 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:45","LK13BE":{"Power_curr":547.00}}
11:27:46.054 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:46","LK13BE":{"Power_curr":544.00}}
11:27:47.049 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:47","LK13BE":{"Power_curr":544.00}}
11:27:48.037 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:48","LK13BE":{"Power_curr":545.00}}
11:27:48.455 MQT: tele/Hauptzaehler/STATE = {"Time":"2024-10-28T11:27:48","Uptime":"0T02:15:01","UptimeSec":8101,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MeinSachsenhagen2WLAN-KS","BSSId":"F0:A7:31:EF:AE:38","Channel":2,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}
11:27:48.465 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:48","LK13BE":{"Power_curr":545.00,"Total_in":963.6}}
11:27:49.048 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:49","LK13BE":{"Power_curr":543.00}}
11:27:50.039 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:50","LK13BE":{"Power_curr":542.00}}
11:27:51.048 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:51","LK13BE":{"Power_curr":544.00}}
11:27:52.036 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:52","LK13BE":{"Power_curr":542.00}}
11:27:53.049 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:53","LK13BE":{"Power_curr":542.00}}
11:27:54.036 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:54","LK13BE":{"Power_curr":545.00}}
11:27:55.052 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:55","LK13BE":{"Power_curr":545.00}}
11:27:56.210 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:56","LK13BE":{"Power_curr":539.00}}
11:27:57.036 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:57","LK13BE":{"Power_curr":545.00}}
11:27:58.032 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:58","LK13BE":{"Power_curr":544.00}}
11:27:58.450 MQT: tele/Hauptzaehler/STATE = {"Time":"2024-10-28T11:27:58","Uptime":"0T02:15:11","UptimeSec":8111,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MeinSachsenhagen2WLAN-KS","BSSId":"F0:A7:31:EF:AE:38","Channel":2,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}
11:27:58.459 MQT: tele/Hauptzaehler/SENSOR = {"Time":"2024-10-28T11:27:58","LK13BE":{"Power_curr":544.00,"Total_in":963.6}}
`

Aber für andere Tasmotasteckdosen habe ich das gleiche Problem mit zu langen laufzeiten.
In meinem Mietkeller habe ich nur einen Stromkreis mit einer 16 A Absicherung. 
Mein EV lasse ich dann da immer laden da dieser Stromkreis auch nach draußen zu meinem Parkplatz geht.
Ich lade mit meiner Steckerladebox schon nur mit max. 10 A. Aber wenn die Waschmaschine angeht zieht die nochmal 10 A und das EV soll dann schnellstmöglichst auf 6 A regeln und bei noch  größerer Last das Laden pausieren.
Bei 10 s Teleperiod ist die Regelung aber zu langsam und die Sicherung fliegt dann gerne mal wenn auch noch der Trockner mit macht.
Daher werde ich mir das PowerDelta nochmal auf den Steckdosenschaltern mit Tasmota ansehen.
Aber für meine 0-Einspeiseregelung hat mir die Hichi-Lösung gereicht.

CosmicDuck, 2024-10-28:
Achso, das ist ein Beispiel das das mit dem Senden von nur einzelnen Werten durchaus funktioniert, auch wenn mein eigentlich Problem die feste minimale TelePeriod war.

grericht, 2024-10-28:
es wäre ja aber praktisch wenn es diese option direkt im webserver der ahoy dtu gäbe. es bräuchte ja nur einen extra haken für "aktuelle produktionswerte sofort senden"