lumapu/ahoy #909: TSOL-M800/TSOL-M800(DE) G3 (16 stellige Seriennummer)

a-wenzel, 2023-04-27:
Hallo zusammen,

ich habe nun den TSOL-M800 bekommen (auf 600W limitiert). Dieser hat eine Seriennummer die mir R*************** beginnt und 16 Stellen hat. Dieser hat auch ein WLAN Modul, jedoch sind die Daten nur über die "wunderbare" Talent-App verfügbar welche völlig Zeitverzögerte Daten anzeigt und auch für mich nicht abgreifbar sind (HomeAssistant). Habe nun eine ganze weile rumgesucht, finde aber niemanden der scheinbar auch ein Gerät mit R beginnend hat sondern nur mit 10/11... Da das Input-field ja begrenzt ist kann ich hier auch nichts eintragen. Gibts hier schon Erfahrungen zu? Anbei mal ein Bild von der Rückseite.
[image]

rmayergfx, 2023-04-27:
Ich habs befürchtet. TSUN hat die Geräte geändert. Sie laufen auch unter dem Namen TSOL-MS800 bzw EVO, je nach Anbieter. Sobald dort "Zugriff über kostenlose Talent Home APP" steht sollte man eigentlich die Finger davon lassen, bis es hoffentlich eine Möglichkeit gibt diese per AhoyDTU abzufragen. Zumindest spart man sich das NRF24L01+.

a-wenzel, 2023-04-28:
> Ich habs befürchtet. TSUN hat die Geräte geändert. Sie laufen auch unter dem Namen TSOL-MS800 bzw EVO, je nach Anbieter. Sobald dort "Zugriff über kostenlose Talent Home APP" steht sollte man [...]

Vielen Dank für die rasche Antwort... Das hatte ich fast befürchtet. Ich hoffe das gelingt, denn diese Talent Home App ist wirklich aus der Hölle... Bis auf den Tagesstand passt da einfach gar nichts. Also bleibt erstmal die Shelly Plug S dran und muss für mich den Strom (wenn auch ungenau) messen.

lumapu, 2023-04-28:
WLAN habe ich bis jetzt nur von Deye gehört, evtl. findet man in dieser Richtung was. Ahoy kann hier erst mal nichts aurichten, tut mir leid.

icepaule, 2023-04-30:
(Leider) das gleiche Problem bei mir. Falls es hierzu eine Lösung geben sollte, wäre ich auch wahhhnsinnig interessiert. 
By the way, ändern der SN in der json-Conf und dann importieren funktioniert NICHT. ;-)

benderl, 2023-04-30:
Leider bieten diese neuen WR der GEN3 Reihe keine lokalen Schnittstellen. Die Korrekte Bezeichnung ist "MS800". Über WLAN konnte ich auch keine offenen Ports finden. Sieht also erst einmal ohne weitere Modifizierungen nach Cloud-only aus.

a-wenzel, 2023-04-30:
Ich bin in dem Thema aktuell nicht wirklich drin, aber sollte das Teil nicht im Zweifel eine hidden SSID haben? Er überträgt die WLAN-Daten ja per BT auf den Inverter. Eventuell werden etwaige Dienste eben nur auf dem eigenen WLAN ausgegeben. Ich hatte allerdings bisher keine Berührungspunkte wie man nach hidden SSIDs sucht...

Alles nur Theorie und ein Versuch wert.

benderl, 2023-04-30:
Ich konnte zumindest kein Netz ohne SSID hier ermitteln.
Die erste Kommunikation und Einrichtung macht die App per BT. Aktuell ist das der einzige Ansatzpunkt, den ich gefunden habe. Ein Portscan auf den WR im Netz ergab nicht einen Treffer.

a-wenzel, 2023-04-30:
> Ich konnte zumindest kein Netz ohne SSID hier ermitteln. Die erste Kommunikation und Einrichtung macht die App per BT. Aktuell ist das der einzige Ansatzpunkt, den ich gefunden habe. Ein Portscan auf den WR im Netz ergab nicht einen Treffer.

Das hab ich leider auch schon festgestellt. Hattest du also bereits nach einer versteckten SSID gesucht? Ich hab das tatsächlich noch nie gemacht bisher.

benderl, 2023-04-30:
Ja, gibt z.B. für Android den "Wifi Analyzer" im Play Store.

a-wenzel, 2023-04-30:
> Ja, gibt z.B. für Android den "Wifi Analyzer" im Play Store.

Bist du sicher das dies mit der App möglich ist? Alles was ich zu "discover hidden ssid" im Internet immer finde ist deutlich aufwendiger (Wireshark, Linux Tools über die man die WLAN-Karte in einen monitoring mode versetzen muss etc. z.b. (https://www.hackingarticles.in/wireless-penetration-testing-detect-hidden-ssid/)).

benderl, 2023-04-30:
Ja, ist relativ einfach. Es wird dann ein Netzt angezeigt, jedoch keine SSID dazu. Hier ein Beispiel.

[image]

benderl, 2023-04-30:
Nächster Schritt wäre, den Traffic des WR zu prüfen. Welche DNS Anfragen kommen, welche Server werden kontaktiert. Wenn wir viel Glück haben, gibt es einen Bug in der Absicherung.

a-wenzel, 2023-05-01:
Vielleicht kannst du damit was anfangen, ich bin absoluter Depp was Wireshark betrifft.

47.254.132.252 (Port: 5005) --> 192.168.20.135 (meine lokale WR IP) (Port: 60470)
192.168.20.135 (meine lokale WR IP) (Port: 60470) --> 47.254.132.252 (Port: 5005)

Aus dem ganzen Hex liest man die SN jedenfalls raus und das Keep-Alive Packages hin und her gehen, für den Rest bin ich aktuell zu "dumm" es zu lesen. Anbei noch ein kleiner Screenshot vom Hex (habe meine Seriennummer vom WR nur weggestrichen. Mit dem Rest kann ich aktuell nix anfangen). 

[image]
[image]

benderl, 2023-05-01:
Das sieht doch schon als Basis ganz gut aus.
Heißt also, der WR verbindet sich über Port 5005 mit der Cloud. Wenn Du Deine Seriennummer dort findest, riecht das irgendwie nach plain Http ohne SSL. Das wäre gut, kann ich aber noch nicht ganz glauben.

Leider habe ich nicht immer Zugriff auf meine Anlage, weil es nicht meine ist. Das Netzwerk dort hat auch keine richtige Firewall, mit der man den Traffic umleiten könnte, um etwas mehr damit zu spielen.

Vom Prinzip her könnte im internen Netz ein Webserver laufen und die Firewall leitet die Anfragen des WR dorthin weiter. Wenn es wirklich ohne Verschlüsselung übertragen wird, könnte man so schon etwas mehr herausfinden. Wenn der WR jedoch eine verschlüsselte Übertragung initiiert und auch das Serverzertifikat prüft (ist eigentlich Standard, wenn es richtig gemacht wird), wird es tricky.

benderl, 2023-05-01:
Gerade auf die Schnelle mal versucht, den Server anzusprechen. Bekomme auf dem Port leider keine Antwort. Die Verbindung wird abgelehnt. Könnte natürlich an speziellen Headern liegen. Z.B. ein spezieller User Agent oder was ganz anderes.

a-wenzel, 2023-05-02:
Ich bin nun gestern auch nicht mehr dazugekommen weiter zu machen, was ich aber versichern kann ist das die S/N dort Plain drin stand, weil mir das direkt auffiel beim durchsehen der Daten.

Was auch noch spannend ist, die großen Datenpakete wo vmtl. die Daten in die Cloud geschickt werden, werden mit großen Zeitabständen (muss noch tracken wie oft genau) verschickt. Das erklärt warum diese App einfach komplett sinnlos ist, die Live-Anzeige über die "aktuellen" Watt kann teilweiße (gefühlt) 5-10 Minuten alt sein, habe auf dem Shelly schon teilweise 500++ Watt gehabt und in der App waren es 50. Lediglich die Tageserzeugung kann man nach Sonnenuntergang gebrauchen, da die hoffentlich den absolut erzeugten Strom korrekt anzeigt.

Das würde sich auch decken mit dem, wenn man sich hier einloggt: https://www.talent-monitoring.com/ kriegt man in den Charts immer nur Werte in 15 Minuten Schritten.

Also alles in allem ist das echt unbefriedigend.

Yacki67, 2023-05-30:
ich habe meinen M800 in der Talent Home App angemeldet. Um die https://www.talent-monitoring.com/ zu nutzen, versuchte ich den Inverter in der Talent Pro App anzumelden, funktioniert scheinbar jedoch taucht das Gerät nicht in der Geräteliste auf. meine unbedarfte Frage muss den M800 erst in der Talent Home App löschen. danke

rmayergfx, 2023-06-02:
@Yacki67 
Schau mal hier: https://github.com/lumapu/ahoy/issues/939
und https://www.photovoltaikforum.com/thread/201739-talent-home-app-tsun-wechselrichter/
Wie beginnt deine Seriennummer? Bitte nicht die Komplette Seriennummer posten (Datenschutz!)

s-allius, 2023-08-11:
Hallo,
 ich habe die Datenpakete der Wechselrichter an die Cloud schickt schonmal angefangen zu analysien. Strings und Integer sind recht eindeutig zu identifizieren.
 Hier ein Übersicht über den aktuellen Stand: https://www.photovoltaikforum.com/thread/205874-tsun-tsol-m800-iobroker/?postID=3303404#post3303404

rmayergfx, 2023-08-12:
Hier noch parallel ein anderes Projekt mit Dokumentation NodeRED: https://www.reddit.com/r/Balkonkraftwerk/comments/14zp7xz/tsun_wwwtalentmonitoringcom_api_per_nodered/

Der Connection-Server logger.talent-monitoring.com scheint in Frankfurt zu stehen und zu Alicloud zu gehören: https://www.ip2location.com/47.254.132.252. Ob dann aber die Daten weiter gesendet werden und für wenn diese alles sichtbar sind habe ich noch nicht weiter geprüft. Meine Systems sind alle vom www getrennt.

lumapu, 2023-08-15:
ich habe auch schon paar Pakete versucht zu entschlüsseln. Aus dem was ich gesehen habe, ist es wirklich unverschlüsselt. Man kann kann eindeutig die Seriennummer finden

s-allius, 2023-11-14:
Aus der Analyse der Daten ist mittlerweile ein Projekt entstanden. Ich habe einen kleiner Proxy gebaut, den man in den Datenverkehr zwischen Wechselrichter und den Talent-Monitoring Service hängt und dann die Daten passt, und an einem MQTT Broker sendet. Ich nutze ihn zusammen mit Home Assistant.

Mehr Details hier: https://github.com/s-allius/tsun-gen3-proxy

Der Proxy unterstützt nur die dritte Generation von TSUN, also die mit der R17... Seriennummer

rmayergfx, 2023-11-14:
Vielen Dank.

benderl, 2023-11-15:
> Vielen Dank. Frage dazu, wenn das System eingerichtet ist und funktioniert ist es dann auch möglich die Cloud abzuklemmen, damit die Daten nur lokal gesammelt werden?

Hast Du die Beschreibung nicht gelesen?

s-allius, 2023-11-15:
> Vielen Dank. Frage dazu, wenn das System eingerichtet ist und funktioniert ist es dann auch möglich die Cloud abzuklemmen, damit die Daten nur lokal gesammelt werden?

Ja, es funktioniert auch ohne Internet und ohne TSUN Cloud. Man kann die Verbindung in der Config abschalten.

stefan123t, 2023-12-21:
Das ist vermutlich wie bei den HMS-xxxx**W** WiFi-Series Modellen von Hoymiles eine DTU-BI (built-in DTU) verbaut.
Das Gerät kommuniziert dann per WLAN und dem sog. NetworkCommands aus den .protobuf Definitionen der DTU Pro.
Für die o.g. WiFi-Series wurde das bereits in #hms-wifi-series in Discord und von @rovo89 hier: https://github.com/rovo89/hoymiles_proto

Für das spezielle Format der Seriennummern der OEM Hersteller die Hoymiles Software verwenden gibt es ebenfalls bereits einige Informationen: https://discord.com/channels/984173303147155506/992022163307638887/1017206826560934028

In der S-Miles Installer App gibt es die folgenden Drittanbieter, für die Hoymiles sozusagen OEM Geräte herstellt:
> HOYMILES_COM -> hoymiles.com
> SOLENSO_COM -> solenso.com
> APTOSSOLAR_COM -> aptosolar.com
> DURACELL_POWERCENTER_COM -> duracell.powercenter.com
> ESTAR_COM -> estar.com

Diese entsprechen nicht der bei Hoymiles üblichen Struktur u.a. aus https://www.mikrocontroller.net/attachment/559626/Micro-inverse_system_related_product_ID_coding_rules-A3-20190712.docx 

> 2. Last eight coding rules
> For products such as micro-inverters, DTUs, repeaters, etc., the rules are as follows:
> The 5th to 12th digits are the production serial number and are used as the RF communication address of the micro-inverter/DTU/repeater, among which:
> The "5" digit is the year of production (time provided by the welder), the "1" is for 2015, and so on
> The "6~7" bits are the production week (the time provided by the welding factory), and the range is "1~52".
> The "8~12" bits are the pipeline coding (currently in decimal), the micro-inverter, repeater, and DTU share the same pipeline sequence, which is coded in sequence and must not be repeated, a total of 99999 bits.

Hoymiles verwendet für die o.g. sog. Sub-Brands eine eigene Transformation der Seriennummern die @rovo89 so hier beschreibt:
https://discord.com/channels/984173303147155506/1146388875204378636/1168573556163100813

> In der Hoymiles S-Miles Installer App in der Methode ProductSNGeneUtils.sn92long():
> ```java
> public static long sn92long(String str) {
>   if (str.length() != 9) {
>     throw new IllegalArgumentException("Illegal SN");
>   }
>   int[] iArr = _FASTDICT;
>   return ((((iArr[str.charAt(0)] << 10) | (iArr[str.charAt(1)] << 5)) | iArr[str.charAt(2)]) << 32)
>           | iArr[str.charAt(8)] | (iArr[str.charAt(3)] << 25) | (iArr[str.charAt(4)] << 20)
>           | (iArr[str.charAt(5)] << 15) | (iArr[str.charAt(6)] << 10) | (iArr[str.charAt(7)] << 5);
> }
> ```
> 
> Die 9 Stellen dürften die vor dem Bindestrich sein, also `A11001TW3`, die `193` ist ein "Extend code" (Prüfnummer?).
> `_FASTDICT` ist ein Mapping für die möglichen Zeichen zu ihrer Position in folgender Liste:
> `0123456789ABCDEFGHJKLMNPRSTUVWXY`
> 
> Da fehlen ein paar Zeichen, die leicht zu verwechseln sind: I, O, Q, Z. So ergeben sich 32 Zeichen, d.h. 0 = 0, A = 10, Y = 31
> Mit dem Code oben wird ein 64-Bit-Wert erzeugt. Da der größte Wert 31 ist, passt er in 5 Bit, und so ergeben die Bitverschiebungen mehr Sinn. Wenn man Seriennummer als `ABCDEFGHI` darstellt, finden sich die jeweiligen Werte nachher so wieder: `0AAA AABB BBBC CCCC 00DD DDDE EEEE FFFF FGGG GGHH HHHI IIII`. 
> 
> ```python
> sn = 'A11001TW4'
> CHARS32 = '0123456789ABCDEFGHJKLMNPRSTUVWXY'
> CHARS64 = 'HMFLGW5XC301234567899Z67YRT2S8ABCDEFGHJKDVEJ4KQPUALMNPRSTUVWXYNB'
> 
> first_char = '1'
> i = CHARS32.find(first_char)
> sum1 = sum(ord(c) for c in sn) & 31
> sum2 = sum(CHARS32.find(c) for c in sn) & 31
> ext = first_char + CHARS64[sum1 + i] + CHARS64[sum2 + i]
> print(ext)
> ```
> 
> Übrigens sind die ersten vier Stellen wohl der Gerätetyp (A=MI, B=ES, C=RSD, D=DTU), dann zwei Stellen für den OEM, dann eine Stelle für die Fabriknummer. Die restlichen 5 Stellen sind eine Sequenznummer. Plus dann halt ein Bindestrich und der "Extend Code".
> 
> Für die "klassischen", 12-stelligen Seriennummern gibt es übrigens auch einen Extend Code:
> ```python
> sum1 = sum(ord(c) for c in sn) & 31
> sum2 = sum(CHARS32.find(c) for c in sn) & 31
> ext = CHARS64[sum1] + CHARS64[sum2]
> print(ext)
> ```
> 
> Letztlich lässt sich jede hex-numerische (12-stellige) in eine alpha-numerische (9-stellige) Seriennummer umrechnen und umgekehrt. Wenn letztere mit A bis D anfangen, entspricht das `2800 0000 0000`  bis `37FF FFFF FFFF`.
> Mein letzter Kommentar ist so gemeint: Wenn man eine 64-Bit (long) Zahl bekommt und sie benutzerfreundlich darstellen möchte, dann sollten Zahlen aus diesem Bereich alpha-numerisch dargestellt werden, alle anderen hex-numerisch.

Das müsste man also mal prüfen, ob das auch auf die von @a-wenzel o.g. Serien Nummer `R17E7305020D01CB` zutrifft.

stefan123t, 2023-12-21:
Siehe auch #1189 für die HMS-xxxx**W** WiFi-Series Modelle

stefan123t, 2024-10-27:
Vermutlich das selbe wie https://github.com/tbnobody/OpenDTU/issues/519#issuecomment-2219830167

Ich habe so eine Serial ID mit dem sn92long code aus https://github.com/tbnobody/OpenDTU/issues/1751#issue-2137109935 und https://github.com/tbnobody/OpenDTU/issues/1751#issuecomment-1946966422 probiert:

SN92 Long: `RE12508005262-1X5`
Serial ID: `61C104502000`