lumapu/ahoy #911: [Info] Yet an other PCB

DB1BMN, 2023-04-29:
## Hardware
  * [ x ] ESP8266

Modelname: WeMos D1 mini or equivalent Clone
Retailer URL: eBay, AliExpres etc.

### Antenna:
* [ x ] circuit board

### Power Stabilization:
* [ x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

[image]

Hello,
I just want to inform you, that I've recently designed also an own PCB carrying the fundamental components i.e. WeMos D1 mini module, nRF24L01+ module & Capacitor and 1.3inch OLED Display.

But it also offers some additional goodies like:
- RX/TX-LEDs (usefull while flash programming or debugging via terminal) - seen on the left short edge of the PCB
- additional space for 3 pcs WS2812B LEDs  on GPIO0 - seen on the top long edge of the PCB
  - compiling Adafruits Neopixel samle code works in Arduino IDE so far but you have to set CPU frequency to 160 MHz since the GPIO0 seems to have some timing limitations...
- additional pushbuttons on A0 port shorting a binary stepped voltage diviver - seen on the bottom edge of the PCB
- some other Gizmos not having to do with the Ahoy-Project but I wanted to test "someday" and had some space left on the PCB...

Since I have some issues with the mechanics (some mounting holes will not fit exactly - obviously there are displays with different hole spacing and dimensions) there is no project data published yet.

But the electronics works so far from V. 0.5.66 till V. 0.6.12.

Best Regards, Mark

stefan123t, 2023-12-21:
@DB1BMN do you have the schematics / PCB layout somewhere online and should we add this to the supported Hardware section ?

stefan123t, 2024-10-27:
@DB1BMN can we close the issue or do you want to donate PCB and schematics ?